raspberrypi/firmware #1687: h264_v4l2m2m -Could not find a valid device

krishnak, 2022-01-20:
Raspberry pi 4B with 8GB running Ubuntu 21.10

I was able to execute ffmpeg and use hardware acceleration by specifying this syntax

`ffmpeg -i SpaceX\ Launches\ 4K\ Demo.mp4 -vf scale="1920x1080" -r 30 -c:v h264_v4l2m2m -b:v 8M -c:a copy spacexdownscaledtohd.mp4`

This was working as intended. However in relation to my other [issue](https://github.com/raspberrypi/rpi-eeprom/issues/392) of USB hard drive boot problems, I had to update the firmware on the device.

**After the firmware update the above command doesn't work as intended, I get an error**

`[h264_v4l2m2m @ 0xaaaaf76fccb0] Could not find a valid device
[h264_v4l2m2m @ 0xaaaaf76fccb0] can't configure encoder
Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height
`

**However when I run the same command with sudo - the encoder works .**

Can you please let me know whether it is a firmware issue or something else.

**Before update**
```

$ sudo rpi-eeprom-update 
[sudo] password for krishna: 
BOOTLOADER: up to date
   CURRENT: Thursday 29 April 2021 04:11:25 PM UTC (1619712685)
    LATEST: Thursday 29 April 2021 04:11:25 PM UTC (1619712685)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
```

**After update**

```
$ rpi-eeprom-update 
BOOTLOADER: up to date
   CURRENT: Thursday 02 December 2021 11:08:03 AM UTC (1638443283)
    LATEST: Thursday 29 April 2021 04:11:25 PM UTC (1619712685)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: version unknown. Try sudo rpi-eeprom-update
   CURRENT: 
    LATEST: 
```

popcornmix, 2022-06-29:
Try:
```sudo adduser pi video```
(replace pi with your username).
If you are using the defailt user, you should probably report this to ubuntu devs, as it would make sense for user to belong to this group by default.

krishnak, 2022-06-29:
OK that solved the issue, but for some reason this occurred only after the eeprom update

closed 2022-06-29 by krishnak