raspberrypi/firmware #1690: Wiki Page - Mailbox Property Interface

michalsc, 2022-01-31:
The Wiki page [Mailbox Property Interface](https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface) says that the Tag 0x00030003, Get Voltage, returns a value which is an offset from 1.2V described in 0.025V units. After looking what results I am getting from VPU, the Get Voltage tag returns an absolute value in microvolts.

Corrected Wiki would remove the disaccord and would save some time on debugging the apparently well working mailbox call.

pelwell, 2022-01-31:
See https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface/61e6456dd263d75d5e4345c7290b6cac8089cc2b.

michalsc, 2022-01-31:
Perfect! The new wiki explains it perfectly. I will close the issue now.

Thank You Phil for fixing it :)

closed 2022-01-31 by michalsc