raspberrypi/firmware #1691: CM4s: bad dt-blob.bin?

nbuchwitz, 2022-02-09:
**Describe the bug**

After booting a CM4s there is an entry in vcdbg log which complains about a potential bad `dt-blob.bin`:
```
004290.256: gpioman: configuration attempt 1 failed (error 1) - bad dt-blob.bin?
004318.190: gpioman: configuration attempt 2 succeeded - using built-in dt-blob
```

I tried to compile a more recent version from https://github.com/raspberrypi/firmware/blob/master/extra/dt-blob.dts (`stable` and `master`), but the message still appears in the log.

**To reproduce**
Boot a CM4s device and run `vcdbg log msg`

**Expected behaviour**
`dt-blob.bin` is loaded and no message is present

**Actual behaviour**
See bug description above

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? CM4s
* Which OS and version (`cat /etc/rpi-issue`)?
```Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage4
```
* Which firmware version (`vcgencmd version`)?
```
vcos_build_version     = 71bd3109023a0c8575585ba87cbb374d2eeb038f (clean)
vcos_build_branch      = stable_20211201
debug_sym: ReadVideoCoreMemoryBySymbol: Symbol not found: 'vcos_build_application'
vcos_build_application = ?
vcos_build_date        = Dec  1 2021
vcos_build_time        = 15:01:54
vcos_build_user        = dom
vcos_build_hostname    = buildbot
vcos_build_platform    = raspberrypi_linux
debug_sym: ReadVideoCoreMemoryBySymbol: Symbol not found: 'vcos_build_type'
vcos_build_type        = ?
```
* Which kernel version (`uname -a`)?
```
Linux gaia 5.10.92-rt60-v7l+ #20 SMP PREEMPT_RT Thu, 03 Feb 2022 11:14:00 +0000 armv7l GNU/Linux
```

**Logs**
vcdbg log msg
```
003798.145: arasan: arasan_emmc_open
004275.403: brfs: File read: /mfs/sd/config.txt
004276.284: brfs: File read: 1935 bytes
004283.935: brfs: File read: /mfs/sd/dt-blob.bin
004290.256: gpioman: configuration attempt 1 failed (error 1) - bad dt-blob.bin?
004318.190: gpioman: configuration attempt 2 succeeded - using built-in dt-blob
004334.608: HDMI0:EDID error reading EDID block 0 attempt 0
004345.640: HDMI0:EDID error reading EDID block 0 attempt 1
004356.669: HDMI0:EDID error reading EDID block 0 attempt 2
004367.700: HDMI0:EDID error reading EDID block 0 attempt 3
004378.730: HDMI0:EDID error reading EDID block 0 attempt 4
004389.761: HDMI0:EDID error reading EDID block 0 attempt 5
004400.790: HDMI0:EDID error reading EDID block 0 attempt 6
004411.821: HDMI0:EDID error reading EDID block 0 attempt 7
004422.851: HDMI0:EDID error reading EDID block 0 attempt 8
004433.882: HDMI0:EDID error reading EDID block 0 attempt 9
004434.895: HDMI0:EDID giving up on reading EDID block 0
004446.161: HDMI1:EDID error reading EDID block 0 attempt 0
004452.186: HDMI1:EDID error reading EDID block 0 attempt 1
004458.209: HDMI1:EDID error reading EDID block 0 attempt 2
004464.233: HDMI1:EDID error reading EDID block 0 attempt 3
[... 77 lines cut ...]
005696.847: brfs: File read: /mfs/sd/overlays/dwc2.dtbo
005710.095: Loaded overlay 'dwc2'
005710.117: dtparam: dr_mode=host
005732.987: brfs: File read: 801 bytes
005740.280: brfs: File read: /mfs/sd/overlays/revpi-core.dtbo
005784.020: Loaded overlay 'revpi-core'
005784.064: dtparam: eth0_mac_hi=0xc83ea700
005786.814: dtparam: eth0_mac_lo=0x2ae5
006033.537: brfs: File read: 3987 bytes
006035.247: brfs: File read: /mfs/sd/cmdline.txt
006035.333: Read command line from file 'cmdline.txt':
006035.358: 'dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait nosplash plymouth.ignore-serial-consoles'
006088.746: CMA size reduced to 256MB (total_mem 1024MB, gpu_mem 76MB)
006290.215: brfs: File read: 154 bytes
007529.849: brfs: File read: /mfs/sd/kernel7l.img
007529.883: Loading 'kernel7l.img' to 0x8000 size 0x6afa20
007529.925: Device tree loaded to 0x2eff3300 (size 0xcc5d)
007540.732: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
009724.525: vchiq_core: vchiq_init_state: slot_zero = 0xded80000, is_master = 1
009732.336: TV service:host side not connected, dropping notification 0x00000002, 0x00000002, 0x00000004
009732.364: TV service:host side not connected, dropping notification 0x00000002, 0x00000002, 0x00000004
```

**Additional context**
Add any other relevant context for the problem.

pelwell, 2022-02-09:
What are you trying to achieve with dt-blob.bin? You are probably better off without it.

popcornmix, 2022-02-09:
dtblob.dts "updated" to a stale version when we bumped to 5.15 from the next branch.
I've updated master branch version to latest (which includes cm4s support).

pelwell, 2022-02-09:
Thanks. My question still remains though, @nbuchwitz - why do you need an external dt-blob.bin?

nbuchwitz, 2022-02-09:
@pelwell  Nothing specific. Just stumbled over the entry in vcdbg logs, while I was debugging another issue. The dt-blob.bin exists in our image mostly for historical reasons and probably can be removed safely. 

@popcornmix I tried with your updated version and the message is gone (also when the file is absent and the included one is used)

If there aren't any objections, I will close this issue. Thanks a lot!

pelwell, 2022-02-09:
We rarely object to issues being closed.

closed 2022-02-09 by nbuchwitz

l1k, 2022-02-09:
<blockquote>
What are you trying to achieve with dt-blob.bin? You are probably better off without it.
</blockquote>

Our use cases for `dt-blob.bin` comprise:
<ul>
<li> Enabling the camera per the <a href="https://www.raspberrypi.com/documentation/computers/compute-module.html#quickstart-guide">official documentation</a>.
<li> Configuring the pin controller to switch to the UART function for RX1 and TX1 to allow for an early console on the mini-UART.
<li> Configuring the pin controller to switch to the UART function for RTS0 and RTS1 so that it can be used as RS-485 Transmit Enable in case either of the UARTs is used as early console.
<li> Deasserting the GPIO used for RS-485 termination to ensure a working early console.
<li> Deasserting chip select of a certain ADC (Microchip MCP-3550) before power-on sequencing, lest the ADC enters continuous conversion mode.
<li> Increasing drive strength of the SPI0 and SPI2 clock to work around suboptimal wiring.
<li> Setting initial polarity of an output GPIO to work around a missing pull-down.
</ul>

We've noticed that most of what is configured in `dt-blob.bin` can also be configured in the HAT EEPROM, albeit only for bank 0. That's a problem for us. Is there any chance that you could extend the HAT EEPROM spec and firmware support to allow for setting up the upper banks as well on boot?

nbuchwitz, 2022-02-09:
Thanks for the clarification, @l1k

l1k, 2022-02-10:
@popcornmix: The <a href="https://raw.githubusercontent.com/raspberrypi/firmware/577aef2bc2f8e9dcd6955a85cf0621cc73b29db7/extra/dt-blob.dts">dt-blob.dts</a> contains two entries for the CM4S which look fishy:

```
            pin_define@EMMC_ENABLE {
               type = "internal";
               number = <49>;
            };
            pin_define@LEDS_DISK_ACTIVITY {
               type = "internal";
               number = <46>;
            };
```

On the CM1, `EMMC_ENABLE` is number 47 and on the CM3, it's number 1.
On both, `LEDS_DISK_ACTIVITY` is `absent`.

Please confirm that these two CM4S settings are correct.

Moreover, the `pin_config` section is missing the entries for `EMMC_ENABLE_N` polarity, `SD` drive strength, `SMPS_SCL/SDA` and `Hotplug`. I'm wondering if that's on purpose or an oversight?