raspberrypi/firmware #1692: This is not a issue, but i have no where to ask , My RP3 just dropped to the ground

Ma-Y, 2022-02-10:
and there's a white component got separated from the main board, i dont know what it is, but it seems that the RP3 can still boot, however the wifi signal is not really good, except the wifi signal , everytime else seems fine, i wonder if you guys know what this component is, should i send to repair or just leave it be
[image]
.

Ma-Y, 2022-02-10:
This image is not my pi, i just used some image online.

pelwell, 2022-02-10:
I don't see a part looking like that on a 3B or 3B+ that I have here. Please post two photos of the damaged board:
* a full-board shot with an indicator of where the missing part is, and
* a close up of the damaged area.

pelwell, 2022-02-10:
A colleague has informed me that the part is a filter - part of the antenna system. If the WiFi and Bluetooth are working well enough for you then hold onto your Pi. If not, then unfortunately there is no repair service - the cost of the repair would be more than the cost of a new board.

However, there is a recycling service: https://www.okdo.com/raspberry-pi-renew/

Ma-Y, 2022-02-11:
[image]
[image]

Hi this is my pi, i believe it's 3B+, i still have that part, it's really small so i put it somewhere safe, in the uploaded images that part is missing.

my Wifi and Bluetooth still works fine except the signal is not really strong.

Thanks a lot, now that i can no longer worry about the damage, i guess i'll just leave it be.

closed 2022-02-12 by Ma-Y