raspberrypi/firmware #1694: After upgrade to 5.10.92 system does not boot from external usb-attached disk

dilyanpalauzov, 2022-02-11:
I have RPI3 booting from external USB harddrive, formatted in btrfs. The sdcard contains only bootcode.bin. After each firmware upgrade, I call `upgrade-initramfs -k version`, in order to lift the initrd-image.  Before the upgrade to kernel 5.10.92 the above procedure has worked.

After upgrading to 5.10.92 the screen hangs with the images below.  Calling rpi-update, thus upgrading to 5.15.21, did not help.

[image]
[image]

pelwell, 2022-02-14:
Is btrfs.ko in your upgraded initramfs? Is the partition UUID still correct in cmdline.txt?

dilyanpalauzov, 2022-02-24:
I repeated the procedure described at https://hackmd.io/@seaQueue/HJOFQphWW on a new hardware, using the latest RPI-image. `cmdline.txt` contains
> console=serial0,115200 console=tty1 root=UUID=9a29d9c4-7607-427c-8eeb-f68d2fa40607 rootfstype=btrfs rootflags=subvol=@root rootwait

The UUID is correct.  According to the mentioned article, Raspberry Pi must use root=UUID instead of root=PARTUUID. 
```
$ sudo blkid 
/dev/sda2: UUID="9a29d9c4-7607-427c-8eeb-f68d2fa40607" UUID_SUB="7b48a943-7340-4e60-a5ab-680d93cbd4d2" BLOCK_SIZE="4096" TYPE="btrfs" PARTLABEL="Linux filesystem" PARTUUID="d84f4fb4-e355-4655-abb4-4f6d06169eb9"
/dev/sda3: UUID="2214cceb-dc3c-4d15-ae1f-128e08054e68" UUID_SUB="9322a052-34d9-4196-9c7c-103636844648" BLOCK_SIZE="4096" TYPE="btrfs" PARTLABEL="Linux filesystem" PARTUUID="e03d42c9-3838-4694-90ac-18275423fc72"
/dev/sda1: SEC_TYPE="msdos" UUID="1870-7EEA" BLOCK_SIZE="512" TYPE="vfat" PARTLABEL="boot" PARTUUID="ce7d1955-c695-448b-97ba-6df7773f4f93"
```

dilyanpalauzov, 2022-02-24:
```
$ cat /ect/fstab
proc            /proc           proc    defaults          0       0
PARTUUID=ce7d1955-c695-448b-97ba-6df7773f4f93  /boot           vfat    defaults          0       2
PARTUUID=d84f4fb4-e355-4655-abb4-4f6d06169eb9  /               btrfs    defaults,noatime  0       1
PARTUUID=e03d42c9-3838-4694-90ac-18275423fc72  /samba          btrfs    defaults,noatime  0       1
```

When I use root=PARTUUID= instead of root=UUID= in cmdconfig.txt the result is kernel panic.