raspberrypi/firmware #1698: /sys/class/net/eth0/carrier stuck in 0

MPParsley, 2022-02-20:
**Describe the bug**
/sys/class/net/eth0/carrier stuck in 0

**To reproduce**
Unplug cable and replug it, /sys/class/net/eth0/carrier becomes 1
Reboot, /sys/class/net/eth0/carrier is now stuck in 0 until the cable is unplugged and replugged again

**Expected behaviour**
Once the cable is plugged I expect /sys/class/net/eth0/carrier to be consistently 1

**Actual behaviour**
/sys/class/net/eth0/carrier stuck in 0

**System**

* Which model of Raspberry Pi? Pi4 8GB
* Which OS and version (`cat /etc/rpi-issue`)? OSMC
* Which firmware version (`vcgencmd version`)? 99c5aad7da0d1e3b2c28669a3941db77734b9f2d
* Which kernel version (`uname -a`)? Linux 5.10.78-2-osmc #1 SMP PREEMPT Wed Nov 17 04:52:11 UTC 2021 aarch64 GNU/Linux

**Additional context**
The device is running on an original Raspberri Pi 4 power supply. This issue came up after a reboot.

pelwell, 2022-02-21:
This works correctly for me on 5.10.63 (the `oldstable` rpi-firmware branch) and on 5.15.24 (the default rpi-firmware branch).

Do you possibly have some device (HDMI or USB) that might be back-powering the Pi4, preventing it from rebooting correctly? Have you tried with a different Ethernet switch?

MPParsley, 2022-02-21:
Same issue with different switch/hdmi.

pelwell, 2022-02-21:
Can you confirm the outcome of the following scenarios (some may have been covered already, but I want them all together)?

Possible outcomes are:
* carrier - `carrier` has the value `1`
* ip - `ifconfig eth0` shows a valid IP address
* both - carrier and IP address are valid
* none - neither a carrier nor an IP address

Note that it shouldn't be possible for eth0 to have an IP address without a carrier, I'm just being careful. The scenarios are (allow a reasonable interval between each step - a few seconds to detect a cable being unplugged longer for plugging it in, perhaps 10 seconds):

1. Power up the Pi4 with the ethernet cable attached, reboot.
2. Power up the Pi4 with the ethernet cable detached, plug in the cable, reboot.
3. Power up the Pi4 with the ethernet cable attached, unplug the cable, reboot, plug in the cable.
4. Power up the Pi4 with the ethernet cable attached, unplug the cable, plug the cable back in reboot.

MPParsley, 2022-02-21:
Hi @pelwell, thank you for following this up.

I started the tests but results aren’t consistent. I keep getting random carrier 1 or 0. As the carrier signal was erratic I tested several different network cables but this resulted in more random statuses, sometimes a cable worked sometimes it didn’t. At first I thought there was an issue with the cable but then I got a carrier signal and could use the device until network connection drops after a random time. I never had issues with any of the cables before.

There’s definitely a hardware issue here.