raspberrypi/firmware #1700: bootcode TFTP client RRQ race - multiple requests to the same file and Early Terminate, TFTP thrashing

arcivanov, 2022-02-21:
**Describe the bug**
Netbootcode seems to have racing concurrency issues, requesting multiple files concurrently, then aborting some or all of them, then re-requesting them again.

**To reproduce**
1. Configure DHCP
2. Configure TFTP
3. Start tcpdump/wireshark able to capture TFTP traffic.
4. Observe `ERROR 0 Early Terminate` for multiple requests after RRQ/OACK/ACK immediately after DATA block 1. Observe that this happens with random files at random times.

**Expected behaviour**
Each file is requested via TFTP one at a time and a sensible UDP retransmission policy is in place. The same file should not be requested multiple times concurrently with 2 - 3 requests for the same file in flight at a time.

**Actual behaviour**
1. Same file is requested multiple times with a difference of 5 - 9 ms.
2. Sometimes all in-flight requests are aborted with Early Terminate, sometimes only a few of them.
3. Which files and how many concurrent requests to them are in flight varies with each boot. 

**System**

* Which model of Raspberry Pi? 
 Pi3B+, Pi2Bv1.1
* Which OS and version (`cat /etc/rpi-issue`)? 
bootcode.bin https://github.com/raspberrypi/firmware/blob/a6496ae5cd5e9b4cd5b38f7274fa94dce315fa7a/boot/bootcode.bin
* Which firmware version (`vcgencmd version`)?
N/A
* Which kernel version (`uname -a`)?
N/A

**Logs**
Attached.

**Additional context**
None.

arcivanov, 2022-02-21:
.129 is RPi
.59 is DHCP/TFTP
```
./main.py 
<snip DHCP negotiation>
tftp: ('192.168.159.129', 49153) b'\x00\x0189eb9bb9/start.elf\x00octet\x00tsize\x000\x00'
tftp read: ('192.168.159.59', 47288) ('192.168.159.129', 49153) 89eb9bb9/start.elf  ==>  start.elf
tftp read connect: ('192.168.159.59', 47288) ('192.168.159.129', 49153) start.elf octet tsize= 0 blksize= None
tftp: ('192.168.159.129', 49153) start.elf 6 0 b'\x00\x06tsize\x002965216\x00'
tftp: ('192.168.159.129', 49153) start.elf ACK block 0
tftp read options: ('192.168.159.59', 47288) ('192.168.159.129', 49153) start.elf octet tsize= 2965216 blksize= 512
tftp: ('192.168.159.129', 49154) b'\x00\x0189eb9bb9/start.elf\x00octet\x00tsize\x000\x00'
Traceback (most recent call last):
  File "src/gevent/greenlet.py", line 906, in gevent._gevent_cgreenlet.Greenlet.run
  File "/home/user/Documents/src/pxefarm/src/main/python/_main.py", line 456, in setup_tftp_transfer
    send(s, packet, block)
  File "/home/user/Documents/src/pxefarm/src/main/python/_main.py", line 380, in send
    handle_ack(block, verbose)
  File "/home/user/Documents/src/pxefarm/src/main/python/_main.py", line 400, in handle_ack
    raise RuntimeError(f"tftp: {addr} {filename} block {block} bytes sent {bytes_sent} ERROR {ack_packet}")
RuntimeError: tftp: ('192.168.159.129', 49153) start.elf block 1 bytes sent 512 ERROR b'\x00\x05\x00\x00Early terminate\x00'
2022-02-21T01:07:40Z <Greenlet at 0x7fbc263ac700: setup_tftp_transfer(('192.168.159.129', 49153), '89eb9bb9/start.elf', 'octet', tsize=0)> failed with RuntimeError

tftp read: ('192.168.159.59', 47868) ('192.168.159.129', 49154) 89eb9bb9/start.elf  ==>  start.elf
tftp read connect: ('192.168.159.59', 47868) ('192.168.159.129', 49154) start.elf octet tsize= 0 blksize= None
tftp: ('192.168.159.129', 49154) start.elf 6 0 b'\x00\x06tsize\x002965216\x00'
tftp: ('192.168.159.129', 49154) start.elf ACK block 0
tftp read options: ('192.168.159.59', 47868) ('192.168.159.129', 49154) start.elf octet tsize= 2965216 blksize= 512
tftp: ('192.168.159.129', 49155) b'\x00\x0189eb9bb9/autoboot.txt\x00octet\x00tsize\x000\x00'
Traceback (most recent call last):
  File "src/gevent/greenlet.py", line 906, in gevent._gevent_cgreenlet.Greenlet.run
  File "/home/user/Documents/src/pxefarm/src/main/python/_main.py", line 456, in setup_tftp_transfer
    send(s, packet, block)
  File "/home/user/Documents/src/pxefarm/src/main/python/_main.py", line 380, in send
    handle_ack(block, verbose)
  File "/home/user/Documents/src/pxefarm/src/main/python/_main.py", line 400, in handle_ack
    raise RuntimeError(f"tftp: {addr} {filename} block {block} bytes sent {bytes_sent} ERROR {ack_packet}")
RuntimeError: tftp: ('192.168.159.129', 49154) start.elf block 1 bytes sent 512 ERROR b'\x00\x05\x00\x00Early terminate\x00'
2022-02-21T01:07:40Z <Greenlet at 0x7fbc263ac940: setup_tftp_transfer(('192.168.159.129', 49154), '89eb9bb9/start.elf', 'octet', tsize=0)> failed with RuntimeError

tftp read: ('192.168.159.59', 47118) ('192.168.159.129', 49155) 89eb9bb9/autoboot.txt  ==>  autoboot.txt
tftp read connect: ('192.168.159.59', 47118) ('192.168.159.129', 49155) autoboot.txt octet tsize= 0 blksize= None
tftp: ('192.168.159.129', 49155) autoboot.txt no such file!
tftp: ('192.168.159.129', 49156) b'\x00\x0189eb9bb9/config.txt\x00octet\x00tsize\x000\x00'
tftp read: ('192.168.159.59', 59077) ('192.168.159.129', 49156) 89eb9bb9/config.txt  ==>  config.txt
tftp read connect: ('192.168.159.59', 59077) ('192.168.159.129', 49156) config.txt octet tsize= 0 blksize= None
tftp: ('192.168.159.129', 49156) config.txt 6 0 b'\x00\x06tsize\x002075\x00'
tftp: ('192.168.159.129', 49156) config.txt ACK block 0
tftp read options: ('192.168.159.59', 59077) ('192.168.159.129', 49156) config.txt octet tsize= 2075 blksize= 512
tftp: ('192.168.159.129', 49156) config.txt blocks=5, bytes=2075 DONE!
tftp: ('192.168.159.129', 49157) b'\x00\x0189eb9bb9/recovery.elf\x00octet\x00tsize\x000\x00'
tftp read: ('192.168.159.59', 39741) ('192.168.159.129', 49157) 89eb9bb9/recovery.elf  ==>  recovery.elf
tftp read connect: ('192.168.159.59', 39741) ('192.168.159.129', 49157) recovery.elf octet tsize= 0 blksize= None
tftp: ('192.168.159.129', 49157) recovery.elf no such file!
tftp: ('192.168.159.129', 49158) b'\x00\x0189eb9bb9/start.elf\x00octet\x00tsize\x000\x00'
tftp read: ('192.168.159.59', 50998) ('192.168.159.129', 49158) 89eb9bb9/start.elf  ==>  start.elf
tftp read connect: ('192.168.159.59', 50998) ('192.168.159.129', 49158) start.elf octet tsize= 0 blksize= None
tftp: ('192.168.159.129', 49158) start.elf 6 0 b'\x00\x06tsize\x002965216\x00'
tftp: ('192.168.159.129', 49158) start.elf ACK block 0
tftp read options: ('192.168.159.59', 50998) ('192.168.159.129', 49158) start.elf octet tsize= 2965216 blksize= 512
Removed request (b"\xb8'\xeb\xeb\x9b\xb9", b'&\xf3\x039') at 343105.882668642
tftp: ('192.168.159.129', 49159) b'\x00\x0189eb9bb9/fixup.dat\x00octet\x00tsize\x000\x00'
tftp: ('192.168.159.129', 49158) start.elf blocks=5792, bytes=2965216 DONE!
tftp read: ('192.168.159.59', 47908) ('192.168.159.129', 49159) 89eb9bb9/fixup.dat  ==>  fixup.dat
tftp read connect: ('192.168.159.59', 47908) ('192.168.159.129', 49159) fixup.dat octet tsize= 0 blksize= None
tftp: ('192.168.159.129', 49159) fixup.dat 6 0 b'\x00\x06tsize\x007221\x00'
tftp: ('192.168.159.129', 49159) fixup.dat ACK block 0
tftp read options: ('192.168.159.59', 47908) ('192.168.159.129', 49159) fixup.dat octet tsize= 7221 blksize= 512
[... cut ...]
tftp read connect: ('192.168.159.59', 41893) ('192.168.159.129', 49182) kernel7.img octet tsize= 0 blksize= None
tftp: ('192.168.159.129', 49182) kernel7.img 6 0 b'\x00\x06tsize\x006596992\x00'
tftp: ('192.168.159.129', 49182) kernel7.img ACK block 0
tftp read options: ('192.168.159.59', 41893) ('192.168.159.129', 49182) kernel7.img octet tsize= 6596992 blksize= 512
tftp: ('192.168.159.129', 49182) kernel7.img blocks=12885, bytes=6596992 DONE!
Removed boot offer 192.168.159.129 at 343156.899704149
```

arcivanov, 2022-02-21:
[image]

arcivanov, 2022-02-21:
[image]

HayleyDeckers, 2022-12-05:
can reproduce on the CM4 and Pi 4. It does "work" but will error out transfers at random as described. Behaves the same regardless of TFTP options (tsize, blocksize) on or off.