raspberrypi/firmware #1702: bcm2835-system-timer absent in some DTBs

asurati, 2022-03-08:
Hello,

Is there a reason for the bcm2835-system-timer device to be absent in the DTBs for RPi 2b/3b (and may be more)? It is present in (at least) RPi1b and RPi4b.

Thanks.

asurati, 2022-03-08:
Thanks.

It seems I saw the descrepancy the wrong way around. I should have asked about the reason it was _present_ in RPi4b.

There's no need for me to want the DTB changed, or to want to use the bcm2835-system-timer on machines above RPi1b. It is more about a common design for handling diverse interrupt-configuration expressions (for e.g. ACPI/DT + GIC/BCMINTC + Device) for a single device than it is about using a specific device; I happen to choose the timer for that purpose because it is the very first (and at present, the only) device that was being initialized after initializing the interrupt controllers.

pelwell, 2022-03-08:
I have no problem with also deleting or disabling the old timer node for BCM2711-based devices - we just need to append the following to `bcm2711-rpi-ds.dtsi`:
```
&system_timer {
	status = "disabled";
};
```

pelwell, 2022-03-08:
Fixed in rpi-5.15.y (and will propagate to newer kernels).

closed 2022-03-09 by asurati