raspberrypi/firmware #1706: Network booting no longer works when Data drive attached

gorky, 2022-03-16:
**Describe the bug**
Add a clear and concise description of what you think the bug is.
name -a
Linux web 5.10.103-v8+ #1530 SMP PREEMPT Tue Mar 8 13:06:35 GMT 2022 aarch64 GNU/Linux

3 PI 3B+s netbooting.
2 are configured with no external devices
1 has a 1TB harddrive attached to a USB port.

Running for 2+ years.  

Recent updates has the PI with the attached HD refusing to boot from the network.  There are no logs available as the boot doesn't start.  No blink codes from the LEDs.

**To reproduce**
Configure PI for network booting.
- Configure SD card for initial boot
- program_usb_boot_mode=1 in /boot/config.txt
- Attach HardDrive to USB Port
- Boot from SD Card
Boot from Network
- Power down
- Remove SD Card
- Power up

**Expected/Previous behaviour**
- PI powers on
- LED Lights come on
- PI sends DHCP Request (can be seen in server system logs)
- DHCP Server replies with Netboot information
- PI contacts TFTP Server and starts downloading kernel and boot files (upload can be seen in server system logs)
- PI finishes downloading kernel and boot files, connects to NFS Server and finishes booting

**Actual behaviour**
When SD card has been removed,
- PI powers on
- LED Lights come on
- Nothing else happens.  No DHCP Request logged on the server, no blink error codes (LEDs stay steadily on), no activity, nothing on the screen.

When SD card is present, PI boots.

**System**
System Information
------------------

Raspberry Pi 3 Model B Rev 1.2
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage2

Linux web 5.10.103-v8+ #1530 SMP PREEMPT Tue Mar 8 13:06:35 GMT 2022 aarch64 GNU/Linux
Revision	: a02082
Serial		: 00000000acd82f6b
Model		: Raspberry Pi 3 Model B Rev 1.2
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Jan 20 2022 13:58:22 
Copyright (c) 2012 Broadcom
version bd88f66f8952d34e4e0613a85c7a6d3da49e13e2 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0
[... 173 lines cut ...]
GPIO 41: level=0 fsel=4 alt=0 func=PWM1
GPIO 42: level=1 fsel=4 alt=0 func=GPCLK1
GPIO 43: level=1 fsel=4 alt=0 func=GPCLK2
GPIO 44: level=1 fsel=0 func=INPUT
GPIO 45: level=1 fsel=0 func=INPUT
BANK2 (GPIO 46 to 53):
GPIO 46: level=1 fsel=0 func=INPUT
GPIO 47: level=1 fsel=1 func=OUTPUT
GPIO 48: level=0 fsel=4 alt=0 func=SD0_CLK
GPIO 49: level=1 fsel=4 alt=0 func=SD0_CMD
GPIO 50: level=1 fsel=4 alt=0 func=SD0_DAT0
GPIO 51: level=1 fsel=4 alt=0 func=SD0_DAT1
GPIO 52: level=1 fsel=4 alt=0 func=SD0_DAT2
GPIO 53: level=1 fsel=4 alt=0 func=SD0_DAT3

vcdbg log messages
------------------

Failed to allocate -1716518942 bytes for message buffer

**Logs**
No logs from Network boot attempt -- no DHCP request sent to DHCP Server

**Additional context**
None known.

pelwell, 2022-03-17:
A 3B+ with no SD card has no firmware to update, unlike a Pi 4 which has an EEPROM.

1. What do you think has changed since it apparently used to work?
2. What happens when you try to boot without the external drive attached?
3. How are the devices powered?

gorky, 2022-04-20:
My apologies for the delay.  The only thing that changed was a kernel update.  Works when the External Drive is attached.  Powered via custom powersupply (setup in 2020).
NOTE: is now working with update 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64

/boot/config.txt -- program_usb_boot_mode=1
This implies SOMETHING is programmable on the board.

Also note: this is not the first time I've had network and even boot from external drive fail on an upgrade.

Since this is working as of 5.15.32-v8+, I no longer have an issue and this ticket can be closed as fixed if you would like.  Although I am curious as to what changed between releases.

timg236, 2024-01-06:
Closed due to inactivity

closed 2024-01-06 by timg236