raspberrypi/firmware #1707: Backport of Video4Linux2 changes (gstreamer-plugins-good) needs one more patch

fduncanh, 2022-03-18:
@6by9 

there are two other of @ndufresne 's commits to (gstreamer)  master that need to be backported (to complete your backport that was pushed to RPi OS updates)  I've checked that they work in 1.18.4 and can be added to your backport

The first one is essential, without it there is a crash if the size of the video is changed, which requires a caps renegotiation

https://gitlab.freedesktop.org/gstreamer/gstreamer/-/commit/e2b2ff26c99723cbbe7129412cc707d0de6d78dd

the second one didn't cause me any issues, but should also be added.

https://gitlab.freedesktop.org/gstreamer/gstreamer/-/commit/d394b8b4bdc229ab3a47dd6b666157b57b3eb8a3

_Originally posted by @fduncanh in https://github.com/raspberrypi/firmware/issues/1673#issuecomment-1068899254_

6by9, 2022-03-29:
Additional patches sent to the relevant person to be merged into Raspberry Pi OS.

fduncanh, 2022-07-05:
@6by9 

Those patches don't yet seem to have made it to  Rasperry Pi OS updates yet.