raspberrypi/firmware #1710: [Ubuntu]: PoE+ hat, fan not spinning even at 80C

samip5, 2022-03-26:
**Describe the bug**
My PoE+ hat doesn't seem to be spinning at all, even when SoC/CPU reaches 80C

**To reproduce**
Install Ubuntu Server 20.04
Install PoE+ hat
Download from this repo master and enable the overlay
Boot and look that it doesn't start spinning

**Expected behaviour**
I expected it to start spinning normally.

**Actual behaviour**
It doesn't start spinning at all, even though the whole board is too hot to touch aka 80C+

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? Pi 4, 8GB
* Which OS and version (`cat /etc/rpi-issue`)? Ubuntu Server 20.04 LTS
* Which firmware version (`vcgencmd version`)?
```
Nov 18 2021 16:16:49 
Copyright (c) 2012 Broadcom
version d9b293558b4cef6aabedcc53c178e7604de90788 (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
`Linux ubuntu 5.4.0-1056-raspi #63-Ubuntu SMP PREEMPT Mon Mar 14 07:48:06 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux`

**Additional context**
Raspinfo paste: https://gist.github.com/samip5/328dc1f789417c70cb933163df5e68c5
measure_temp: `temp=84.2'C`

dtoverlay -l: `No overlays loaded`

samip5, 2022-03-26:
Tested also on Ubuntu 21.04, still no overlay being loaded and loading it fails.

```
DTOVERLAY[debug]: using platform 'bcm2711'
DTOVERLAY[debug]: overlay map loaded
run_cmd: which dtoverlay-pre >/dev/null 2>&1 && dtoverlay-pre
DTOVERLAY[debug]: loading file 'rpi-poe.dtbo'
DTOVERLAY[debug]: wrote 2673 bytes to '/tmp/.dtoverlays/0_rpi-poe.dtbo'
DTOVERLAY[debug]: wrote 2673 bytes to '/sys/kernel/config/device-tree/overlays/0_rpi-poe/dtbo'
* Failed to apply overlay '0_rpi-poe' (kernel)
run_cmd: which dtoverlay-post >/dev/null 2>&1 && dtoverlay-post
```

pelwell, 2022-03-26:
The PoE+ HAT needs a different overlay - `rpi-poe-plus`.
Notr that the temperature threshold parameters might not work unless your firmware is no older than March 2022.

samip5, 2022-03-26:
> The PoE+ HAT needs a different overlay - `rpi-poe-plus`. Notr that the temperature threshold parameters might not work unless your firmware is no older than March 2022.

I'm aware, but I'm saying neither loads.

```
DTOVERLAY[debug]: using platform 'bcm2711'
DTOVERLAY[debug]: overlay map loaded
run_cmd: which dtoverlay-pre >/dev/null 2>&1 && dtoverlay-pre
DTOVERLAY[debug]: loading file 'rpi-poe-plus.dtbo'
DTOVERLAY[debug]: wrote 2938 bytes to '/tmp/.dtoverlays/0_rpi-poe-plus.dtbo'
DTOVERLAY[debug]: wrote 2938 bytes to '/sys/kernel/config/device-tree/overlays/0_rpi-poe-plus/dtbo'
* Failed to apply overlay '0_rpi-poe-plus' (kernel)
run_cmd: which dtoverlay-post >/dev/null 2>&1 && dtoverlay-post
```

pelwell, 2022-03-26:
That's probably because the device tree files that come with your old kernel don't match. Running `dmesg` might give a more detailed error message.

samip5, 2022-03-26:
> That's probably because the device tree files that come with your old kernel don't match. Running `dmesg` might give a more detailed error message.

dmesg says nothing relating the overlays nor poe.

pelwell, 2022-03-27:
> The gist link has dmesg output too.

But any such messages will only appear (if at all) after using the dtoverlay command.

There should be no need to load the overlay manually - the firmware does that automatically. That may even be the cause of the load failure. Are you sure that your old kernel includes all of the necessary drivers?

samip5, 2022-03-27:
> But any such messages will only appear (if at all) after using the dtoverlay command.
>
> There should be no need to load the overlay manually - the firmware does that automatically. That may even be [...]

What are all of the needed drivers? The Pi is now running Ubuntu 21.04, and still it doesn't appear to work and dmesg says nothing about it, even after the dtoverlay command.

samip5, 2022-03-27:
It didin't want to find newer eeprom either..

New gist with raspinfo and after trying to enable the overlay manually: https://gist.github.com/samip5/399886441d3345fadc869d1702f37155

juergh, 2022-03-30:
Please report this in the Ubuntu bug tracker by running the following on the affected system:
```
$ ubuntu-bug linux-raspi
```

samip5, 2022-03-30:
> Please report this in the Ubuntu bug tracker by running the following on the affected system:
> 
> ```
> $ ubuntu-bug linux-raspi
> ```

https://bugs.launchpad.net/ubuntu/+source/linux-raspi/+bug/1967178

closed 2022-03-30 by samip5