raspberrypi/firmware #1711: Raspberry Pi 3 - 0x0b vs 0xef

kallisti5, 2022-03-27:
**Describe the bug**
What partition types will the Raspberry Pi 3 boot from?  Haiku moved to using 0xef for their boot partition which works on pretty much every SoC we can test on except the Raspberry Pi 3.

**To reproduce**
Setup an SD card with the following format:

* 0xef  EFI System Partition FAT32
  * https://github.com/haiku/firmware/raw/master/u-boot/arm/rpi3/README.md
  * https://github.com/haiku/firmware/raw/master/u-boot/arm/rpi3/config.txt
  * https://github.com/haiku/firmware/raw/master/u-boot/arm/rpi3/u-boot.bin
  * https://github.com/raspberrypi/firmware/raw/07937c7d48bcd44cc1015c6257ae2cfa5da51298/boot/bootcode.bin
  * https://github.com/raspberrypi/firmware/raw/07937c7d48bcd44cc1015c6257ae2cfa5da51298/boot/start.elf
  * https://github.com/raspberrypi/firmware/raw/master/boot/LICENCE.broadcom
* 0xeb Haiku's partition.

**Expected behavior**
The Raspberry Pi 3 should launch start.elf as usual.

**Actual behavior**
Green activity LED remains off.   Changing the first boot partition from 0xef to 0x0b fixes the issue and u-boot starts as expected.

pelwell, 2022-03-27:
If you are referring to the initial step of loading bootcode.bin from the attached storage (usually an SD card, but could be a USB MSD or a TFTP server) then there is no possibility to change the behaviour - that code is in the ROM.

kallisti5, 2022-03-27:
@pelwell at minimum, i'm seeing any documentation from Rasberry Pi foundation on what the requirements are for the boot partition.  We know the Rpi4 will boot from a partition marked 0x0b or 0xef.  It appears that the Rpi3 only boots from 0x0b.

These closed early-loader blobs are annoying

pelwell, 2022-03-28:
The ROM does not recognise the 0xef partition type, and there is no possibility to change this. It was a later addition to the Pi 4 EEPROM after a user request (https://github.com/raspberrypi/rpi-eeprom/issues/126).

kallisti5, 2022-03-28:
@pelwell thanks for confirming.. I guess we're going to have to add a partition ID change when users try to use our images on the Rpi3.

ARM needs to standardize on EFI... between AllWinner's fixed offsets and Broadcom's blobs the ecosystem is a mess.

closed 2022-03-28 by kallisti5