raspberrypi/firmware #1712: RPi3 B 1.2: 'flip_done timed out' when turning monitor off and on again

z0rti, 2022-03-27:
I got this error with my RPi3 using 'dtoverlay=vc4-kms-v3d' (on bullseye)
I'm using 'hdmi_mode=4' in config.txt
It boots up fine (HDMI monitor is on) 
but when i turn the monitor off and on again i don't get a picture anymore, until i do a hard reset (pull power off)

```
Mar 26 20:08:28 pi3 kernel: [ 2500.612845] ------------[ cut here ]------------
Mar 26 20:08:28 pi3 kernel: [ 2500.612857] Timeout waiting for VC4_HDMI_SCHEDULER_CONTROL_HDMI_ACTIVE
Mar 26 20:08:28 pi3 kernel: [ 2500.612978] WARNING: CPU: 2 PID: 6289 at drivers/gpu/drm/vc4/vc4_hdmi.c:1336 vc4_hdmi_encoder_post_crtc_enable+0x1064/0x11a8 [vc
4]
Mar 26 20:08:28 pi3 kernel: [ 2500.612983] Modules linked in: ccm rfcomm algif_hash algif_skcipher af_alg bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q
garp stp llc snd_soc_hdmi_codec rt2800usb rt2800lib rt2x00usb rt2x00lib joydev brcmfmac mac80211 brcmutil crc_ccitt vc4 libarc4 cec cfg80211 drm_kms_helper rfk
ill snd_soc_core snd_compress snd_pcm_dmaengine syscopyarea sysfillrect sysimgblt fb_sys_fops raspberrypi_hwmon i2c_bcm2835 bcm2835_codec(C) bcm2835_v4l2(C) bc
m2835_isp(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 snd_bcm2835(C) videobuf2_common snd_pcm
snd_timer snd vc_sm_cma(C) videodev mc uio_pdrv_genirq uio i2c_dev aes_neon_bs aes_neon_blk crypto_simd cryptd drm drm_panel_orientation_quirks backlight fuse
ip_tables x_tables ipv6
Mar 26 20:08:28 pi3 kernel: [ 2500.613222] CPU: 2 PID: 6289 Comm: kworker/2:2 Tainted: G         C        5.10.92-v8+ #1514
Mar 26 20:08:28 pi3 kernel: [ 2500.613226] Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
Mar 26 20:08:28 pi3 kernel: [ 2500.613294] Workqueue: events output_poll_execute [drm_kms_helper]
Mar 26 20:08:28 pi3 kernel: [ 2500.613304] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO BTYPE=--)
Mar 26 20:08:28 pi3 kernel: [ 2500.613331] pc : vc4_hdmi_encoder_post_crtc_enable+0x1064/0x11a8 [vc4]
Mar 26 20:08:28 pi3 kernel: [ 2500.613357] lr : vc4_hdmi_encoder_post_crtc_enable+0x1064/0x11a8 [vc4]
Mar 26 20:08:28 pi3 kernel: [ 2500.613361] sp : ffffffc0132538a0
Mar 26 20:08:28 pi3 kernel: [ 2500.613365] x29: ffffffc0132538a0 x28: 0000000000000001
Mar 26 20:08:28 pi3 kernel: [ 2500.613375] x27: ffffffc0090e43e8 x26: 0000000000000500
Mar 26 20:08:28 pi3 kernel: [ 2500.613385] x25: ffffff80076d3c50 x24: ffffff80076d3c30
Mar 26 20:08:28 pi3 kernel: [ 2500.613395] x23: ffffffc010e1d178 x22: ffffff80076d3c28
Mar 26 20:08:28 pi3 kernel: [ 2500.613405] x21: 0000024637dffa7e x20: ffffff80076d34b8
Mar 26 20:08:28 pi3 kernel: [ 2500.613415] x19: ffffff80076d3080 x18: 0000000000000000
Mar 26 20:08:28 pi3 kernel: [ 2500.613425] x17: 0000000000000000 x16: 0000000000000000
Mar 26 20:08:28 pi3 kernel: [ 2500.613434] x15: 0000007f8adf46a8 x14: 0000000000000000
Mar 26 20:08:28 pi3 kernel: [ 2500.613444] x13: 0000000000000000 x12: ffffffc0113154f8
Mar 26 20:08:28 pi3 kernel: [ 2500.613453] x11: 0000000000000003 x10: ffffffc0112fd4b8
Mar 26 20:08:28 pi3 kernel: [ 2500.613463] x9 : ffffffc0100e6558 x8 : 0000000000017fe8
Mar 26 20:08:28 pi3 kernel: [ 2500.613473] x7 : c0000000ffffefff x6 : ffffffc013253500
Mar 26 20:08:28 pi3 kernel: [ 2500.613483] x5 : 0000000000000001 x4 : 0000000000000000
Mar 26 20:08:28 pi3 kernel: [ 2500.613492] x3 : 0000000000000001 x2 : 0000000000000000
Mar 26 20:08:28 pi3 kernel: [ 2500.613502] x1 : bcd5c0829e029300 x0 : 0000000000000000
Mar 26 20:08:28 pi3 kernel: [ 2500.613511] Call trace:
Mar 26 20:08:28 pi3 kernel: [ 2500.613539]  vc4_hdmi_encoder_post_crtc_enable+0x1064/0x11a8 [vc4]
Mar 26 20:08:28 pi3 kernel: [ 2500.613565]  vc4_crtc_atomic_enable+0x3d0/0x4a8 [vc4]
Mar 26 20:08:28 pi3 kernel: [ 2500.613609]  drm_atomic_helper_commit_modeset_enables+0x210/0x258 [drm_kms_helper]
Mar 26 20:08:28 pi3 kernel: [ 2500.613635]  vc4_atomic_complete_commit+0x1dc/0x6a8 [vc4]
Mar 26 20:08:28 pi3 kernel: [ 2500.613660]  vc4_atomic_commit+0x1b0/0x200 [vc4]
Mar 26 20:08:28 pi3 kernel: [ 2500.613785]  drm_atomic_commit+0x54/0x68 [drm]
Mar 26 20:08:28 pi3 kernel: [ 2500.613877]  drm_client_modeset_commit_atomic.isra.16+0x1ec/0x290 [drm]
Mar 26 20:08:28 pi3 kernel: [ 2500.613963]  drm_client_modeset_commit_locked+0x68/0x1a8 [drm]
Mar 26 20:08:28 pi3 kernel: [ 2500.614050]  drm_client_modeset_commit+0x38/0x60 [drm]
Mar 26 20:08:28 pi3 kernel: [ 2500.614094]  __drm_fb_helper_restore_fbdev_mode_unlocked+0xd4/0x118 [drm_kms_helper]
Mar 26 20:08:28 pi3 kernel: [ 2500.614139]  drm_fb_helper_set_par+0x44/0x80 [drm_kms_helper]
Mar 26 20:08:28 pi3 kernel: [ 2500.614184]  drm_fb_helper_hotplug_event.part.24+0xb4/0xe8 [drm_kms_helper]
Mar 26 20:08:28 pi3 kernel: [ 2500.614228]  drm_fbdev_client_hotplug+0x44/0x1b8 [drm_kms_helper]
Mar 26 20:08:28 pi3 kernel: [ 2500.614314]  drm_client_dev_hotplug+0x80/0xc8 [drm]
Mar 26 20:08:28 pi3 kernel: [ 2500.614357]  drm_kms_helper_hotplug_event+0x3c/0x50 [drm_kms_helper]
Mar 26 20:08:28 pi3 kernel: [ 2500.614401]  output_poll_execute+0x1e0/0x230 [drm_kms_helper]
Mar 26 20:08:28 pi3 kernel: [ 2500.614413]  process_one_work+0x1f4/0x4d8
Mar 26 20:08:28 pi3 kernel: [ 2500.614419]  worker_thread+0x50/0x480
Mar 26 20:08:28 pi3 kernel: [ 2500.614427]  kthread+0x148/0x158
Mar 26 20:08:28 pi3 kernel: [ 2500.614435]  ret_from_fork+0x10/0x30
Mar 26 20:08:28 pi3 kernel: [ 2500.614440] ---[ end trace cca87e7ca102e151 ]---
Mar 26 20:08:28 pi3 kernel: [ 2500.614515] ------------[ cut here ]------------
[...]
Mar 26 20:08:39 pi3 kernel: [ 2510.824308] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:76:crtc-3] flip_done timed out
```
```
$ lsb_release -d
Description:    Debian GNU/Linux 11 (bullseye)
$ uname -a
Linux pi3 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux

```
So i'm currently using 'dtoverlay=vc4-fkms-v3d' where this doesn't happen.

popcornmix, 2022-03-27:
Can you try running `rpi-update` to switch to the 5.15 kernel branch? Is the issue still there?
This is intended to become the stable kernel in apt in the very near future.

simnalamburt, 2022-03-28:
@popcornmix I could reproduce this issue with `5.15.30-v8+`.

popcornmix, 2022-03-28:
That's good. That should be default kernel pretty soon.

pelwell, 2022-03-28:
It doesn't sound good if @simnalamburt can reproduce the issue with 5.15.30.

popcornmix, 2022-03-28:
Sorry - I misread his post. I'll see if I can reproduce.

popcornmix, 2022-03-28:
I didn't see this on my monitor when switching into/out of standby.
I didn't see when switching AV indput on monitor.
I didn't see when unplugging/replugging hdmi cable.
I did see it when unplugging monitor from mains / plugging back in (specifically on plugging back in).

(note: what happens to hotplug signal and ability to read edid in each of these cases tends to be monitor specific).