raspberrypi/firmware #1715: This demo application is not available on the Pi4

xingyun86, 2022-04-02:
[pi4]
I want to use egl and dispmanx create window and set window's layer.But I found that it cannot be success. 
I use /opt/vc/.../hello_triangle2 to compile and run it. But response me "This demo application is not available on the Pi4" 。
How I can do? Please help me.

popcornmix, 2022-04-02:
dispmanx was a proprietary interface only used on raspberry pi.
That interface is deprecated (and was never supported for gl on pi4) in favour of standard linux APIs.
Some discussion of alternatives  [here](https://forums.raspberrypi.com/viewtopic.php?t=243707)

closed 2022-04-02 by popcornmix

xingyun86, 2022-04-06:
> 
But I cannot set my app ui‘s layer on monitor？Could you give me some supply？
I tried drm， but also give me show：
```
pi@raspberrypi: $ sudo ./drmtest 
resolution: 1280x1024
* failed to add service - already in use?
```

popcornmix, 2022-04-06:
Are you launching from X? Try launching from console outisde of X (or change VT with ctrl-alt-F<n>)