raspberrypi/firmware #1716: This demo application is not available on the Pi4, and drm not working!!!

xingyun86, 2022-04-06:
[pi4]
I want to use egl and dispmanx create window and set window's layer.But I found that it cannot be success.
I use /opt/vc/.../hello_triangle2 to compile and run it. But response me "This demo application is not available on the Pi4" 。
How I can do? Please help me.
[version info]

BOOTLOADER: up to date
   CURRENT: Thu 29 Apr 2021 04:11:25 PM UTC (1619712685)
    LATEST: Thu 29 Apr 2021 04:11:25 PM UTC (1619712685)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
===============================================
pi@raspberrypi: $ uname -a
Linux raspberrypi 5.10.17-v7l+ https://github.com/raspberrypi/firmware/issues/1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux
================================================================
But I cannot set my app ui‘s layer on monitor？Could you give me some supply？
I tried drm， but also give me show：

pi@raspberrypi: $ sudo ./drmtest 
resolution: 1280x1024
* failed to add service - already in use?

xingyun86, 2022-04-06:
Please treat it in heart!!!

pelwell, 2022-04-06:
The Issues sections of our GitHub repos are for reporting bugs, not asking for help - the Raspberry Pi Forums (https://forums.raspberrypi.com/) are the right place for that.

There is an argument that the old demo applications that rely on the firmware-based graphics and camera stacks should be deleted from our repos, apt packages and images, because they no longer run on images are they are shipped and because they are not supported.

closed 2022-04-06 by pelwell