raspberrypi/firmware #1718: libfdt cannot apply miniuart-bt.dtbo

jimpo, 2022-04-08:
The `miniuart-bt.dtbo` overlay is invalid according to `libfdt`, the reference implementation used by linux and u-boot. When applying the overlay to `bcm2711-rpi-4-b.dtb` it errors with `FDT_ERR_NOTFOUND`.

**To reproduce**
```bash
% fdtoverlay -v -i bcm2711-rpi-4-b.dtb -o bcm2711-rpi-4-b-with-overlay.dtb miniuart-bt.dtbo                                                                                                                                                                                                                                                                                                       [INS][16:43:02]
input  = bcm2711-rpi-4-b.dtb
output = bcm2711-rpi-4-b-with-overlay.dtb
overlay[0] = miniuart-bt.dtbo

Failed to apply 'miniuart-bt.dtbo': FDT_ERR_NOTFOUND
```

**Expected behaviour**
The DTB overlay should be applied.

**Actual behaviour**
The DTB overlay cannot be applied.

**System**

* Which model of Raspberry Pi? e.g. Pi4B
* Which OS and version (`cat /etc/rpi-issue`)? [dtc](https://git.kernel.org/pub/scm/utils/dtc/dtc.git) version master as of 2022-04-08

**Additional context**

This may very well be a bug with dtc. Even if the error is valid, I find that it is possible to compile invalid overlays from reasonable-looking DTS syntax. A minimal example to reproduce is the following overlay.

```
/dts-v1/;
/plugin/;

/ {
	fragment@0 {
		target = <&minibt>;
		minibt_frag: __overlay__ {
		};
	};
};
```

pelwell, 2022-04-08:
You need miniuart-bt-pi4.dtbo.

jimpo, 2022-04-09:
OK, I still have so many questions. First, there is no `miniuart-bt-pi4.dtbo` in this repository under `boot/overlays`.

But, moreover, this issue applies to other overlays as well. In the official `2022-01-28-raspios-bullseye-arm64.img` Raspberry Pi OS image, the config.txt contains

```
# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
```

so you would expect this overlay to be well-formed. However, if I use `fdtoverlay` to merge that overlay in, I get the same error

```
% fdtoverlay -v -i bcm2711-rpi-4-b.dtb -o bcm2711-rpi-4-b-with-overlay.dtb vc4-kms-v3d.dtbo
input  = bcm2711-rpi-4-b.dtb
output = bcm2711-rpi-4-b-with-overlay.dtb
overlay[0] = vc4-kms-v3d.dtbo

Failed to apply 'vc4-kms-v3d.dtbo': FDT_ERR_NOTFOUND
```

and the same thing happens if I use the `vc4-kms-v3d-pi4.dtbo` overlay

```
% fdtoverlay -v -i bcm2711-rpi-4-b.dtb -o bcm2711-rpi-4-b-with-overlay.dtb vc4-kms-v3d-pi4.dtbo                                                                                                                                                                                                                                                                                                   [INS][19:59:07]
input  = bcm2711-rpi-4-b.dtb
output = bcm2711-rpi-4-b-with-overlay.dtb
overlay[0] = vc4-kms-v3d-pi4.dtbo

Failed to apply 'vc4-kms-v3d-pi4.dtbo': FDT_ERR_NOTFOUND
```

pelwell, 2022-04-11:
That will teach me not to be so quick to close issues on a Friday night.

Even in verbose mode fdtoverlay is unhelpful about what the problem is, but after adding a lot of debug output to a custom build of fdtoverlay I was able to get to the bottom of it.

RPi overlays get some of their flexibility by being allowing fragments to modify other fragments before they are merged. This is not possible using the standard fdtoverlay, which was released _after_ the equivalent RPi utility - `dtmerge`.

Try this instead:
```
$ dtmerge bcm2711-rpi-4-b.dtb bcm2711-rpi-4-b-with-overlay.dtb miniuart-bt.dbo
```
Add the `-d` flag for debug output shows what it is doing:
```
$ dtmerge -d bcm2711-rpi-4-b.dtb bcm2711-rpi-4-b-with-overlay.dtb overlays/miniuart-bt.dtbo
DTOVERLAY[debug]: using platform 'bcm2711'
DTOVERLAY[debug]: overlay map loaded
DTOVERLAY[debug]: /aliases:i2c=i2c0
DTOVERLAY[debug]: /__symbols__:i2c=i2c0
DTOVERLAY[debug]: /__overrides__:i2c=i2c0
DTOVERLAY[debug]: /aliases:i2c_arm=i2c0
DTOVERLAY[debug]: /__symbols__:i2c_arm=i2c0
DTOVERLAY[debug]: /__overrides__:i2c_arm=i2c0
DTOVERLAY[debug]: /aliases:i2c_vc=i2c1
DTOVERLAY[debug]: /__symbols__:i2c_vc=i2c1
DTOVERLAY[debug]: /__overrides__:i2c_vc=i2c1
DTOVERLAY[debug]: /__overrides__:i2c_baudrate=i2c0_baudrate
DTOVERLAY[debug]: /__overrides__:i2c_arm_baudrate=i2c0_baudrate
DTOVERLAY[debug]: /__overrides__:i2c_vc_baudrate=i2c1_baudrate
DTOVERLAY[debug]: overlay = '/miniuart-bt.dtbo', file = 'overlays/miniuart-bt.dtbo'
DTOVERLAY[debug]: merge_fragment(/soc/serial@7e201000,/fragment@0/__overlay__)
DTOVERLAY[debug]:   +prop(pinctrl-names)
DTOVERLAY[debug]:   +prop(pinctrl-0)
DTOVERLAY[debug]:   +prop(status)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: merge_fragment(/soc/serial@7e201000/bluetooth,/fragment@1/__overlay__)
DTOVERLAY[debug]:   +prop(status)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: merge_fragment(/soc/serial@7e215040,/fragment@2/__overlay__)
DTOVERLAY[debug]:   +prop(pinctrl-names)
DTOVERLAY[debug]:   +prop(pinctrl-0)
DTOVERLAY[debug]:   +prop(status)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: merge_fragment(/soc/gpio@7e200000/uart0_pins,/fragment@3/__overlay__)
DTOVERLAY[debug]:   +prop(brcm,pins)
DTOVERLAY[debug]:   +prop(brcm,function)
DTOVERLAY[debug]:   +prop(brcm,pull)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: merge_fragment(/soc/gpio@7e200000/uart1_pins,/fragment@4/__overlay__)
DTOVERLAY[debug]:   +prop(brcm,pins)
DTOVERLAY[debug]:   +prop(brcm,function)
DTOVERLAY[debug]:   +prop(brcm,pull)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: merge_fragment(/soc/gpio@7e200000,/fragment@5/__overlay__)
DTOVERLAY[debug]: merge_fragment(/soc/gpio@7e200000/fake_bt_cts,/fragment@5/__overlay__/fake_bt_cts)
DTOVERLAY[debug]:   +prop(brcm,pins)
DTOVERLAY[debug]:   +prop(brcm,function)
DTOVERLAY[debug]:   +prop(phandle)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: merge_fragment(/aliases,/fragment@6/__overlay__)
DTOVERLAY[debug]:   +prop(serial0)
DTOVERLAY[debug]:   +prop(serial1)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: merge_fragment(/soc/serial@7e215040/bluetooth,/fragment@7/__overlay__)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: wrote 52120 bytes to 'bcm2711-rpi-4-b-with-overlay.dtb'
```
Even without the debug output there are helpful error messages:
```
$ dtmerge bcm2711-rpi-4-b.dtb bcm2711-rpi-4-b-with-overlay.dtb overlays/vc4-kms-v3d.dtbo 
DTOVERLAY[error]: can't find symbol 'i2c2'
* Exiting with error code 1
```

jimpo, 2022-04-12:
I appreciate you looking into this! I can confirm that using `dtmerge` does apply the overlay successfully.

However, this still presents an issue when using U-Boot on a pi because U-Boot uses `libfdt` to apply overlays. So if you load an overlay and run `fdt apply ...`, it will error with `FDT_ERR_NOTFOUND`. So the workaround seems to be using `dtmerge` ahead of time and having U-Boot load the merged FDT, but that's not ideal.

jimpo, 2022-04-12:
Oh silly me, I just realized I can pass through the FDT passed to U-Boot instead of reloading it from a file and applying overlays, which pretty much solves the issue as far as I'm concerned.

pelwell, 2022-04-12:
That's the recommended way of using U-boot on a Pi.

closed 2022-04-16 by jimpo

pelwell, 2023-05-08:
Which base fdt are you applying it to? Posting any debug logs you have may save time.

pelwell, 2023-05-08:
Note that if you are running on a Pi 4 (including the Pi 400 and CM4) then asking for `vc4-kms-v3d` will result in `vc4-kms-v3d-pi4.dtbo` being loaded.

sahib, 2023-05-08:
> Which base fdt are you applying it to? Posting any debug logs you have may save time.

Sorry, didn't mention that. It's the ``bcm2710-rpi-cm3.dts`` (this one her of kernel 5.15 [here](https://github.com/raspberrypi/linux/blob/14b35093ca68bf2c81bbc90aace5007142b40b40/arch/arm/boot/dts/bcm2710-rpi-cm3.dts#L1)). I'm using this [vc4 overlay](https://github.com/raspberrypi/linux/blob/14b35093ca68bf2c81bbc90aace5007142b40b40/arch/arm/boot/dts/overlays/vc4-kms-v3d-overlay.dts#L1). Hardware is a CM3.

If I use regular ``fdtoverlay`` it errors out like that:

```
$ fdtoverlay -v -i /tmp/base_dt.dtb -o /tmp/out /tmp/vc4.dtbo
input  = /tmp/base_dt.dtb
output = /tmp/out
overlay[0] = /tmp/vc4.dtbo

Failed to apply '/tmp/vc4.dtbo': FDT_ERR_NOTFOUND
```

``dtmerge`` (freshly compiled) works and prints this:

```
$ ./bin/dtmerge -d /tmp/base_dt.dtb /tmp/out.dtb /tmp/vc4.dtbo
DTOVERLAY[debug]: /aliases:i2c=i2c0
DTOVERLAY[debug]: /__symbols__:i2c=i2c0
DTOVERLAY[debug]: /__overrides__:i2c=i2c0
DTOVERLAY[debug]: /aliases:i2c_arm=i2c0
DTOVERLAY[debug]: /__symbols__:i2c_arm=i2c0
DTOVERLAY[debug]: /__overrides__:i2c_arm=i2c0
DTOVERLAY[debug]: /aliases:i2c_vc=i2c1
DTOVERLAY[debug]: /__symbols__:i2c_vc=i2c1
DTOVERLAY[debug]: /__overrides__:i2c_vc=i2c1
DTOVERLAY[debug]: /__overrides__:i2c_baudrate=i2c0_baudrate
DTOVERLAY[debug]: /__overrides__:i2c_arm_baudrate=i2c0_baudrate
DTOVERLAY[debug]: /__overrides__:i2c_vc_baudrate=i2c1_baudrate
DTOVERLAY[debug]: Found fragment 0 (offset 36)
DTOVERLAY[debug]: merge_fragment(/reserved-memory/linux,cma,/fragment@0/__overlay__)
DTOVERLAY[debug]:   +prop(size)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: Found fragment 1 (offset 504)
DTOVERLAY[debug]: merge_fragment(/soc/i2c@7e805000,/fragment@1/__overlay__)
DTOVERLAY[debug]:   +prop(status)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: Found fragment 2 (offset 580)
DTOVERLAY[debug]: merge_fragment(/soc/fb,/fragment@2/__overlay__)
DTOVERLAY[debug]:   +prop(status)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: Found fragment 3 (offset 660)
DTOVERLAY[debug]: merge_fragment(/soc/pixelvalve@7e206000,/fragment@3/__overlay__)
DTOVERLAY[debug]:   +prop(status)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: Found fragment 4 (offset 736)
DTOVERLAY[debug]: merge_fragment(/soc/pixelvalve@7e207000,/fragment@4/__overlay__)
DTOVERLAY[debug]:   +prop(status)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: Found fragment 5 (offset 812)
DTOVERLAY[debug]: merge_fragment(/soc/pixelvalve@7e807000,/fragment@5/__overlay__)
DTOVERLAY[debug]:   +prop(status)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: Found fragment 6 (offset 888)
DTOVERLAY[debug]: merge_fragment(/soc/hvs@7e400000,/fragment@6/__overlay__)
DTOVERLAY[debug]:   +prop(status)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: Found fragment 7 (offset 964)
DTOVERLAY[debug]: merge_fragment(/soc/hdmi@7e902000,/fragment@7/__overlay__)
DTOVERLAY[debug]:   +prop(status)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: Found fragment 8 (offset 1040)
DTOVERLAY[debug]: merge_fragment(/soc/v3d@7ec00000,/fragment@8/__overlay__)
DTOVERLAY[debug]:   +prop(status)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: Found fragment 9 (offset 1116)
DTOVERLAY[debug]: merge_fragment(/soc/gpu,/fragment@9/__overlay__)
DTOVERLAY[debug]:   +prop(status)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: Found fragment 10 (offset 1192)
DTOVERLAY[debug]: merge_fragment(/soc/cprman@7e101000,/fragment@10/__overlay__)
DTOVERLAY[debug]:   +prop(claim-clocks)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: Found fragment 11 (offset 1276)
DTOVERLAY[debug]: fragment 11 disabled
DTOVERLAY[debug]: Found fragment 12 (offset 1352)
DTOVERLAY[debug]: merge_fragment(/soc/txp@7e004000,/fragment@12/__overlay__)
DTOVERLAY[debug]:   +prop(status)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: Found fragment 13 (offset 1428)
DTOVERLAY[debug]: fragment 13 disabled
DTOVERLAY[debug]: Found fragment 14 (offset 1496)
DTOVERLAY[debug]: merge_fragment(/soc/mailbox@7e00b840/bcm2835_audio,/fragment@14/__overlay__)
DTOVERLAY[debug]:   +prop(brcm,disable-hdmi)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: Wrote 30630 bytes to '/tmp/out.dtb'
```

I added some debug logs to ``fdtoverlay`` and ``libfdt`` that probably only make sense to me. The call stack however is with the command above roughly this:

1. [ftd_overlay_apply()](https://github.com/qemu/dtc/blob/2cdf93a6d402a161edf16de6011bd5ad76382e92/libfdt/fdt_overlay.c#LL815C5-L815C22) calls ``overlay_symbol_update()``.
2. [fdt_overlay_target_offset](https://github.com/qemu/dtc/blob/2cdf93a6d402a161edf16de6011bd5ad76382e92/libfdt/fdt_overlay.c#L769) in ``overlay_symbol_update()`` fails (fragment name is ``fragment@3/__overlay__``).
3. [fdt_overlay_target_offset()](https://github.com/qemu/dtc/blob/2cdf93a6d402a161edf16de6011bd5ad76382e92/libfdt/fdt_overlay.c#L64) fails in ``fdt_node_offset_by_phandle()`` due to non-existing phandle (I don't enough yet to understand why that is happening)

sahib, 2023-05-08:
I might have worked around this by pre-applying the overlay on the base dt with ``dtmerge`` and using that as base tree. ``vc4`` driver is loaded and does not complain, although I don't have display output yet (likely another issue though). That's pretty hacky though, so if there's another way with regular overlays I'd happy to go it.

pelwell, 2023-05-08:
After a brief look I think the problem is caused by `frag0`, which is a label on a fragment but not on the payload of the fragment - something libfdt doesn't expect or understand. It should not be difficult to get libfdt to skip over it when creating a list of labels to add to the base dtb, but it would mean using a custom libfdt or upstreaming the patch.

pelwell, 2023-05-09:
I take it back - libfdt _does_ understand how to deal with labels on fragment@ nodes.  What I think you are seeing is the effect of a mismatched overlay and kernel.  Download this cm3 dtb file (https://github.com/raspberrypi/rpi-firmware/blob/f5c4fc199c8d8423cb427e509563737d1ac21f3c/bcm2710-rpi-cm3.dtb) and this vc4-kms-v3d overlay (https://github.com/raspberrypi/rpi-firmware/blob/f5c4fc199c8d8423cb427e509563737d1ac21f3c/overlays/vc4-kms-v3d.dtbo) and try again.

sahib, 2023-05-09:
Hmm, that still gets me a ``FDT_ERR_NOTFOUND``. Maybe a version mismatch? I have ``Version: DTC 1.6.1``   (but ``1.7.0``  the same result).                                                                                                                                             

For now I have a workaround I can live with: I removed ``fragment@0`` completely as it concerns itself itself with cma settings only. Since I set that on the kernel commandline anyways it should not be necessary. Without this fragment, it applies fine.

pelwell, 2023-05-09:
Gah - tried it again and it's now borked. It looks like the confusion was at my end.

It now looks like it fails when trying to find the linux,cma node in order to overwrite the size, even though it has previously found it. It's as if the fdt has become corrupted.

pelwell, 2023-05-09:
OK, I understand what's going wrong.

1. Labels in FDT are stored as phandles - unique integer values. The node with the label gets a `phandle` property, and references to it (`&cma`) are replaced with the value - 0x53 in this case. Anything trying to find the thing at the other end of that reference has to search the tree for a node with a `phandle` property with that value.
2. `fragment@0` contains a payload (the contents of the `__overlay__` subnode) that gets written over the node with the `cma` label (`linux,cma`, which we know has the `phandle` value 0x53). But, because the `__overlay__` has a label (`frag0`), it also has a phandle value (and property) of 1, because it is the first and only label in the overlay.
3. Before the overlay can be applied, its phandles have to be renumbered so they don't clash with any in the base DTB. In this case, that 1 becomes 0x8b.
4. When that fragment is _applied_ to the `linux,cma` node, the `phandle` in `__overlay__` is also copied, which has the effect of changing the apparent phandle value of `linux,cma` to 0x8b - 0x35 is no longer a valid phandle value.
5. The fdt overlay application process then tries to merge the relevant labels from the overlay into the base dtb, and decides that it needs to make `frag0` another label on the `linux,cma` node, but when doing so discovers that phandle 0x35 no longer exists, leading to the error you have seen.

Our dtoverlay library, as used by dtmerge, was written before the overlay support was added to libfdt. It doesn't suffer from this problem for two reasons:
i. Certain properties (`name`and `phandle`) are filtered out and not applied (although `phandles` contained in subnodes of the `__overlay__` _are_ applied).
ii. Only labels that are explicitly exported get merged with the base dtb.

sahib, 2023-05-10:
@pelwell Thanks for finding that out :+1: 

Is that something one could patch in libfdt easily? Or should rather the vc4 overlay be changed? 
In my case I can't really patch libfdt/u-boot since I also need to be able to boot with the new overlay on an old u-boot (which cannot be updated easily). If there's something I can help with regarding upstreaming a fix, let me know.

pelwell, 2023-05-10:
We rely on the ability to patch the content of fragments from the parameters/overrides, so without changing the way that mechanism works there is no way to modify the vc4 overlay to remove the label in the general case. However, if you are prepared to do a custom build of the vc4 overlay then ovmerge (https://github.com/raspberrypi/utils/blob/master/ovmerge/ovmerge) can do the conversion for you:
```
# Note the trailing comma, which means apply the (no) parameters and remove any local labels
$ ovmerge vc4-kms-v3d-overlay.dts, > my_vc4_overlay.dts
```
Then compile my_vc4_overlay.dts in the usual way (but note that you'll have to run it through cpp first to convert the clock names to numbers, or do it by hand).

pelwell, 2023-05-10:
I consider this to be a bug/limitation in libfdt - the fact it explicitly caters for putting labels on `__overlay__` (added in 2019) but then doesn't cope with the fallout just shows how little this feature is used in general. The change shouldn't be too difficult, one just needs to add the filtering on the property name, including a way to distinguish a recursive call to `overlay_apply_node` (I used a depth count, but an is-top-level flag would also work.

sahib, 2023-05-11:
Alright, I'll add it to my TODO list and see if I can cook up a patch. If not I'll bring it up at https://github.com/dgibson/dtc.

pelwell, 2023-05-11:
Here's how the dtoverlay library does it: https://github.com/raspberrypi/utils/blob/master/dtmerge/dtoverlay.c#L580

pelwell, 2023-05-12:
Be careful - libfdt is a shared library. Try with STATIC_BUILD=1 (after a clean) to be sure you are testing the new code.