raspberrypi/firmware #1720: RPi3b+ boards are detected as RPi3b when using os_prefix option

alfonsosanchezbeato, 2022-04-11:
**Describe the bug**
When using the os_prefix option, RPi3b+ boards are detected as RPi3b. 

**To reproduce**
Copy dtbs, kernel, initrd, and cmdline.txt to a folder in the boot partition, and set `os_config=<dir>` in config.txt.

**Expected behaviour**
An RPI3b+ should be detected:
```
$ cat /proc/device-tree/compatible
raspberrypi,3-model-b-plus\0brcm,bcm2837\0
```
(this is the behaviour when os_prefix is not used)
**Actual behaviour**
An RPI3b is detected:
```
$ cat /proc/device-tree/compatible
raspberrypi,3-model-b\0brcm,bcm2837\0
```
One of the side effects that is produced is that the ethernet MAC is set to a randon number on each boot.

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi3B+
* Which OS and version (`cat /etc/rpi-issue`)?
Ubuntu 22.04
* Which firmware version (`vcgencmd version`)?
pi-5.15.y (2022-03-16)
* Which kernel version (`uname -a`)?
5.15.0

**Logs**
NA

**Additional context**
NA

pelwell, 2022-04-11:
Please post the content of config.txt? You can omit blank lines and comments.

alfonsosanchezbeato, 2022-04-12:
This would be the config.txt:
```
[all]
kernel=kernel.img
cmdline=cmdline.txt
initramfs initrd.img followkernel
os_prefix=/systems/20220411/kernel/

[pi4]
max_framebuffers=2
arm_boost=1

[all]
dtparam=audio=on
dtparam=i2c_arm=on
dtparam=spi=on

[cm4]
dtoverlay=dwc2,dr_mode=host

[all]
dtoverlay=vc4-fkms-v3d,cma-128

arm_64bit=1
dtoverlay=dwc2
enable_uart=1
```

juergh, 2022-04-12:
Not sure if this is related but overlay_map.dtb and vc4-fkms-v3d are not loaded if os_prefix is set. But dwc2 is.

And the DTB for 3b is loaded instead of 3b+?? I guess that would explain the MAC issue...

Without os_prefix:

```
MESS:00:00:04.772705:0: brfs: File read: /mfs/sd/initrd.img
MESS:00:00:04.776583:0: Loading 'initrd.img' to 0x0 size 0x119bee7
MESS:00:00:04.803253:0: initramfs loaded to 0x2de64000 (size 0x119bee7)
MESS:00:00:04.812604:0: dtb_file 'bcm2710-rpi-3-b-plus.dtb'
MESS:00:00:04.816487:0: brfs: File read: 18464487 bytes
MESS:00:00:04.829552:0: brfs: File read: /mfs/sd/bcm2710-rpi-3-b-plus.dtb
MESS:00:00:04.834642:0: Loading 'bcm2710-rpi-3-b-plus.dtb' to 0x100 size 0x7c42
MESS:00:00:04.856150:0: brfs: File read: 31810 bytes
MESS:00:00:04.871718:0: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
MESS:00:00:04.959349:0: brfs: File read: 1783 bytes
MESS:00:00:04.965703:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:04.969694:0: dtparam: audio=on
MESS:00:00:04.984547:0: dtparam: i2c_arm=on
MESS:00:00:04.996004:0: dtparam: spi=on
MESS:00:00:05.006739:0: brfs: File read: 414 bytes
MESS:00:00:05.016282:0: brfs: File read: /mfs/sd/overlays/vc4-fkms-v3d.dtbo
MESS:00:00:05.042239:0: Loaded overlay 'vc4-fkms-v3d'
MESS:00:00:05.045591:0: dtparam: cma-128=true
MESS:00:00:05.087963:0: brfs: File read: 1446 bytes
MESS:00:00:05.096529:0: brfs: File read: /mfs/sd/overlays/dwc2.dtbo
MESS:00:00:05.108476:0: Loaded overlay 'dwc2'
```

With os_prefix:

```
MESS:00:00:04.802471:0: brfs: File read: /mfs/sd/piboot/ubuntu/pi-kernel_447.snap/initrd.img
MESS:00:00:04.809219:0: Loading '/piboot/ubuntu/pi-kernel_447.snap/initrd.img' to 0x0 size 0x119bee7
MESS:00:00:04.838832:0: initramfs loaded to 0x2de64000 (size 0x119bee7)
MESS:00:00:04.848180:0: dtb_file 'bcm2710-rpi-3-b-plus.dtb'
MESS:00:00:04.852073:0: brfs: File read: 18464487 bytes
MESS:00:00:04.861950:0: dtb_file 'bcm2710-rpi-3-b.dtb'
MESS:00:00:04.871646:0: brfs: File read: /mfs/sd/piboot/ubuntu/pi-kernel_447.snap/bcm2710-rpi-3-b.dtb
MESS:00:00:04.879180:0: Loading '/piboot/ubuntu/pi-kernel_447.snap/bcm2710-rpi-3-b.dtb' to 0x100 size 0x79df
MESS:00:00:04.902948:0: brfs: File read: 31199 bytes
MESS:00:00:05.006741:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.010738:0: dtparam: audio=on
MESS:00:00:05.025268:0: dtparam: i2c_arm=on
MESS:00:00:05.036669:0: dtparam: spi=on
MESS:00:00:05.046726:0: brfs: File read: 413 bytes
MESS:00:00:05.065435:0: Failed to load overlay 'vc4-fkms-v3d'
MESS:00:00:05.074673:0: brfs: File read: /mfs/sd/piboot/ubuntu/pi-kernel_447.snap/overlays/dwc2.dtbo
MESS:00:00:05.089478:0: Loaded overlay 'dwc2'
```

pelwell, 2022-04-12:
With `os_prefix=foo/` (and `os_prefix=/foo/`) it's working for me:
```
Raspberry Pi Bootcode
Read File: config.txt, 2121
Read File: start.elf, 2966048 (bytes)
Read File: fixup.dat, 7221 (bytes)
MESS:00:00:01.450876:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.456133:0: brfs: File read: 2121 bytes
MESS:00:00:01.551765:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:02.407803:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:02.416190:0: *** Restart logging
MESS:00:00:02.418681:0: brfs: File read: 2121 bytes
MESS:00:00:02.459392:0: HDMI0: hdmi_pixel_encoding: 162000000
MESS:00:00:02.483199:0: dtb_file 'bcm2710-rpi-3-b-plus.dtb'
MESS:00:00:02.494207:0: brfs: File read: /mfs/sd/foo/bcm2710-rpi-3-b-plus.dtb
MESS:00:00:02.499653:0: Loading '/foo/bcm2710-rpi-3-b-plus.dtb' to 0x100 size 0x7c2e
MESS:00:00:02.521824:0: brfs: File read: 31790 bytes
MESS:00:00:02.536119:0: brfs: File read: /mfs/sd/foo/overlays/overlay_map.dtb
MESS:00:00:02.624321:0: brfs: File read: 1783 bytes
MESS:00:00:02.632774:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:02.637266:0: dtparam: audio=on
MESS:00:00:02.652310:0: brfs: File read: 2121 bytes
MESS:00:00:02.670759:0: brfs: File read: /mfs/sd/foo/overlays/vc4-kms-v3d.dtbo
MESS:00:00:02.725628:0: Loaded overlay 'vc4-kms-v3d'
MESS:00:00:02.839741:0: brfs: File read: 2703 bytes
MESS:00:00:02.847211:0: brfs: File read: /mfs/sd/foo/cmdline.txt
MESS:00:00:02.851558:0: Read command line from file '/foo/cmdline.txt':
MESS:00:00:02.857857:0: 'console=serial0,115200 console=tty1 root=PARTUUID=ae7b131e-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles'
MESS:00:00:02.891406:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:03.349380:0: brfs: File read: 148 bytes
MESS:00:00:03.790304:0: brfs: File read: /mfs/sd/foo/kernel7.img
MESS:00:00:03.794596:0: Loading '/foo/kernel7.img' to 0x8000 size 0x64ae60
MESS:00:00:03.801199:0: Device tree loaded to 0x2eff7d00 (size 0x82c6)
MESS:00:00:03.808647:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:03.815130:0: uart: Baud rate change done...
MESS:00:00:03.818544:0: uart: Baud rate
Raspbian GNU/Linux 11 raspberrypi ttyS0

raspberrypi login: pi
Password:
Linux raspberrypi 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue Apr 12 11:43:09 BST 2022 on tty1
pi@raspberrypi:~$ tail /proc/cpuinfo
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : 2a020d3
Serial          : 00000000eb4751ef
Model           : Raspberry Pi 3 Model B Plus Rev 1.3
pi@raspberrypi:~$
```
What do you have in your os_prefix folder? I created mine with:
```
$ cd /boot/
$ sudo mkdir -p foo/overlays
$ sudo cp *.img *.dtb cmdline.txt foo
$ sudo cp overlays/* foo/overlays/
```

juergh, 2022-04-12:
`foo` works but `piboot/ubuntu/pi-kernel_999.snap` doesn't. Is there a size limit on `os_prefix`? Or a hierarchical limitation? If I change `foo` to a 40-char name it falls over.

juergh, 2022-04-13:
> It's looking as though there is an overall path limit of 63 bytes
> including the /mfs/sd/ prefix) somewhere, which is shorter than I
> expected. The longest dtb filename is 24 characters, which currently would
> give a maximum os_prefix length of 30 characters. However, the longest
> overlay name is 37 characters, plus 9 characters for overlays/, which
> brings the maximum safe os_prefix directory name length down to a mere 8
> characters.
>
> This restriction is completely unexpected - I thought the limit for prefix
> + file path (without /mfs/sd/) would be over 80 characters, leading to a
> maximum os_prefix of 40 characters, but the bottleneck appears to be lower
> down the file handling stack.

So this is treated as a pi firmware bug then?

pelwell, 2022-04-13:
It's somewhere between a bug and a limitation. I hope the limit can be increased, but there will always be a limit.

What would you consider an acceptable maximum path length?

pelwell, 2022-04-13:
Would you be happy with a maximum `os_prefix` length (excluding the trailing `/`) of 64?:
```
001418.401: brfs: File read: /mfs/sd/config.txt
001423.817: brfs: File read: 2243 bytes
001528.106: brfs: File read: /mfs/sd/config.txt
002392.671: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
002400.298: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
002407.389: *** Restart logging
002410.017: brfs: File read: 2243 bytes
002450.889: HDMI0: hdmi_pixel_encoding: 162000000
002480.057: dtb_file 'bcm2710-rpi-3-b-plus.dtb'
002491.181: dtb_file 'bcm2710-rpi-3-b.dtb'
002502.935: brfs: File read: /mfs/sd/0123456789012345678901234567890123456789012345678901234567890123/bcm2710-rpi-3-b.dtb
002513.247: Loading '0123456789012345678901234567890123456789012345678901234567890123/bcm2710-rpi-3-b.dtb' to 0x100 size 0x79cb
002549.162: brfs: File read: 31179 bytes
002568.470: brfs: File read: /mfs/sd/0123456789012345678901234567890123456789012345678901234567890123/overlays/overlay_map.dtb
002700.593: brfs: File read: 1783 bytes
002709.126: brfs: File read: /mfs/sd/config.txt
002713.633: dtparam: audio=on
002744.989: brfs: File read: 2243 bytes
002754.515: brfs: File read: /mfs/sd/0123456789012345678901234567890123456789012345678901234567890123/cmdline.txt
002764.168: Read command line from file '0123456789012345678901234567890123456789012345678901234567890123/cmdline.txt':
002775.670: 'console=serial0,115200 console=tty1 root=PARTUUID=ae7b131e-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles'
003343.709: brfs: File read: 148 bytes
003805.451: brfs: File read: /mfs/sd/0123456789012345678901234567890123456789012345678901234567890123/kernel7.img
003815.049: Loading '0123456789012345678901234567890123456789012345678901234567890123/kernel7.img' to 0x8000 size 0x64ae60
003826.859: Device tree loaded to 0x2eff8000 (size 0x7f9e)
...
```

alfonsosanchezbeato, 2022-04-13:
To me, 64 for max for os_prefix sounds reasonable, yes.

pelwell, 2022-04-13:
Thanks. A patch implementing that change has been merged to our internal software repo, and will be in future firmware builds.

juergh, 2022-04-13:
```
002480.057: dtb_file 'bcm2710-rpi-3-b-plus.dtb'
002491.181: dtb_file 'bcm2710-rpi-3-b.dtb'
002502.935: brfs: File read: /mfs/sd/0123456789012345678901234567890123456789012345678901234567890123/bcm2710-rpi-3-b.dtb
002513.247: Loading '0123456789012345678901234567890123456789012345678901234567890123/bcm2710-rpi-3-b.dtb' to 0x100 size 0x79cb
```
Is this a 3b or 3b+? And curious, how do you account for potentially growing dtb filenames? Can the FW spit out some error if the resulting pathname is too long?

pelwell, 2022-04-13:
Ah - that was the output from an intermediate version. A build with the final patch says:
```
001419.659: brfs: File read: /mfs/sd/config.txt
001425.074: brfs: File read: 2243 bytes
001527.408: brfs: File read: /mfs/sd/config.txt
002392.007: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
002399.631: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
002406.721: *** Restart logging
002409.352: brfs: File read: 2243 bytes
002450.354: HDMI0: hdmi_pixel_encoding: 162000000
002479.581: dtb_file 'bcm2710-rpi-3-b-plus.dtb'
002494.141: brfs: File read: /mfs/sd/0123456789012345678901234567890123456789012345678901234567890123/bcm2710-rpi-3-b-plus.dtb
002504.892: Loading '0123456789012345678901234567890123456789012345678901234567890123/bcm2710-rpi-3-b-plus.dtb' to 0x100 size 0x7c2e
002541.653: brfs: File read: 31790 bytes
002560.996: brfs: File read: /mfs/sd/0123456789012345678901234567890123456789012345678901234567890123/overlays/overlay_map.dtb
002695.569: brfs: File read: 1783 bytes
002704.099: brfs: File read: /mfs/sd/config.txt
002708.607: dtparam: audio=on
002740.680: brfs: File read: 2243 bytes
002750.312: brfs: File read: /mfs/sd/0123456789012345678901234567890123456789012345678901234567890123/cmdline.txt
002759.965: Read command line from file '0123456789012345678901234567890123456789012345678901234567890123/cmdline.txt':
002771.466: 'console=serial0,115200 console=tty1 root=PARTUUID=ae7b131e-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles'
003351.952: brfs: File read: 148 bytes
003813.461: brfs: File read: /mfs/sd/0123456789012345678901234567890123456789012345678901234567890123/kernel7.img
003823.059: Loading '0123456789012345678901234567890123456789012345678901234567890123/kernel7.img' to 0x8000 size 0x64ae60
003834.869: Device tree loaded to 0x2eff7d00 (size 0x8201)
...
```
> how do you account for potentially growing dtb filenames?

The internal limit has been raised by 64 bytes, all of which are notionally given to os_prefix. This will leave enough headroom for an overlay name to be longer than the current champion by 7-8 bytes.

> Can the FW spit out some error if the resulting pathname is too long?

That would be a more invasive change than I would like.

popcornmix, 2022-04-13:
Updated firmware with increased `os_prefix` length is available from rpi-update

alfonsosanchezbeato, 2022-04-18:
I have checked that with latest firmware in master (https://github.com/raspberrypi/firmware/commit/4fe2e4af02058f88f991469349a335a25a2c719d) things work as expected and my RPi3b+ is detected properly while having the relatively long `os_prefix=/piboot/ubuntu/pi-kernel_x1.snap/`. Thanks a lot!

closed 2022-04-18 by pelwell