raspberrypi/firmware #1721: Double buffering

JeanValjean2, 2022-04-14:
Hi everyone,

First, sorry if this message doesn't belong here, I know that but out of desperation I cannot think of any other place...

I'd like to implement double buffering on a Pi (be it a Pi Zero, Pi 3 etc) in a bare metal context and I cannot do it.

I'm playing with my Pis to learn ARM assembly but the visuals are not as nice as they could be because of the single buffering scheme I currently use.

I've done some memcpy tests to fake a double buffer but it takes a bit of time each frame (not ideal) and it seems stupid : there must be an other way and the VC4 surely has "instructions" to do it.

Is there a way to implement this double buffering using the mailboxes of the VC4 ? I think it's obvious but I cannot do it by myself.

Again, I'm sorry to "pollute" this space, and if someone could direct me to another list/forum/repo I'd be so thankful !

Cheers.

6by9, 2022-04-14:
You don't say which mailbox calls you're currently using - presumably the same as used by [bcm2708_fb](https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/video/fbdev/bcm2708_fb.c), in which case there is no option to double buffer.

If you use the mailbox calls used by [vc4_firmware_kms](https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/gpu/drm/vc4/vc4_firmware_kms.c) then you can double buffer as the allocations are controlled by the kernel (or your code if bare metal).

Quelconque, 2022-04-19:
@JeanValjean2 
Tu peux jeter un coup d'oeil vers:
http://raspberrycompote.blogspot.com/2014/03/low-level-graphics-on-raspberry-pi-part_14.html 
http://raspberrycompote.blogspot.com/2014/03/low-level-graphics-on-raspberry-pi-part_16.html

kumaashi, 2022-04-20:
@JeanValjean2 

I used to look for similar topics as a hobby. The following issues will help you if you writing from baremetal asm or something.

https://github.com/raspberrypi/firmware/issues/67

Also above IRQ can able to receive user irq vector from GPU side.
I recently implemented it. I think you should refer to it as follows if you need.
https://github.com/kumaashi/RaspberryPI/tree/master/RPIZEROW/Sample_Flip_00

I using mailbox message of SET_VIRTUAL_OFFSET when buffer flip.

I hope this information helps.

JeanValjean2, 2022-04-22:
Thanks a lot to the three of you, I think I will have a very prolific week-end.

I didn't expect this much info !

And it shows that I'm not alone in this "hobby" :)

closed 2022-04-22 by JeanValjean2