raspberrypi/firmware #1723: Pi Zero 2 W with BCM43430/1 wifi issues

killah78, 2022-04-21:
Hi, 
I bought a Raspi Zero 2 W and downloaded the Octopi Image via the Image loader. 
I have strange errors with the wifi. First it works, then lot of error, finally freezing and wifi is gone. :-(

I tried a lot with switching firmware files. I found out that there are at least 2 revisions with different wifi chip. I have the one with BCM43430/1. 
The firmware file brcmfmac43436s-sdio.txt is been read within standard configuration. 
I also tried to remove the 43436s, then the 43430 is loaded. But this isn't working better. 

Instead of the octopi image I also tried the standard raspbian lite image, but same issue. 

So the question is: How can I run the wifi on Zero 2 W with BCM43430/1 wifi chip? What I am doing wrong?

Thanks for any hint. 

see attached dmeg.
[dmesg.txt](https://github.com/raspberrypi/firmware/files/8534336/dmesg.txt)

pelwell, 2022-04-29:
Those errors suggest you have a hardware problem - possibly undervoltage. Try adding `over_voltage=2` to `config.txt` and rebooting. Also make sure you are using a quality power supply with a good cable.

pelwell, 2022-04-29:
It seems that the voltage calibration that happens at boot time on a Zero 2 is a bit marginal on a small number of devices - this isn't the first time that a small over_voltage has fixed an issue.

popcornmix, 2022-04-29:
@killah78 Can you test with `over_voltage=1` and see if that is sufficient?

killah78, 2022-05-03:
I just tested with over_voltage=1 and it freezed again while downloading a file. With over_voltage=2 all works fine.

pelwell, 2022-05-16:
@killah78 We'd like to check the manufacturing logs for your board to help us locate it on the spectrum of silicon variability. Would you mind posting your serial number (from `tail /proc/cpuinfo`) and a photo of the QR code on the board? There should be no security implications of doing that, but you can email them to me (`phil@raspberrypi.com`) if you prefer.

Thanks in advance.

killah78, 2022-05-22:
Hi, 
update: over_voltage=2 does not solve all issues. It is much better, but with a big download (i tried several parallel downloads of the octopi image) the wifi still freezes. I now created a 5 min cronjob that checks the kernellog and do a reload of the brcmfmac kernelmodule. 

cpuinfo:
`
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : 902120
Serial          : 00000000976914e4
Model           : Raspberry Pi Zero 2 W Rev 1.0
`
See attached image.

[image]

pelwell, 2022-05-23:
Thanks - I've passed that on, in the hope that we spot some common factor in the affected boards.

popcornmix, 2022-05-24:
The latest rpi-update firmware increases the default voltage on Pi02.
Can anyone affected update and remove any over_voltage setting and confirm system is still reliable?

The over_voltage setting shouldn't be required going forward (and if left in, will result in a higher than required setting).

killah78, 2022-05-24:
After rpi-update and reboot in idle:
May 24 19:27:03 octopi kernel: [  429.891974] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
May 24 21:00:04 octopi kernel: [ 6010.891750] brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -110
May 24 21:00:04 octopi kernel: [ 6010.891786] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
May 24 21:00:04 octopi kernel: [ 6010.916202] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
May 24 21:26:19 octopi kernel: [ 7586.020570] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame

So I can not confirm the problem is solved with that. I now re set to over_voltage = 2. This seems better. It does not solve the issue, but only throws the error messages on large downloads. If there is anything else I can test, just let me know.

pelwell, 2022-06-09:
@killah78 We would like to get hold of your Zero 2 W for testing, if at all possible. We will make it worth your while, but I will understand if you would rather avoid the disruption.

Email me - phil@raspberrypi.com - if interested; if not, no apology needed.

tuna83, 2022-07-28:
Has there been a fix with this?  I have the same problem on my Zero 2W running pihole.  The latest update didn't seem to help much. 
With the over_voltage set to 2, I can usually run a few days before the issue hits and kills my wifi.

I should note that I have 1 small difference.  My error trace shows -81 instead of -110.

Jul 27 12:48:30 raspberrypi kernel: [275690.723848] brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -84
Jul 27 12:48:30 raspberrypi kernel: [275690.723869] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
Jul 27 12:48:30 raspberrypi kernel: [275690.724736] brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -84
Jul 27 12:48:30 raspberrypi kernel: [275690.724754] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
Jul 27 12:48:30 raspberrypi kernel: [275690.725573] brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -84
Jul 27 12:48:30 raspberrypi kernel: [275690.725588] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK

luedi0, 2023-11-15:
@pelwell
Hi Phil,
I seem to have a similar issue with my Pi Zero 2 W (rev1.0).
The onboard wifi module (wlan0) is not recognised at all with a stock os image (lite 64bit).
I have run a full update and forced a firmware update as well as over voltage 1 & 2 without success.
An external usb wifi / ethernet adapter was recognised and worked.

Any further test/ environment recommended?

Best,
Max

pelwell, 2023-11-15:
Has it ever worked? If not, it sounds like a hardware fault.

What does `dmesg  | grep mmc1` report? You may need to reboot if there's a lot of logging output since you last booted.

luedi0, 2023-11-15:
Hi Phil,
It has never worked. I had this pi zero in a sealed antistatic bag until yesterday and then wanted to use it for a IoT thing.
I carefully soldered a header to it. First use and no wlan0 interface available. 

`dmesg  | grep mmc1` does not output anything. dmesg alone outputs a lot though incl. mmc0.

pelwell, 2023-11-15:
That's odd. I'll compare a known good Zero 2 W tomorrow.

pelwell, 2023-11-15:
Do you have any non-standard config.txt settings - overlays, etc.?

pelwell, 2023-11-15:
`dtoverlay=disable-wifi` would be a good one to avoid.

luedi0, 2023-11-15:
I am running a mint 
Linux raspberrypi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
for testing. 
No customisations.

LightBridge16, 2024-12-05:
hello I have also the wifi issue on my zero 2 W

psoloviov, 2025-01-12:
Hello, I have encountered the same issue with my Raspberry Pi Zero 2 W.

I bought the Pi about 2 months ago, and while working with it via SSH, my connection would randomly freeze, and the Pi would become unresponsive.

To troubleshoot, I swapped the SD card with the one from my Raspberry Pi 5 to check if the issue was related to the SD card. Unfortunately, it didn’t solve the problem. Card from Raspberry Pi Zero 2 W works perfectly on Raspberry Pi 5, wifi working.

Next, I bought a mini-HDMI to HDMI adapter to connect the Raspberry Pi Zero 2 W to a display and see what was happening. When the Zero 2 W boots into the system (PiOS Desktop 32-bit november 2024 release), I repeatedly see SDIO errors (-84 and -110). The number of these errors varies with each boot. Here some logs `dmesg | grep mmc`:

```
[    3.154857] mmc-bcm2835 3f300000.mmcnr: mmc_debug:0 mmc_debug2:0
[    3.154882] mmc-bcm2835 3f300000.mmcnr: DMA channel allocated
[    3.236317] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
[    3.269530] mmc0: host does not support reading read-only switch, assuming write-enable
[    3.274382] mmc0: Host Software Queue enabled
[    3.274409] mmc0: new high speed SDXC card at address 5048
[    3.277655] mmcblk0: mmc0:5048 SD64G 58.0 GiB
[    3.280903]  mmcblk0: p1 p2
[    3.281664] mmcblk0: mmc0:5048 SD64G 58.0 GiB
[    3.288506] mmc1: error -110 whilst initialising SDIO card
[    3.530754] mmc1: error -84 whilst initialising SDIO card
[    3.548799] mmc1: Failed to initialize a non-removable card
[    8.436858] EXT4-fs (mmcblk0p2): mounted filesystem 74f968bb-7c40-44cd-8d3e-6963921bf08b ro with ordered data mode. Quota mode: none.
[   11.610615] EXT4-fs (mmcblk0p2): re-mounted 74f968bb-7c40-44cd-8d3e-6963921bf08b r/w. Quota mode: none.
```

I have since discovered several threads discussing similar issues, where the Raspberry Pi Zero 2 W suddenly stops working with WiFi.

Based on one of those threads, I tried adding `over_voltage=2` to `/boot/firmware/config.txt`, but it didn’t resolve the issue.

I don’t have the original power supply, so I’ve tried powering the Zero 2 W with several sources:
- Directly from my PC
- A power bank/power station ( 3A, 5V )
- A mobile charger ( 2A, 5V )

None of these power sources made any difference.
I really don’t understand why my WiFi stopped working. I can’t even figure out what caused it, which makes me concerned that if I buy another Zero 2 W, it might fail in the same way.

Is there any way to fix this issue?

patrikzudel, 2025-01-29:
@killah78 Mind reopening the issue? It seems quite many people are having the same issue (including me). 

Steps to replicate for me:
1. Flashed newest 64bit release of raspbian (2024-11-19)
2. Installed speedtest CLI (Ookla)
3. Run speedtest
4. Freezes at some point during the speedtest

`dmesg | grep mmc` output:
```
[   11.161161] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   11.178210] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   11.179159] usbcore: registered new interface driver brcmfmac
[   11.446107] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[   11.446854] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7
[   17.740014] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
```

For me this happens on OpenWRT as well (with correct drivers according to https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=e9f9cd14cc71826957877999fd063dd080de4751)

patrikzudel, 2025-01-29:
> [@patrikzudel](https://github.com/patrikzudel) I'm not sure, but looks like you have different case. This issue related to Pis where wifi stop working at all (-84 and -110 errors), but works previously.
> 
> In your case, can you try to turn off power management and check if it solves your issue? See many times when issue with wi-fi was related to power management (or bad power supply)

I also got

```
kernel log: 
[  438.865461] brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -110
[  438.873357] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
```

after the freeze on OpenWRT. I am getting a Mini HDMI cable today to better debug this because I was doing it without it, and it was a proper pain in the ass with constantly losing connection. Will comment more if I find out anything.

Thank you for the suggestion though, I will try it!

pelwell, 2025-01-29:
Errors like that from the SDIO level suggest a power problem.
1. Check your power supply and cable.
2. Remove unpowered peripherals that may be consuming more power than can be spared.
3. Add over_voltage=2 to config.txt.