raspberrypi/firmware #1724: QT6 eglfs is not working on raspberry pi 4

PhysicsX, 2022-05-01:
Hello

I cross compiled the qt6.3 for raspberry pi. But I can not run the eglfs platform.
It gives these erros:

[image]

Configuration output is like that 

```
QPA backends:
  DirectFB ............................... no
  EGLFS .................................. yes
  EGLFS details:
    EGLFS OpenWFD ........................ no
    EGLFS i.Mx6 .......................... no
    EGLFS i.Mx6 Wayland .................. no
    EGLFS RCAR ........................... no
    eglfs_egldevice ...................... yes
    eglfs_gbm ............................ yes
    EGLFS VSP2 ........................... no
    EGLFS Mali ........................... no
    EGLFS Raspberry Pi ................... no
    EGLFS X11 ............................ yes
```
EGLFS Raspberry Pi  is "no". It should be on also?
What should I do ? Is eglfs_egldevice not enough ?

**System**
Model           : Raspberry Pi 4 Model B Rev 1.1
Linux raspberrypi 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux

Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 226b479f8d32919c9fe36dd5b4c20c02682f8180, stage4

Thank you very much.

carlonluca, 2022-07-15:
EGLFS Raspberry Pi is based on dispmanx, you need the old libbrcm libs for that. Also, I think I remember it is still not ported to cmake, I had to build it manually. KMS eglfs is probably what you want.

PhysicsX, 2022-07-16:
Thanks for the answer.

KMS eglfs may be okay for me.

What should I do to enable kms eglfs ? Should I compile again the Qt?

Thank you.

carlonluca, 2022-07-16:
You already did, eglfs GBM is enabled. For the rest not sure sorry, I saw those errors in quite a few cases.

popcornmix, 2022-07-18:
This question would be better directed at eglfs/QT6 developers.

gbm generally means running fullscreen as the drm master.
This means it won't work from an X desktop session.
You can either boot to console and run it, or switch away from X VT (e.g. ctrl-alt-F2).

But I don't use eglfs (just using knowledge of gbm apps like kodi), hence the suggestion to ask eglfs/QT6 devs.

closed 2024-01-06 by timg236