raspberrypi/firmware #1725: Looking for help: how to build from source

fluffynuts, 2022-05-05:
Good day

I'd really like to build a customised version of the kernel presented in this repo. Are there build instructions and upstream sources that I can get started with? In particular, I'd like to build in support for older CIFS versions (1.0) to interface with a media player that still uses this.

Kernel 5.15.32-v8+ is built without support for /proc/config.gz, otherwise I could have already had a go at this; but if there are official docs for building all the parts required, I'd appreciate a pointer in the right direction.

pelwell, 2022-05-05:
There is documentation on building your own kernel here: https://www.raspberrypi.com/documentation/computers/linux_kernel.html#building

popcornmix, 2022-05-06:
> Kernel 5.15.32-v8+ is built without support for /proc/config.gz, otherwise I could have already had a go at this; 

It is - it's just a module.
```
pi@pi4:~ $ uname -a
Linux pi4 5.15.36-v8+ #1549 SMP PREEMPT Thu Apr 28 19:53:21 BST 2022 aarch64 GNU/Linux
pi@pi4:~ $ sudo modprobe configs
pi@pi4:~ $ zcat /proc/config.gz | grep RPIVID
CONFIG_RPIVID_MEM=m
CONFIG_VIDEO_RPIVID=m
```

fluffynuts, 2022-05-06:
thanks to @pelwell and @popcornmix - I'll follow the links and see where I end up (: