raspberrypi/firmware #1726: latest firmware version breaks boot on Raspberry Pi 3 (and others) with upstream Linux when no HDMI is connected

stapelberg, 2022-05-08:
**Describe the bug**

Commit https://github.com/raspberrypi/firmware/commit/949c8989df4d67dab93400acd67dea0a5addee6a broke booting on my Raspberry Pi 3B, Pi 3B+ and Pi Zero 2 W. It seems to not have affected my Pi 4B.

The last line when booting Linux 5.17.5 (latest upstream, not the Raspberry Pi foundation version) is:
```
[    2.373010] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops)
```

The commit description mentions HDMI clocks, so I think that’s related.

Note that I’m running my Raspberry Pis without an HDMI display connected. As soon as I connect an HDMI display, the device boots again.

popcornmix, 2022-05-09:
I'm trying to reproduce. I have confirmed that using rpi-update (i.e. firmware you indicate with rpi 5.15.38-v7+ kernel) does boot okay on a Pi3+ with hdmi disconnected.

I'll see if I can test with upstream kernel.

popcornmix, 2022-05-09:
I managed to get upstream kernel working, but it booted okay for me without hdmi connected:
```
pi@a:~ $ uname -a
Linux a 5.17.6-multi #606 SMP Mon May 9 19:55:57 BST 2022 armv7l GNU/Linux
```
This is with latest rpi-update firmware.

stapelberg, 2022-05-09:
Weird. Are you using the upstream DTBs, too?

I just double-checked on my Raspberry Pi 3 B+, and I upgraded to 5.17.6, just like you’re using, but I still see the same symptom: with HDMI cable connected, it boots, without HDMI cable connected, it hangs.

You can find the full disk image I’m using at https://t.zekjur.net/2022-05-09-raspberry-pi-boot-hang.img.bz2 (57 MB). You can find my kernel / dtb in the first partition (`losetup -Pf 2022-05-09-raspberry-pi-boot-hang.img`)

popcornmix, 2022-05-10:
Yes I was using upstream dtbs.

Can you try with this patch added?
https://www.spinics.net/lists/arm-kernel/msg975291.html

I think currently firmware and upstream kernel disagree with the number of clocks (which is passed through a mailbox call), so undefined behaviour is possible.

stapelberg, 2022-05-11:
Thanks for the tip!

I applied the following patches:
1. https://github.com/torvalds/linux/commit/12c90f3f27bb3ad0dd3fad1550fec87091aa3329
2. https://github.com/torvalds/linux/commit/542acfec4e313c001f9b82332f4fa2848ec7bf58
3. https://github.com/torvalds/linux/commit/e9d6cea2af1cf8d84287ff2287b6cd776f7475d2
4. https://www.spinics.net/lists/arm-kernel/msg975291.html

And that indeed makes the new firmware version work.

Is there any chance you can fix this on the firmware side until the above changes are in a released Linux kernel version?

stapelberg, 2022-05-12:
Thanks for the details. I’ll carry the kernel patches for now.

popcornmix, 2022-05-17:
There is a [firmware update](https://drive.google.com/uc?id=1SR90gsUxuJJUA44_MmeKZn_zy62E3DEw&export=download) that I believe resolves this issue. Are you able to test it seems okay for you?

stapelberg, 2022-05-18:
Can confirm: with the current kernel and firmware, the issue is still reproducible. With the firmware update you linked to, the issue is fixed.

Thanks for resolving this!

popcornmix, 2022-05-19:
The test firmware is now in rpi-update repo.

colemickens, 2022-05-19:
@popcornmix Is there a tag that corresponds to "test firmware" "in rpi-update"?

A branch would be super helpful as it would just work with my update/testing automation.

pelwell, 2022-05-19:
The default branch ("master") is essentially a "test" branch - only the "stable" branch is likely to have had any significant usage.

timg236, 2022-11-28:
It sounds like the issue was resolved. Closing as fixed.

closed 2022-11-28 by timg236