raspberrypi/firmware #1728: /opt/vc/src/hello_pi fails to build

jonshouse1, 2022-06-15:
Installed the latest "pi OS" whatever that is.

Did an rpi-update
Tried to build examples in /opt/vc/src

```
make -C hello_font
make[1]: Entering directory '/opt/vc/src/hello_pi/hello_font'
cc -DSTANDALONE -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -g -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -ftree-vectorize -pipe -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -I./ -I/opt/vc/src/hello_pi/libs/ilclient -I/opt/vc/src/hello_pi/libs/vgfont -I/opt/vc/src/hello_pi/libs/revision -g -c main.c -o main.o -Wno-deprecated-declarations
cc -o hello_font.bin -Wl,--whole-archive main.o -lvgfont -lfreetype -lz -lrevision -L/opt/vc/lib/ -lbrcmGLESv2 -lbrcmEGL -lopenmaxil -lbcm_host -lvcos -lvchiq_arm -lpthread -lrt -lm -L/opt/vc/src/hello_pi/libs/ilclient -L/opt/vc/src/hello_pi/libs/vgfont -L/opt/vc/src/hello_pi/libs/revision -Wl,--no-whole-archive -rdynamic
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glxx_buffer_info_set'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glBufferSubData'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glxx_client_GetFramebufferAttachmentParameteriv'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glxx_client_state_free'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glxx_client_DeleteRenderbuffers'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glxx_set_error'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glxx_client_GenFramebuffers'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glxx_set_error_api'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glxx_client_BindRenderbuffer'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glxx_client_FramebufferTexture2D'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glxx_client_GenRenderbuffers'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glxx_client_DeleteFramebuffers'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glxx_client_GetRenderbufferParameteriv'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glxx_client_RenderbufferStorage'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glxx_client_IsFramebuffer'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glintAttribPointer'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glxx_client_GenerateMipmap'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glPointSizePointerOES'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `gl20_client_state_init'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `gl11_client_state_init'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glxx_client_FramebufferRenderbuffer'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glxx_client_IsRenderbuffer'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glxx_buffer_info_get'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glxx_client_BindFramebuffer'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glxx_client_CheckFramebufferStatus'
/usr/bin/ld: /opt/vc/lib//libbrcmEGL.so: undefined reference to `glDiscardFramebufferEXT'
collect2: error: ld returned 1 exit status
make[1]: *** [../Makefile.include:19: hello_font.bin] Error 1
rm main.o
make[1]: Leaving directory '/opt/vc/src/hello_pi/hello_font'
make: *** [Makefile:18: apps] Error 2
root@raspberrypi:/opt/vc/src/hello_pi# uname -a
Linux raspberrypi 5.15.45-v7l+ #1562 SMP Thu Jun 9 13:57:42 BST 2022 armv7l GNU/Linux

```

popcornmix, 2022-06-16:
Code in rpi-update and userland repo is updated. Does that build for you?

Note: much of the hello_pi code was written for legacy interfaces (openmax/mmal/dispmanx/openvg),
and on bullseye we are moving to standard linux apis (v4l2/kms/drm), so while these should build,
they may not run on bullseye running kms.

jonshouse1, 2022-06-16:
Thanks, I can now confirm they build.  They don't work ... but they do build !

> legacy interfaces (openmax/mmal/dispmanx/openvg)
The main job I was doing was to try and build omxplayer (from popcorn github). When that did not work I tried to build the demos to see if they where still functional.
My customer has many devices based on Pi3/Compute modules in the field doing media playback. It is all very well to wave a hand and say "legacy", but I can't find equivalent functionality with any other software. I want accelerated video playback with audio routing options without any desktop environment.
So far i've wasted a couple of days trying to build and rum omxplayer  on a Pi 4 using the latest Pi OS, with a struggle I can build it but its just exists with a error code and no comments :-(  I feel either the Pi foundation or Broadcom have screwed me a bit at this point.

JamesH65, 2022-06-16:
You can use the Legacy Raspberry Pi OS edition (Buster) which still supports the older mechanisms by default. Installable via the Raspberry Pi Imager.   Its also possible to down configure Bullseye to use them, but that a bit more complex.

jonshouse1, 2022-06-16:
>You can use the Legacy Raspberry Pi OS edition (Buster) which still supports the older mechanisms by default

Thanks, I will give that a go.

closed 2022-06-16 by jonshouse1