raspberrypi/firmware #1730: raspberry pi 4

khangTi, 2022-07-11:
i am using 1 raspberry pi 4 model b, but when i install android 10 it gives me error attached,
Is there any way to fix this error? 
"start4x.elf: is not compatible
This board requires newer software
Get the latest software from https://www.raspberrypi.com/software/"

JamesH65, 2022-07-11:
We don't support Android here, try our forums, which have a third party Android section.

closed 2022-07-11 by JamesH65

timg236, 2022-07-11:
As James says Android isn’t supported but you can copy the *.elf and *dat files from this repo on to your Android repo. You might want to ping the Android maintainers and ask them to update the build.

khangTi, 2022-07-12:
thanks so much, I will copy 2 files the *.elf and *dat to test run

khangTi, 2022-07-12:
yeah I copied 4 files fixup4.dat, fixup4x.dat, start4x.elf, start4.self replaces the boot files and is already working, thank you very much