raspberrypi/firmware #1732: [Pi4] Video is Static during boot only, when booting from HDMI 2 only, at 1080x1920 (NOT 1920x1080)

ChaosBlades, 2022-07-18:
**Describe the bug**
When booting the Pi 4 only on the HDMI 2 port connected to a 1080x1920 monitor (NOT 1920x1080) the entire boot process is scrambled like static on a TV or it is all miss aligned depending on what is on screen.

It does not happen when using the HDMI 1 port nor does it happen on a 1920x1080 monitor.

Once boot has finished and the OS Desktop loads the video returns to normal.

**To reproduce**
Connect the P4 HDMI-2 port to a 1080x1920 monitor and watch the boot process. The entire boot process will be scrambled.

I am using this specific monitor but I suspect it will occure on any 9:16 aspect ratio monitor. Perhaps even any monitor that is taller then wide.
https://www.waveshare.com/5.5inch-hdmi-amoled.htm

**Expected behaviour**
The boot process to not be scrambled so I can set a boot logo for a clean looking boot process.

**Actual behaviour**
Scrambled boot with static all over the monitor.

**System**
https://gist.github.com/ChaosBlades/6dd079a0ba039552bb97fd9d8269eec3

**Logs**
https://gist.github.com/ChaosBlades/110e423604d00ed2cfd0f1651517a968

**Additional context**
I understand this is a VERY specific bug but I have limited space in my enclosure so only using HDMI 2 port is optimal.

I don't have issues on the Pi 3 (only has one HDMI port).
Does not matter how I power this specific monitor. Either from the Pi or its own source.
Does not matter what HDMI cable I use.
Does not matter if it is a 32-bit OS or 64-bit
Does not matter if the OS is headless or with Desktop Environment.

During first boot the blue screen that comes up when resizing the SD Card is scrambled but on a headless install the screen that wants you to enter your username is NOT scrambled as well as the end of the boot process right before that screen.

JamesH65, 2022-07-18:
Does enabling 4K with the following in config.txt make any difference? I expect not but worth a try.

hdmi_enable_4kp60=1

popcornmix, 2022-07-18:
This looks suspicious:
```
hdmi_group:0=2
hdmi_mode:0=87
hdmi_timings:0=1080 1 26 4 50 1920 1 8 2 6 0 0 0 60 0 135580000 3
hdmi_timings:1=
```
so you a custom hdmi mode set for hdmi0, but not hdmi1.
But you are using kms driver which won't use the custom mode.

What if behaviour if you remove all `hdmi_*` lines from config.txt?

ChaosBlades, 2022-07-18:
I added the following per the WaveShare Wiki: https://www.waveshare.com/wiki/5.5inch_HDMI_AMOLED
That is why those settings are present.
```
max_framebuffer_height=1920
config_hdmi_boost=10
hdmi_group=2
hdmi_force_hotplug=1
hdmi_mode=87
hdmi_timings=1080 1 26 4 50 1920 1 8 2 6 0 0 0 60 0 135580000 3
```
So I am guessing it assumes HDMI 0 instead of both when a specific one not called for.
I changed it to...
```
max_framebuffer_height=1920
config_hdmi_boost=10
hdmi_group:0=2
hdmi_group:1=2
hdmi_force_hotplug=1
hdmi_mode:0=87
hdmi_mode:1=87
hdmi_timings:0=1080 1 26 4 50 1920 1 8 2 6 0 0 0 60 0 135580000 3
hdmi_timings:1=1080 1 26 4 50 1920 1 8 2 6 0 0 0 60 0 135580000 3
```
Now HDMI 2 does not even display anything during boot.
I also figured out that the reason it works after it finishes booting is because I was using the following
```
dtoverlay=vc4-kms-v3d
max_framebuffers=2
```
Once KMS is loaded then the display looks normal.
If I comment this out then the display remains static even after boot.
If I apply the hdmi setting above to both HDMI ports then the display remains black just on HDMI 2.

I also updated my config.txt to `display_lcd_rotate` since `display_rotate` is depreciated. This did not change things.

`hdmi_enable_4kp60=1` does not change anything.

If I uncomment all the WaveShare HDMI settings then both HDMI ports are static even after boot with KMS disabled.

So am I just setting HDMI 2 settings wrong above?

popcornmix, 2022-07-18:
> If I uncomment all the WaveShare HDMI settings then both HDMI ports are static even after boot with KMS disabled.

I want you to test a default configuration. So kms enabled and no `hdmi_*` settings.

ChaosBlades, 2022-07-18:
Already did that. Static on both HDMI ports until KMS engages then a clean image. It appears the current issue is why the `hdmi_*` settings are not working on HDMI 2.

DillonC-Spec, 2023-09-23:
Has anyone found a fix to this?  I have the same problem with the Waveshare 5.5" on my Pi 4.

ChaosBlades, 2023-09-23:
You need to change the HDMI timings that waveshare provides on the wiki.

In my testing this issue is resolved without HDMI timings by replacing the HDMI 180 degree connection with one that is properly shielded.

closed 2023-09-23 by ChaosBlades