raspberrypi/firmware #1733: system does not boot from specified partition, when using GPT

shaxi1, 2022-07-28:
**Describe the bug**
No matter the value of "boot_partition" inside the autoboot.txt, system always finally boots from the default partition. (anyother_part.txt)
(uboot_fdt.txt)
Value "PARTITION=" inside config.txt has no effect at all, from my testing autoboot.txt seems to be an only viable option to choose boot partition.

When setting boot_partition=2 the system does not boot at all. AFAIK that behaviour only occurs with "boot_partition" set to 2 (Attached file: bootpart2.txt)

**Additional context**
To my knowledge problem seems relevant when using GPT. 
The problem doesn't seem to occur when using MBR with one exception - when partition type is different than "e". 

**Expected behaviour**
System boots from the partition specified inside autoboot.txt, and properly sets property "partition" in the fdt, inside /chosen/bootloader to the number of partition it was booted from.

**System**
 * Tested on RPI4 Model B
 * 5.10.95-arm-rpi4-v7l
 * newest firmware from this repo

**Logs**

[bootpart2.txt](https://github.com/raspberrypi/firmware/files/9206794/bootpart2.txt)
[anyother_part.txt](https://github.com/raspberrypi/firmware/files/9206889/anyother_bootpart.txt)
[uboot_fdt.txt](https://github.com/raspberrypi/firmware/files/9206799/uboot_fdt.txt)

timg236, 2022-07-28:
That's correct. The firmware does not support specifying the partition number within GPT table. It just looks for the first "FAT / EFI" partition within the GPT

srcshelton, 2023-05-02:
> That's correct. The firmware does not support specifying the partition number within GPT table. It just looks for the first "FAT / EFI" partition within the GPT

What's being looked-for, though?  A specified GPT UUID (I think `EBD0A0A2-B9E5-4433-87C0-68B6B72699C7` is `0700`/`Microsoft basic data`) or some magic signature bytes known to exist at the start of any FAT32 partition?

If they differ, is the physical ordering on disk or the logical numbering in the partition table given priority?

abaumgaertner, 2023-11-09:
Booting from a 1TB USB SSD with GPT works fine. maybe consider a similar partition layout.

```
root@dc1:~# sgdisk -p /dev/sda
Disk /dev/sda: 1953525168 sectors, 931.5 GiB
Model:
Sector size (logical/physical): 512/512 bytes
Disk identifier (GUID): C7EC176E-D8CD-4041-9C5D-111111111111
Partition table holds up to 128 entries
Main partition table begins at sector 2 and ends at sector 33
First usable sector is 34, last usable sector is 1953525134
Partitions will be aligned on 2048-sector boundaries
Total free space is 1076589 sectors (525.7 MiB)

Number  Start (sector)    End (sector)  Size       Code  Name
   1            2048        16779263   8.0 GiB     0700  boot
   2        16779264        20973567   2.0 GiB     8200  swap
   3        20973568      1952450559   921.0 GiB   BF00  rpool
root@dc1:~#
```

srcshelton, 2023-11-09:
> Booting from a 1TB USB SSD with GPT works fine. maybe consider a similar partition layout.

I'm afraid that you've missed the point - this is concerning the situation where you have _multiple_ `boot` partitions and wish to use the firmware configuration files to select which one to boot from (to allow for A/B-switching for firmware updates or rollback, for example).

onorua, 2025-01-05:
I confirm this, autoboot is useless with GPT partition. It is always boots from the first one. It looks like a bug of autoboot mechanism in rpi bootloader to me.

timg236, 2025-01-15:
Closing as duplicated of https://github.com/raspberrypi/rpi-eeprom/issues/654

closed 2025-01-15 by timg236

timg236, 2025-03-06:
The updated start4.elf should now be available via APT

Get:1 http://archive.raspberrypi.com/debian bookworm/main arm64 raspi-firmware all 1:1.20250305-1 [12.8 MB]