raspberrypi/firmware #1734: PoE+ Hat fan not spinning unless manually setting the overlay.

matty0005, 2022-07-30:
**Describe the bug**
I'm finding that the fan on the PoE+ hat is not turning on. I'm using a Pi 4 4GB and netbooting it.

Nothing pops up in dmesg regarding "fan" or "poe"
```
pi@pi4gb :~ $ dmesg | grep poe
pi@pi4gb :~ $ dmesg | grep fan
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
```

```
pi@pi4gb:~ $ dtoverlay -l
No overlays loaded
```
But I have the following in the config.txt file in /boot
```
dtoverlay=rpi-poe-plus

dtparam=poe_fan_temp0=45000,poe_fan_temp0_hyst=3000
dtparam=poe_fan_temp1=50000,poe_fan_temp1_hyst=3000
dtparam=poe_fan_temp2=62000,poe_fan_temp2_hyst=4000
dtparam=poe_fan_temp3=70000,poe_fan_temp3_hyst=5000
```
Other settings in the config.txt file are applied, such as `arm_freq`.

The fan only starts spinning after I run the following command
```
pi@pi4gb:~ $ sudo dtoverlay -v rpi-poe-plus
DTOVERLAY[debug]: using platform 'bcm2711'
DTOVERLAY[debug]: overlay map loaded
run_cmd: which dtoverlay-pre >/dev/null 2>&1 && dtoverlay-pre
DTOVERLAY[debug]: loading file '/boot/overlays/rpi-poe-plus.dtbo'
DTOVERLAY[debug]: fragment 5 disabled
DTOVERLAY[debug]: fragment 6 disabled
DTOVERLAY[debug]: merge_fragment(/fragment@4/__overlay__/poe@51,/fragment@11/__overlay__)
DTOVERLAY[debug]: merge_fragment(/fragment@4/__overlay__/poe@51/rpi-poe-power-supply@f2,/fragment@11/__overlay__/rpi-poe-power-supply@f2)
DTOVERLAY[debug]:   +prop(compatible)
DTOVERLAY[debug]:   +prop(reg)
DTOVERLAY[debug]:   +prop(status)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: wrote 4337 bytes to '/tmp/.dtoverlays/0_rpi-poe-plus.dtbo'
DTOVERLAY[debug]: wrote 4337 bytes to '/sys/kernel/config/device-tree/overlays/0_rpi-poe-plus/dtbo'
run_cmd: which dtoverlay-post >/dev/null 2>&1 && dtoverlay-post
```

After this, if i then run `dtoverlay -l` i get
```
pi@pi4gb:~ $ dtoverlay -l
Overlays (in load order):
free(): invalid pointer
Aborted
```

Finally the following is now present in dmesg:
```
pi@pi4gb:~ $ dmesg | grep "poe"
[ 1127.444519] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__overrides__/poe_fan_temp0
[ 1127.444545] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__overrides__/poe_fan_temp0_hyst
[ 1127.444556] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__overrides__/poe_fan_temp1
[ 1127.444566] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__overrides__/poe_fan_temp1_hyst
[ 1127.444575] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__overrides__/poe_fan_temp2
[ 1127.444582] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__overrides__/poe_fan_temp2_hyst
[ 1127.444589] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__overrides__/poe_fan_temp3
[ 1127.444596] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__overrides__/poe_fan_temp3_hyst
[ 1127.444603] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /__overrides__/poe_fan_i2c
```
I have also tried what was suggested in #1610 but that didn't change anything.

I have also ran `sudo rpi-update` to update to the latest kernal, but nothing changed with regards to the fan. With all logs being the same, minus the `dtoverlay -l` command which outputs
```
pi@pi4gb:~ $ dtoverlay -l
Overlays (in load order):
DTOVERLAY[error]: failed to open '/tmp/.dtoverlays/0_rpi-poe-plus.dtbo'
Segmentation fault
```

**To reproduce**
Not sure if this is PXE related, but use a Pi4 4GB and netboot it, and find the pi doesn't have its fan on.

**Expected behaviour**
The fan turn on when the CPU gets hot

**Actual behaviour**
The fan does not turn on at all even if temperatures rise to above 84C.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? Pi 4B 4GB
* Which OS and version (`cat /etc/rpi-issue`)?
```
pi@pi4gb:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 27a8050c3c06e567c794620394a8c2d74262a516, stage2
```
* Which firmware version (`vcgencmd version`)?
```
pi@pi4gb:~ $ vcgencmd version
Mar 24 2022 13:19:26
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
```
Linux pi4gb  5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
```

pelwell, 2022-07-30:
It should not be necessary to load the overlay explicitly. What happens if you remove the dtoverlay line from config.txt? You can keep the dtparam lines, which should continue to work as long as they appear before any other dtoverlay line.

matty0005, 2022-07-30:
Unfortunately the fan still doesn't spin when I remove the dtoverlay line in the config.txt file and reboot it - same exact behaviour.

pelwell, 2022-08-11:
Testing with a fresh POE+ HAT and a fresh image it just works - no configuration is required. If you have a spare SD card, install a fresh image and boot from the card to see if that makes a difference.

pelwell, 2022-08-11:
P.S. Full disclosure - this is testing with a regular USB-C power supply and without PoE (not a recommended set-up), because that's what I have.

matty0005, 2022-08-12:
Ok, so I flashed a brand new SD card with the latest version of PiOS lite (64bit) and the fan does turn on (even when powered through PoE).

I also inserted the SD card I used to originally configure the pi before netbooting. It too seemed to work. (This also didn't have the dtoverlay configuration in config.txt)

So it seems to be related to netbooting.

If I run the following commands (also executed above) for comparison, I get
```
pi@pi4gb:~ $ dmesg | grep fan
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
pi@pi4gb:~ $ dmesg | grep poe
pi@pi4gb:~ $ dtoverlay -l
No overlays loaded
pi@pi4gb:~ $ sudo dtoverlay -v rpi-poe-plus
sudo: unable to resolve host pi4gb: Name or service not known
DTOVERLAY[debug]: using platform 'bcm2711'
DTOVERLAY[debug]: overlay map loaded
run_cmd: which dtoverlay-pre >/dev/null 2>&1 && dtoverlay-pre
DTOVERLAY[debug]: loading file '/boot/overlays/rpi-poe-plus.dtbo'
DTOVERLAY[debug]: fragment 5 disabled
DTOVERLAY[debug]: fragment 6 disabled
DTOVERLAY[debug]: merge_fragment(/fragment@4/__overlay__/poe@51,/fragment@11/__overlay__)
DTOVERLAY[debug]: merge_fragment(/fragment@4/__overlay__/poe@51/rpi-poe-power-supply@f2,/fragment@11/__overlay__/rpi-poe-power-supply@f2)
DTOVERLAY[debug]:   +prop(compatible)
DTOVERLAY[debug]:   +prop(reg)
DTOVERLAY[debug]:   +prop(status)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: wrote 4337 bytes to '/tmp/.dtoverlays/0_rpi-poe-plus.dtbo'
DTOVERLAY[debug]: wrote 4337 bytes to '/sys/kernel/config/device-tree/overlays/0_rpi-poe-plus/dtbo'
* Failed to apply overlay '0_rpi-poe-plus' (kernel)
run_cmd: which dtoverlay-post >/dev/null 2>&1 && dtoverlay-post
pi@pi4gb:~ $ dmesg | grep poe
pi@pi4gb:~ $ dmesg | grep fan
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
pi@pi4gb:~ $ dtoverlay -l
No overlays loaded****
```

What I find interesting is that theres no 
```
free(): invalid pointer
Aborted
```
Output on the second `dtoverlay -l`
In addition I noticed that on the SD card, It seems to have failed to apply the overlay
```
* Failed to apply overlay '0_rpi-poe-plus' (kernel)
```

When I get a bit more time, I'll do a fresh netboot setup to make sure it's nothing on my end in that regard.

GNewbury1, 2022-11-02:
@matty0005 Did you get anywhere with your testing? I've just flashed RPi OS Lite 64-bit on one of my RPi 4s, and I'm running in to the exact same issue you've mentioned above. 

Interestingly, the `type` given `/sys/class/thermal/cooling_device0/type` by reads:
```
$ cat /sys/class/thermal/cooling_device0/type
pwm-fan
```

Shouldn't this be `rpi-poe-fan`?

pelwell, 2022-11-02:
The PoE fan support makes use of the standard `pwm-fan` driver (which knows how to control a fan using any PWM) with a custom PWM driver that knows how to control a specific PWM via the firmware (this is the Linux way - separating out the different layers of functionality to maximise reusability).

pelwell, 2022-11-02:
By the way, it's working for me:
```
pi@raspberrypi:~$ uname -a
Linux raspberrypi 5.15.73-v8+ #1593 SMP PREEMPT Wed Oct 12 14:24:59 BST 2022 aarch64 GNU/Linux
pi@raspberrypi:~$ dtoverlay -l
Overlays (in load order):
0:  rpi-poe-plus
```

GNewbury1, 2022-11-02:
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.15.76-v8+ #1596 SMP PREEMPT Mon Oct 31 17:15:15 GMT 2022 aarch64 GNU/Linux
pi@raspberrypi:~ $ dtoverlay -l
No overlays loaded
```

```
~ $ sudo dtoverlay -v rpi-poe-plus
DTOVERLAY[debug]: using platform 'bcm2711'
DTOVERLAY[debug]: overlay map loaded
run_cmd: which dtoverlay-pre >/dev/null 2>&1 && dtoverlay-pre
DTOVERLAY[debug]: loading file '/boot/overlays/rpi-poe-plus.dtbo'
DTOVERLAY[debug]: fragment 5 disabled
DTOVERLAY[debug]: fragment 6 disabled
DTOVERLAY[debug]: merge_fragment(/fragment@4/__overlay__/poe@51,/fragment@11/__overlay__)
DTOVERLAY[debug]: merge_fragment(/fragment@4/__overlay__/poe@51/rpi-poe-power-supply@f2,/fragment@11/__overlay__/rpi-poe-power-supply@f2)
DTOVERLAY[debug]:   +prop(compatible)
DTOVERLAY[debug]:   +prop(reg)
DTOVERLAY[debug]:   +prop(status)
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: merge_fragment() end
DTOVERLAY[debug]: wrote 4337 bytes to '/tmp/.dtoverlays/0_rpi-poe-plus.dtbo'
DTOVERLAY[debug]: wrote 4337 bytes to '/sys/kernel/config/device-tree/overlays/0_rpi-poe-plus/dtbo'
* Failed to apply overlay '0_rpi-poe-plus' (kernel)
run_cmd: which dtoverlay-post >/dev/null 2>&1 && dtoverlay-post
```

pelwell, 2022-11-02:
Hang on - if `/sys/class/thermal/cooling_device0/type` contains `pwm-fan` then the overlay has been applied (which is should be, automatically, by the firmware). If it is (and you can check by looking for the existence of `/proc/device-tree/pwm-fan`) then the kernel will refuse to apply it again at runtime because it is illegal to overwrite an existing node.

N.B. `dtoverlay -l` only lists overlays that have been applied at runtime using the `dtoverlay` command. Overlays applied using the config.txt `dtoverlay=` directive become part of the base dtb seen by the kernel, and therefore not visible as overlays.

GNewbury1, 2022-11-02:
Right, so are we expecting `/sys/class/thermal/cooling_device0/type` to contain `pwm-fan` instead of `rpi-poe-fan` with newer versions of Debian/RPi OS/Ubuntu running the `5.15` kernel? Does this mean the `dtparams` for controlling the fan speed need to be updated?

pelwell, 2022-11-02:
> Right, so are we expecting /sys/class/thermal/cooling_device0/type to contain pwm-fan instead of rpi-poe-fan with newer versions of Debian/RPi OS/Ubuntu running the 5.15 kernel?

Yes, but it's only linked to the kernel version - 5.15 has the implementation.

> Does this mean the dtparams for controlling the fan speed need to be updated?

No - we kept the parameters the same, only adding `i2c` to bypass the firmware and control the fan directly from i2c0.

TheNON75, 2024-09-30:
hi @pelwell,

sorry for the newbie question, but can you explain, in the light of the above, how should the config.txt look like to set custom temperature levels? I assume i2c shall be added, but no idea what exactly

thanks in advance

pelwell, 2024-10-01:
Unless you've got very old firmware, the overlay should be loaded automatically. The fan can be configured by a number of overlay parameters:
```
pi@phil-pi5:~$ dtoverlay -a | grep poe
  pineboards-hatdrive-poe-plus
  rpi-poe
  rpi-poe-plus
pi@phil-pi5:~$ dtoverlay -h rpi-poe-plus
Name:   rpi-poe-plus

Info:   Raspberry Pi PoE+ HAT fan

Usage:  dtoverlay=rpi-poe-plus,<param>[=<val>]

Params: poe_fan_temp0           Temperature (in millicelcius) at which the fan
                                turns on (default 40000)
        poe_fan_temp0_hyst      Temperature delta (in millicelcius) at which
                                the fan turns off (default 2000)
        poe_fan_temp1           Temperature (in millicelcius) at which the fan
                                speeds up (default 45000)
        poe_fan_temp1_hyst      Temperature delta (in millicelcius) at which
                                the fan slows down (default 2000)
        poe_fan_temp2           Temperature (in millicelcius) at which the fan
                                speeds up (default 50000)
        poe_fan_temp2_hyst      Temperature delta (in millicelcius) at which
                                the fan slows down (default 2000)
        poe_fan_temp3           Temperature (in millicelcius) at which the fan
                                speeds up (default 55000)
        poe_fan_temp3_hyst      Temperature delta (in millicelcius) at which
                                the fan slows down (default 5000)
        i2c                     Control the fan via Linux I2C drivers instead of
                                the firmware.
```