raspberrypi/firmware #1736: It is no longer possible to reduce the transmission power

kosmosgit, 2022-07-31:
Hello, so far I have reduced the transmission power of my RPi's using

sudo iw wlan0 set txpower limit 1800    # = 18dB

Since the standard setting of 31 dB (1 watt transmission power) is mercilessly too high, it is probably also not legally permissible. In addition, it does not help the RPi in terms of reception and only disturbs other participants

lategoodbye, 2022-09-11:
Please provide more information about the setup (which Raspberry PI?, which wifi device?, which kernel version?, ...)

kosmosgit, 2022-09-11:
Raspberry Pi4, 4GB, First Board Edition, Kernel 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022, included WLAN0

after change with 

sudo iw wlan0 set txpower limit 1800

iw dev, shows the change, because the router interface shows every time the same signal quality. Earlier, a change has always had an effect.

[image]

all my 3 Raspberry Pi 4 shows the same error.

It looks to me that no more changes are allowed, but that's completely wrong to work with the highest possible transmission power. I have about 30 WLAN clients and none has such a high transmission performance as the Raspberrys.

seamusdemora, 2022-09-11:
Just my 2 pence worth, but I wonder: 

1. `mBm` was a brain fart - nobody uses it, and there was an [action to change it to dBm](https://patchwork.kernel.org/project/linux-wireless/patch/1547731764-14037-1-git-send-email-bpothuno@codeaurora.org/). Maybe that's happened? 
2. At the conclusion of `iw help`: 
>Do NOT screenscrape this tool, we don't consider its output stable. 
3. 31 dB***m*** is 1 watt - not 31 dB  
4. 1 Watt of *conducted* power (power delivered to the antenna) is allowed under most regulatory regimes. I'm not quite sure how operating at that limit gets to be "completely wrong", but I think you should be able to reduce power.
5. Other than defining hard power limits, `iw` also provides the `power_save` option. This setting may be verified with:
```
$ sudo iw dev wlan0 get power_save
Power save: on
```
I don't know exactly how `power_save` is implemented, but perhaps there is some sort of feedback loop that influences the Tx power level? Have you tried this?

kosmosgit, 2022-09-13:
Of course I meant dBm, powersave on used to cause problems for me that the Raspberry could only be reached via the network after repeated attempts or sometimes not at all.

rubienr, 2024-01-04:
My two cents:

[...]

>     3. 31 dB_**m**_ is 1 watt - not 31 dB

Simplified explained (excluding EIRP and antenna gain): given a TX power of `31 dB mW`, the `mW` power is `10^(31/10)*mW = 1258,9mW`.

>     4. 1 Watt of _conducted_ power (power delivered to the antenna) is allowed under most regulatory regimes. I'm not quite sure how operating at that limit gets to be "completely wrong", but I think you should be able to reduce power.

This is absolutely not very accurate!

*"[Equivalent isotropically radiated power](https://en.wikipedia.org/wiki/Equivalent_isotropically_radiated_power) (EIRP) in the [European Union](https://en.wikipedia.org/wiki/European_Union) is limited to 20 [dBm](https://en.wikipedia.org/wiki/DBm) (100 mW)."* [1]

[1] https://en.wikipedia.org/wiki/Wi-Fi#Transmitter_power

Apart of this, i did not see many WiFi final stages that dissipate 1W on the 2,4GHz band. Even many of the high end MiKroTik AP wireless cards do not dissipate more than about 23dBm per output stage.

More examples of wrongly reported power:

- Raspberry Pi 4 Model B Rev 1.4 - reports `31dBm`
  - `Linux octoprint 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Raspberry Pi Zero W Rev 1.1 - reports `31dBm`
  - `Linux pihole 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`
- Variscite DART-MX8M-PLUS - reports `30dBm`
  - even if regulatory domain is set manually in the device tree to DE, AT or ETSI, the reported value stays constant: `30dBm`

Example with reasonable result:

- Intel Wi-Fi 6 AX200 - reports `22dBm`
  - `Linux r9 6.5.0-14-generic #14-Ubuntu SMP PREEMPT_DYNAMIC Tue Nov 14 14:59:49 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux`

So far i have seen wrong `iw` reports only on ARM devices.

kosmosgit, 2024-01-05:
For me it wasn't/isn't the deciding factor here whether it was DBm, 31 or 22....it was simply extremely noticeable that my Raspberry had the strongest signal to the router despite the worst antennas and the worst location in the house.

See the picture

seamusdemora, 2024-01-05:
> My two cents:
> 
> [...]
> 
> > 4. 1 Watt of _conducted_ power (power delivered to the antenna) is allowed under most regulatory regimes. I'm not quite sure how operating at that limit gets to be "completely wrong", but I think you should be able to reduce power.
> > ```
> 
> This is absolutely not very accurate! 

For the *technically challenged*, I'll just [quote this](http://www.wiki.robotz.com/index.php/FCC_Regulations_on_WiFi): 
>One Watt (1000mw) is the FCC limit on WiFi devices

> 
> _"[Equivalent isotropically radiated power](https://en.wikipedia.org/wiki/Equivalent_isotropically_radiated_power) (EIRP) in the [European Union](https://en.wikipedia.org/wiki/European_Union) is limited to 20 [dBm](https://en.wikipedia.org/wiki/DBm) (100 mW)."_ [1]

Whatever you say... [perhaps you can explain this](https://w.wol.ph/2015/08/28/maximum-wifi-transmission-power-country/) - or not; I really don't care. But you should stop running off at the mouth on things you're apparently not very knowledgeable about. 

Have a good day.

rubienr, 2024-01-06:
> Whatever you say... [perhaps you can explain this](https://w.wol.ph/2015/08/28/maximum-wifi-transmission-power-country/) - or not; I really don't care. But you should stop running off at the mouth on things you're apparently not very knowledgeable about.

Hi,

First of all that is a private page, it could state anything. However this section seems reasonable:

```
Short overview of the ETSI standard:
– 2.4 GHz: 100 mW (20 dBm)
– 5 GHz channel 36 to 64: 200 mW (23 dBm)
– 5 GHz channel 100 to 140: 1000 mW (30 dBm)
– 5 GHz channel 155 to 171: 4000 mW (36 dBm)
```

As you see, on the `2.4GHz` band `20dBm` are allowed (there might be exceptions w.r.t. regulatory domains). 

Notes:
- On the `5GHz` band `1W` is not allowed on all channels (outdoors due to Radars etc.).
- This are maximum allowed `EIRP` values.  Assuming a Wifi equipment operating on `2.4GHz` and an antenna with `3dB` gain , the allowed TX power in the ETSI regulatory domain would be `(20-3)dBm`.

seamusdemora, 2024-01-06:
> > Whatever you say... [perhaps you can explain this](https://w.wol.ph/2015/08/28/maximum-wifi-transmission-power-country/) - or not; I really don't care. But you should stop running off at the mouth on things you're apparently not very knowledgeable about.
> 
> Hi,
> 
> First of all that is a private page, it could state anything. However this section seems reasonable:
> 
> ```
> Short overview of the ETSI standard:
> – 2.4 GHz: 100 mW (20 dBm)
> – 5 GHz channel 36 to 64: 200 mW (23 dBm)
> – 5 GHz channel 100 to 140: 1000 mW (30 dBm)
> – 5 GHz channel 155 to 171: 4000 mW (36 dBm)
> ```
> 
> As you see, on the `2.4GHz` band `20dBm` are allowed (there might be exceptions w.r.t. regulatory domains).
> 
> Notes:
> 
>     * On the `5GHz` band `1W` is not allowed on all channels (outdoors due to Radars etc.).
> 
>     * This are maximum allowed `EIRP` values.  Assuming a Wifi equipment operating on `2.4GHz` and an antenna with `3dB` gain , the allowed TX power in the ETSI regulatory domain would be `(20-3)dBm`.
> 
>     * The hydrogen bond has a good absorption at about `2.4GHz` (where microwave ovens operate), hence the regulation limits `EIRP` in this area, to not induce energy into the human body (there are also other RF bands limited due to resonance with human body, i.e. `2m` band bacause wavelength and avg. body height are somewhat similar).

Off into the fog with ya'...  maintain your fantasy as long as you like - it won't bother me. But other than insulting me, and making a fool of yourself, I cannot figure what your post adds to this thread. Comic relief?