raspberrypi/firmware #1740: Turn Official Raspberry Pi 7" Touchscreen Display Off / On

Vito303, 2022-08-18:
**Describe the bug**
Upgrade from 5.15.32-v8+ to 5.15.56-v8+ causes vcgencmd display_power does not work anymore. 

**To reproduce**
`sudo vcgencmd display_power 0`
returns
`display_power=-1`
Does not affect on display.

**Expected behaviour**
`sudo vcgencmd display_power 0`
normal returns
`display_power=0`
The display is then turned off.

**Actual behaviour**
Does not affect on display. Echos 'display_power=-1' in terminal.

**System**

**System Information**

Raspberry Pi 3 Model B Rev 1.2
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 27a8050c3c06e567c794620394a8c2d74262a516, stage4

Linux aftNew 5.15.56-v8+ #1575 SMP PREEMPT Fri Jul 22 20:31:26 BST 2022 aarch64 GNU/Linux
Revision        : a02082
Serial          : 00000000ef3ffb27
Model           : Raspberry Pi 3 Model B Rev 1.2
Throttled flag  : throttled=0x20000
Camera          : supported=0 detected=0, libcamera interfaces=0

**Videocore information**

Aug  9 2022 13:46:06
Copyright (c) 2012 Broadcom
version 273b410636cf8854ca35af91fd738a3d5f8b39b6 (clean) (release) (start)

mickym2, 2022-08-19:
I have the same problem since the last kernel update on my Raspberry Pi4 4Gb. 
`vcgencmd display_power 0`

returns
`display_power=1`

I have not a touchscreen but a normal monitor attached to the HDMI port and switched this off and on with my IoT system and this does no longer work.

Here are my specs:
` uname -a
Linux MWHome2 5.15.56-v8+ #1575 SMP PREEMPT Fri Jul 22 20:31:26 BST 2022 aarch64 GNU/Linux`

mickym2, 2022-08-20:
I restored the functionality with xrandr commands. However I found no way how to get the status, if display is currently on or off.

6by9, 2022-08-20:
vcgencmd issues commands to the firmware.
Bullseye has always used the vc4-kms-v3d driver, where all displays are controlled by the Linux kernel, and the firmware has no involvement. ```vcgencmd display_power``` will therefore do nothing.

Blackbox-git, 2022-08-23:
I have the same issue with my Raspberry Pi 2B on a normal HDMI-Screen.

`sudo vcgencmd display_power 0`

returns
`display_power=1`

`sudo vcgencmd display_power 0 0`

returns
`display_power=0`

But both dosent work for me. I have the issu since my last update.

The Display don´t turn off. My System is:
`Linux DisplayPi 5.15.56-v7+ #1575 SMP Fri Jul 22 20:28:11 BST 2022 armv7l GNU/Linux`

MichaIng, 2022-08-30:
> vc4-kms-v3d driver, where all displays are controlled by the Linux kernel, and the firmware has no involvement

We have this reported as well and this was my first thought as well, but the docs don't state that it works with legacy graphics only and in our case downgrading to one of the last Linux 5.10 commit solved it. Of course some users switch to legacy for fKMS graphics driver. We'll do some tests to verify where it works and if there really something changed/broke with Linux 5.15.

popcornmix, 2022-08-31:
`vcgencmd display_power` is not supported when using kms. Even if it had worked in the past, it's inherently unsafe to allow the firmware and the kernel to access the same hardware without knowledge that the other is also accessing it.

It looks like it stopped working with firmware commit:
`platform: Block "vcgencmd display_power" if vc4-kms-v3d is enabled`

so it was completely intentional. What needs to be done is to work out the correct linux side way of achieving what you want.

If you are booting to desktop then perhaps:
`xset dpms force on`
`xset dpms force off`

popcornmix, 2022-08-31:
See [here](https://forums.raspberrypi.com/viewtopic.php?t=331817) for context

mickym2, 2022-08-31:
If this behavior is intended and the feature or the usage of vcgencmd display_power will never come back, 
I found a workaround - but it is more complicated to implement - and I have not really an understanding - what happens in the background.

fire1ce, 2022-09-11:
+1 following

wuast94, 2022-11-18:
+1 following

rcarmo, 2023-03-16:
Also looking into this. I can't reliably control the official 7" display backlight on or off. Worse, using `vcgencmd display_power 0` now seems to also disable the touchscreen.

6by9, 2023-03-16:
> I can't reliably control the official 7" display backlight on or off

What commands are you using?

I don't have a panel set up here, but backlight should be controllable through nodes under ```/sys/class/backlight```, but I forget the full path used
```sudo ch -c 'echo 0 > /sys/class/backlight/10-0045/rpi_touchscreen_attiny/brightness```
(0 = off, with range up to 255)

```sudo sh -c 'echo 1 > /sys/class/backlight/10-0045/rpi_touchscreen_attiny/bl_power'```
1 for on, 0 for off.

> Worse, using vcgencmd display_power 0 now seems to also disable the touchscreen.

As already stated above, vcgencmd display_power is incompatible with using vc4-kms-v3d and should not be used. IIRC it has already been blocked for HDMI, but may have been missed for DSI. Do not expect it to be coming back.

Antislag, 2023-04-12:
Hi,
I have the same issue. I tried to replace the command sudo vcgencmd display_power 0 with xrandr --output HDMI-1 --off. The display is disabled for about 1-2 sec and then turns on again. Can i adapt this, so the display will stay off?

Thanks in advance

wuast94, 2023-08-05:
is there any dev or progress on this? im using a pi as a smart mirror, and normally i turn the screen off when nobody is there.

its almost a year now without any news or progress or usable workaround

wuast94, 2023-08-05:
Found a working solution:

1. Install ddcutil: `sudo apt install ddcutil`
1.1  if you dont want to need sudo to use ddcutil add pi to the i2c group: `sudo usermod -G i2c -a pi`
3. search if you can change the power state with: `ddcutil getvcp scan | grep DPMS` should output something like:
`VCP code 0xd6 (Power mode                    ): DPM: On,  DPMS: Off (sl=0x01)`
The importand part here is the d6
4: Search for possible states with: `ddcutil vcpinfo d6 --verbose` which ouputs something like:
```
Output level:               Verbose
Reporting DDC data errors:  false
Trace groups active:        none
Traced functions:           none
Traced files:               none
Force I2C slave address:    false
User defined features:      disabled

Performance and Retry Options:
   Deferred sleep enabled:                      false
   Sleep suppression (reduced sleeps) enabled:  false
   Dynamic sleep adjustment enabled:            false

VCP code D6: Power mode
   DPM and DPMS status
   MCCS versions: 2.0, 2.1, 3.0, 2.2
   MCCS specification groups: Control, Miscellaneous
   ddcutil feature subsets:
   Attributes: Read Write, Non-Continuous (simple)
   Simple NC values:
      0x01: DPM: On,  DPMS: Off
      0x02: DPM: Off, DPMS: Standby
      0x03: DPM: Off, DPMS: Suspend
      0x04: DPM: Off, DPMS: Off
      0x05: Write only value to turn off display
```
In my case 0x01 turns the monitor on and 0x05 off
5.  Controll the monitor with `ddcutil setvcp d6 5` or `ddcutil setvcp d6 1`

dont know if needed but i have added `dtoverlay=vc4-kms-v3d` to my /boot/config.txt

lbt, 2023-10-17:
The path in @6by9's comment here https://github.com/raspberrypi/firmware/issues/1740#issuecomment-1471695013 isn't quite correct.

On kernel 6.1.25-v8+ with an official 7" DSI display 

Turn off backlight:
`sudo sh -c 'echo 0 > /sys/class/backlight/10-0045/brightness'`

Turn on backlight:
`sudo sh -c 'echo 255 > /sys/class/backlight/10-0045/brightness'`

Dim backlight:
`sudo sh -c 'echo 10 > /sys/class/backlight/10-0045/brightness'`