raspberrypi/firmware #1741: Revision number changed after update? 5.15.56-v8+

akumpf, 2022-08-19:
I just updated my Pi Zero W 2 running Bullseye to the latest firmware (5.15.56-v8+) and noticed that some code I was using that checked the revision (via `/proc/cpuinfo` or the like) stopped working. 

The board revision was previously showing as `0x00902120` but after the update now shows `0x02902120` -- is that from the firmware update or something else? Could the latest firmware update have done that? Seems unexpected, and not sure how to set it back. I tried reverting to previous firmware versions, but it seems to continue showing the new revision number now.

6by9, 2022-08-20:
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#raspberry-pi-revision-codes

W | Warranty bit2 | 0: Warranty is intact
-- | -- | --
  |   | 1: Warranty has been voided by  [overclocking](https://www.raspberrypi.com/documentation/computers/config_txt.html#overclocking-options)

Any code processing the revision code should always deal with it as a bitmask, rather than assuming absolute values.

akumpf, 2022-08-22:
Good to know! I don't believe I was overclocking anything (I am using `force_turbo=1` and `arm_boost=1` to keep it from reducing clock speed when idle, but not explicitly setting any over voltage or cpu speeds). 

That said, I'll assume it's on my end somewhere. And that bit definitely shouldn't be changing any proper downstream code that keying off of the board revision (i.e. it should mask those bits out for any lookup/comparison). I'll debug the offending library and post an issue over there. Thanks so much for the explanation -- I'll close this out.

akumpf, 2022-08-22:
unlikely related to the firmware update; closed.

closed 2022-08-22 by akumpf