raspberrypi/firmware #1742: dtparam settings for power led no longer function since upgrade to `5.15.56-v7+ #1575`, RPi 3B+

seamusdemora, 2022-08-21:
**Is this the right place for my bug report?**
This repository contains the GPU firmware used on the Raspberry Pi. This software is the closed source part of the Raspberry Pi system, it includes booting (including network booting and USB booting), low-level power and clock control, FKMS and legacy HDMI control (not full KMS that is in the linux kernel), hardware legacy codecs (MPEG2, H264 and VC1), encode hardware including the ISP (image sensor pipeline) and camera control, audio output (analogue and HDMI audio).

If you believe that the issue you are seeing is within this area, this is the right place. If not, we have other repositories for the linux kernel at [github.com/raspberrypi/linux](https://github.com/raspberrypi/linux) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
Prior to a recent upgrade that resulted in a kernel update to `5.15.56-v7+ #1575`, the power led could be extinguished using the following lines in `/boot/config.txt`. Since this upgrade, these `dtparam` settings seem to have no effect: 

```
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off
```

**To reproduce**
List the steps required to reproduce the issue.

Restore the firmware prior to the current version, and the power led stays OFF.

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

Power LED remains OFF after boot

**Actual behaviour**
Add a clear and concise description of what actually happened.

Power LED remains ON after boot

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

System Information
------------------

Raspberry Pi 3 Model B Plus Rev 1.3
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Raspberry Pi reference 2021-10-30
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 288b21fc27e128ea6b330777aca68e0061ebf4fe, stage2

Linux raspberrypi3b 5.15.56-v7+ #1575 SMP Fri Jul 22 20:28:11 BST 2022 armv7l GNU/Linux
Revision	: a020d3
Serial		: 00000000eccd2fff
Model		: Raspberry Pi 3 Model B Plus Rev 1.3
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0, libcamera interfaces=0

Videocore information
---------------------

Aug  9 2022 13:46:06
Copyright (c) 2012 Broadcom
version 273b410636cf8854ca35af91fd738a3d5f8b39b6 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem                  1K-blocks       Used   Available Use% Mounted on
/dev/root                    30395256    1862420    27266484   7% /
[... 610 lines cut ...]
[64983.619239]  loop0: p1 p2
[64983.621540]  loop0: p1 p2
[64992.932453] EXT4-fs (loop0p2): mounted filesystem with ordered data mode. Opts: (null). Quota mode: none.
[65113.066573] loop0: detected capacity change from 0 to 6463488
[65113.139605]  loop0: p1 p2
[324186.198682] loop0: detected capacity change from 0 to 6463488
[324187.328954]  loop0: p1 p2
[324188.733526] EXT4-fs (loop0p2): mounted filesystem with ordered data mode. Opts: (null). Quota mode: none.
[324222.467344] loop0: detected capacity change from 0 to 6463488
[324222.530971]  loop0: p1 p2
[496987.402100] loop0: detected capacity change from 0 to 6463488
[496995.653273]  loop0: p1 p2
[496995.658627]  loop0: p1 p2
[496996.660508] EXT4-fs (loop0p2): mounted filesystem with ordered data mode. Opts: (null). Quota mode: none.
[497012.318769] loop0: detected capacity change from 0 to 6463488
[497012.383517]  loop0: p1 p2
[497012.385550]  loop0: p1 p2

NOTE: Output of `raspinfo` did not return to shell prompt.

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

writepatrick, 2022-08-23:
same with Raspberry Pi 4 Model B.

pelwell, 2022-08-23:
When did it last work?

pelwell, 2022-08-23:
i.e. do you have a kernel version when it used to work?

pelwell, 2022-08-24:
That took a while, but I can now see a change in behaviour with releases since Aug 08 2022. However, the new behaviour is correct and the old behaviour was wrong.

This isn't easy to say succinctly, so I'm going to explain it as a numbered list of statements that you can process in any order you like. Please refer to the facts by number if you dispute them (or simply want more information).

1. The PWR LED on the affected boards is connected to the GPIO expander.
2. The PWR LED is controlled by an active-low signal; driving the GPIO _low_ turns the LED _on_, driving the GPIO _high_ turns the LED _off_.
3. By default, the PWR LED is declared to the kernel (by Device Tree) as being active-low.
4. By default, the PWR LED is assigned (by Device Tree) the "default-on" trigger which puts the GPIO into the active state.
5. If the "none" trigger is used, the kernel decides what to do based on the `default-state` property.
6. On the Pi 4 family, `default-state = "keep"`, which means use the value set by the firmware.
7. On older Pis (3B, 3B+ etc.), `default-state` is not set, which means turn the LED off.

The following statements assume `dtparam=pwr_led_trigger=none` and `dtparam=pwr_led_activelow=off` have been specified.
8. On an older Pi, the kernel will try to turn off the LED, which means driving the GPIO low.
9. On a Pi 4, the kernel sees that the firmware has left the GPIO low (causing it to think the LED is off because it thinks the GPIO is active-high), so it will drive the GPIO low.

Now to complicate matters further:
10. The GPIO expander is actually driven by the firmware, under the direction of the kernel.
11. The kernel has two methods to change a GPIO's state - by making it an output with a specific value, or assuming it is an output and just setting the value.
12. In the case of the GPIO expander, the two methods above result in different messages to he firmware.
13. The "none" trigger just uses the "make it an output..." method, but the "default-on" trigger goes on to also use the "just set the value" method.

And finally:
14. Firmware prior to Aug 08 2022 has a bug in the "make it an output..." method which causes it to ignore the polarity (active-high or active-low),
15. Newer firmware fixes the bug.

Combining 8&9, 14&15, `dtparam=pwr_led_trigger=none` and `dtparam=pwr_led_activelow=off` we get:
* old Pi, old firmware -> kernel thinks GPIO low+LED off, actually GPIO high+LED off
* Pi 4, old firmware -> kernel thinks GPIO low+LED off, actually GPIO high+LED off
* old Pi, new firmware -> kernel thinks GPIO low+LED off, actually GPIO low+LED on
* Pi 4, new firmware -> kernel thinks GPIO low+LED off, actually GPIO low+LED on

The safest way to turn off the PWR LED on all Pis with either old or new firmware is to use the default-on method but declare the output as being active-high:
```
dtparam=pwr_led_trigger=default-on   # The default
dtparam=pwr_led_activelow=off
```
This uses the "set a value" method, overriding the incorrect GPIO level set by the buggy "make it an output..." path in the older firmware.

If you think you have results that contradict these findings then please state:
a. which Pi you are running,
b, the firmware version (`vcgencmd version`), and
c. the relevant config.txt settings.

writepatrick, 2022-08-24:
@pelwell  Thank you for the detailed explanation! Even that I barely understand it. I get the concept.
It seems like it was never intended to be used the way it was by most users (but worked somehow)

I tried it now and „default-on“ works like a charm on my pi4. 
Since it was a „bug“ in the firmware the logic should be now fixed for the next updates.

I am really happy that I found this here and was fixed immediately. Great work!

closed 2022-08-26 by pelwell

seamusdemora, 2022-08-27:
> If you think you have results that contradict these findings then please state:
> a. which Pi you are running,
> b, the firmware version (`vcgencmd version`), and
> c. the relevant config.txt settings. 

```bash
pi@raspberrypi1bp:~ $ cat /proc/device-tree/model && echo
Raspberry Pi Model B Plus Rev 1.2
pi@raspberrypi1bp:~ $ vcgencmd version
Dec  1 2021 15:07:06
Copyright (c) 2012 Broadcom
version 71bd3109023a0c8575585ba87cbb374d2eeb038f (clean) (release) (start)
```
fm /boot/config.txt:

### This does not work (pwr led remains illuminated after boot completes):
>dtparam=act_led_trigger=mmc0  
dtparam=act_led_activelow=off  
dtparam=pwr_led_trigger=default-on   # The default  
dtparam=pwr_led_activelow=off  

### This still works as before (pwr led turns OFF after boot completes): 
>dtparam=act_led_trigger=mmc0
dtparam=act_led_activelow=off
dtparam=pwr_led_trigger=none

pelwell, 2022-08-27:
Here is a complete (or as complete as I'm going to give) list of the Pi models and their PWR LED support:
* No (or uncontrollable) PWR LED:
    Raspberry Pi Model A
    Raspberry Pi Model B
    Raspberry Pi Model B rev2
    Raspberry Pi Compute Module
    Raspberry Pi Compute Module 2
    Raspberry Pi Compute Module 3
    Raspberry Pi Compute Module 4S
    Raspberry Pi Zero
    Raspberry Pi Zero W
    Raspberry Pi Zero 2 W
* Active-high PWR LED on GPIO 35:
    Raspberry Pi Model A+
    Raspberry Pi Model B+
    Raspberry Pi 2 Model B
    Raspberry Pi 2 Model B rev 1.2
* Active-high PWR LED on expander GPIO 7:
    Raspberry Pi 3 Model B
* Active-low PWR LED on expander GPIO 2:
    Raspberry Pi 3 Model B+
    Raspberry Pi 4 Model B
    Raspberry Pi 400
    Raspberry Pi Compute Module 4

Only the last group of 4 models are affected by the firmware bug that has just been fixed.

You should probably bookmark this post for all of your PWR LED needs.

seamusdemora, 2022-08-27:
If I read all of this correctly, the following (from an [earlier reply](https://github.com/raspberrypi/firmware/issues/1742#issuecomment-1225910618)) is **not** a configuration that will extinguish the power led??

>The safest way to turn off the PWR LED on all Pis with either old or new firmware is to use the default-on method but declare the output as being active-high:

>dtparam=pwr_led_trigger=default-on   # The default
dtparam=pwr_led_activelow=off

Just as a counter-suggestion, if this were included in the documentation, there would be no need to bookmark this post. And since extinguishing the power led is a question that comes up from time-to-time, that might serve the needs of all those users?

ar51an, 2022-08-29:
> That took a while, but I can now see a change in behaviour with releases since Aug 08 2022. However, the new behaviour is correct and the old behaviour was wrong.

Yes the last kernel version where trigger=none used to work was 5.15.55-v8+ and it worked for years this way.

> The safest way to turn off the PWR LED on all Pis with either old or new firmware is to use the default-on method but declare the output as being active-high:
> dtparam=pwr_led_trigger=default-on   # The default
> dtparam=pwr_led_activelow=off

Thank you. Tested on RP4b with kernel 5.15.61-v8+ and it worked.

The correct way is to only specify below line in config.txt:
`dtparam=pwr_led_activelow=off`

Removing trigger=none line sets the default trigger to "default-on":
`dtparam=pwr_led_trigger=none`

Arathonk, 2023-01-18:
Hi! So I'm on Libreelec and  since the new update I can't get the red led to be off,
Normally it would turn of after booting, now it doesn't work,
I tried editing the config file but I don't know what I'm doing wrong.
Libreelec 10.0.4(Kodi build 19.5.0)
RaspPi 4.

Config file text:

# SPDX-License-Identifier: GPL-2.0-or-later
# Copyright (C) 2009-2014 Stephan Raue (stephan@openelec.tv)
# Copyright (C) 2016-present Team LibreELEC (https://libreelec.tv)
################################################################################
# Bootloader configuration
# config.txt version v1 (do not remove or change this line!)
################################################################################
# For more options and information see
# http://rpf.io/configtxt
################################################################################

# Default GPU memory split, 76MB are needed for H264 decoder
gpu_mem=76

# Don't send initial active source message.
# Avoids bringing CEC (enabled TV) out of standby and channel switch when
# rebooting.
hdmi_ignore_cec_init=1

################################################################################
# Include distribution specific config file if it exists.
################################################################################
[all]
include distroconfig.txt

# uncomment to enable infrared remote recevier connected to GPIO 18
#dtoverlay=gpio-ir,gpio_pin=18

# Turn off Power LED
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off

# Turn off Activity LED
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off

# Turn off Ethernet ACT LED
dtparam=eth_led0=4

# Turn off Ethernet LNK LED
dtparam=eth_led1=4

dtparam=i2c=on
enable_uart=1

ar51an, 2023-01-18:
> I can't get the red led to be off
> ... 
> Turn off Power LED
> 
> dtparam=pwr_led_trigger=none
> dtparam=pwr_led_activelow=off

Solution is mentioned in multiple comments of this PR. Remove below line from your config.txt
`dtparam=pwr_led_trigger=none`

pelwell, 2023-01-19:
In the 6.2 kernel there was non-trivial upstream DT change, and as part of the rework of the downstream patches the LED default-state values have been changed to "off", which makes `dtparam=pwr_led_trigger=none` work without the need to make the GPIO active-low. Back-porting at least this aspect of the change to rpi-6.1.y would not be difficult, so I may do it before it becomes the master branch (assuming Greg KH eventually makes 6.1 the next LTS).

pelwell, 2023-01-24:
[#5335 ](https://github.com/raspberrypi/linux/pull/5335) implements the default-state change on rpi-6.1.y.

ar51an, 2023-01-26:
@pelwell Thank you for the heads up and quick turn around

ar51an, 2023-02-28:
Validated on kernel 6.1.13-v8+. Works fine for both pwr and act. Thank you for making it consistent for pwr and act led. That's all needed for switching off pwr & act leds in kernel 6.1.*

dtparam=pwr_led_trigger=none
dtparam=act_led_trigger=none

PThePy, 2023-03-16:
> That took a while, but I can now see a change in behaviour with releases since Aug 08 2022. However, the new behaviour is correct and the old behaviour was wrong.
>
> This isn't easy to say succinctly, [...]

After hours of research, thank you. 
After the upgrade to bullseye the red pw led woudn't turn off even though I've corrected the `dtparam=pwr_led_trigger=default-o` line.
After uprading my firmware using `sudo rpi-update` it worked.
Finally

toy, 2023-12-18:
@Arathonk The problem with `dtparam=eth_led0` and `dtparam=eth_led1` not taken into account should be caused by `config.txt` being moved to `/boot/firmware/config.txt` and `/boot/config.txt` being a symlink to it. So if during editing `/boot/config.txt` symlink gets replaced with a file, it contains config that is not processed. See https://github.com/raspberrypi/firmware/issues/1855

toy, 2023-12-18:
@pelwell Ah, then it must be something else