raspberrypi/firmware #1744: USB boot doesn't work anymore

paolo1983, 2022-08-25:
USB boot no longer works since the August 23 update. The 24 August update Has the same issue. I had to downgrade to the previous One.
I read on archlinux arm forum that a lot of people have the issue since 23 August. Waiting for the fix. Thanks

jvdsn, 2022-08-25:
I have the same issue. Some debugging information:
* Raspberry Pi 4 Model B - 4GB
* bootloader: `c2f8c388 Apr 29 2021` (I don't know why it shows that version, I had the `raspberrypi-bootloader-20220823` package installed)
* update-ts: `1625930123`
* boot: `mode USB-MSD 4 order f14 retry 0/0 restart 0/-1`
* Kernel: `Linux pi 5.18.1-1-aarch64-ARCH`
* Disklabel type: dos
* Partitions:
```
Device     Boot   Start       End   Sectors   Size Id Type
/dev/sda1          2048   1026047   1024000   500M  c W95 FAT32 (LBA)
/dev/sda2       1026048 240353279 239327232 114,1G 83 Linux
```

Failing (package version 20220823) bootloader log (manually transcribed, might contain typo's):
```
USB2[1] 400202e1 connected
USB2 root HUB port 1 init
HUB [01:00] 2.16 000000:01 init port 4 speed 2
USB3[2] 00021203 connected enabled
USB MSD timed out after 20 seconds
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
USB3[2] 000002b1 connected
USB2[1] 400202e1 connected
USB2 root HUB port 1 init
MSD [02:00] 3.32 000000:02 register MSD
MSD [02:00] 3.32 000000:02 LUN 0
MSD INQUIRY [02:00] 3.32 000000:02
MSD [02:00] 3.32 000000:02 lun 0 block-count 240353280 block-size 512
Trying partition: 0
lba: 2048 oem: 'mkfs.fat' volume: '   V       ^ '
rsc 16 fat-sectors 256 c-count 63965 c-size 16 r-dir 1 r-sec 32
Trying partition: 0
lba: 2048 oem: 'mkfs.fat' volume: '   V       ^ '
rsc 16 fat-sectors 256 c-count 63965 c-size 16 r-dir 1 r-sec 32
Read config.txt bytes       65 hnd 0x0000163d
Read pieeprom.upd bytes   524288 hnd 0x00001d98
SELF-UPDATE timestamp current 1625930123 new 1625930123 skip
HUB [01:00] 2.16 000000:01 init port 4 speed 2
Read start4.elf bytes  2249184 hnd 0x000028c1
Read fixup4.dat bytes     5397 hnd 0x0000274b
Firmware: bf881c824bac214626a9f78094d614f7db462c34 Aug 23 2022 14:23:34
0x00c03114 0x00000000 0x00001fff
MEM GPU: 76 ARM: 947 TOTAL: 1023
Starting start4.elf @ 0xfeb00200 partition 0
<nothing>
```

Succeeding (package version 20220812) bootloader log:
```
USB2[1] 400202e1 connected
USB2 root HUB port 1 init
HUB [01:00] 2.16 000000:01 init port 4 speed 2
USB3[2] 00021203 connected enabled
USB MSD timed out after 20 seconds
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
USB3[2] 000002b1 connected
USB2[1] 400202e1 connected
USB2 root HUB port 1 init
MSD [02:00] 3.32 000000:02 register MSD
MSD [02:00] 3.32 000000:02 LUN 0
MSD INQUIRY [02:00] 3.32 000000:02
MSD [02:00] 3.32 000000:02 lun 0 block-count 240353280 block-size 512
Trying partition: 0
lba: 2048 oem: 'mkfs.fat' volume: '   V       ^ '
rsc 16 fat-sectors 256 c-count 63965 c-size 16 r-dir 1 r-sec 32
Trying partition: 0
lba: 2048 oem: 'mkfs.fat' volume: '   V       ^ '
rsc 16 fat-sectors 256 c-count 63965 c-size 16 r-dir 1 r-sec 32
Read config.txt bytes       65 hnd 0x0000163d
Read pieeprom.upd bytes   524288 hnd 0x00001d98
SELF-UPDATE timestamp current 1625930123 new 1625930123 skip
HUB [01:00] 2.16 000000:01 init port 4 speed 2
Read start4.elf bytes  2249024 hnd 0x000028c1
Read fixup4.dat bytes     5399 hnd 0x0000274b
Firmware: 273b410636cf8854ca35af91fd738a3d5f8b39b6 Aug  9 2022 13:44:40
0x00c03114 0x00000000 0x00001fff
MEM GPU: 76 ARM: 947 TOTAL: 1023
Starting start4.elf @ 0xfeb00200 partition 0
<kernel boot log>
```

zhuykovkb, 2022-08-25:
i have the same problem. how to downgrade? write me @paolo1983  please

jvdsn, 2022-08-25:
> i have the same problem. how to downgrade? write me @paolo1983 please

Cleanest way:
* Install Arch Linux ARM on an SD card
* Boot from the SD card
* Install `arch-install-scripts`
* `arch-chroot` in your USB
* Downgrade using a cached package in `/var/cache/pacman/pkg`

cleverca22, 2022-08-25:
@jvdsn can you repeat with `uart_2ndstage=1` in `config.txt` on both versions?

riconero, 2022-08-26:
> i have the same problem. how to downgrade? write me @paolo1983 please

you may try replace start*.elf and fixup*.elf files with the ones in stable channel.
https://github.com/raspberrypi/firmware/tree/stable/boot

kitalia, 2022-08-26:
same here, I also removed the cached packages and reinstalled all the packages in one of the many attempts to restore the system, if you are having the same issue 

chroot into your system and copy and do this:

`cd /home/$USER`
`mkdir rraspberrypi-bootloader`
`cd raspberrypi-bootloader`
`nano raspberrypi-bootloader`
now paste this into nano
```
pkgname=raspberrypi-bootloader
pkgver=20220725
pkgrel=1
 _commit=f95b607c72f9f6d4ea27140635f1e55078ef6f74
arch=('any')
 url="https://github.com/raspberrypi/firmware"
 license=('custom')
 options=(!strip)
 source=("bootloader-$pkgver-${_commit:0:10}.tar.gz::https://github.com/raspberrypi/firmware/archive/${_commit}.tar.gz")
 md5sums=('af86b61362a4012430d92b437d2a07a4')

 package() {
   pkgdesc="Bootloader files for Raspberry Pi"
  replaces=('raspberrypi-firmware-bootloader' 'raspberrypi-bootloader-x' 'raspberrypi-firmware-bootloader-x')
  conflicts=('raspberrypi-bootloader-x')
  mkdir -p "${pkgdir}"/boot
  cp "${srcdir}"/firmware-${_commit}/boot/{*.dat,*.bin,*.elf} "${pkgdir}"/boot
}
```
`makepkg -i`

now exit chroot, shutdown, eject the SD card/USB drive you chroot from and boot

jvdsn, 2022-08-26:
> @jvdsn can you repeat with `uart_2ndstage=1` in `config.txt` on both versions?

I did, but it doesn't seem to add any output in the bootloader log. Note: I'm not using UART per se, simply connecting the pi to an HDMI monitor.

peterharperuk, 2022-08-26:
I can reproduce this on rpi os 64 bit using rpi-update 0006a29c09947f8699d460787a101c34d9c38856
As reported downgrading to 4dde751755c0fa0635678b4f0986471f4bfaa7f2 and it works as does the stable version
Looking for the cause...

riconero, 2022-08-26:
> I can reproduce this on rpi os 64 bit using rpi-update 0006a29c09947f8699d460787a101c34d9c38856 As reported downgrading to 4dde751755c0fa0635678b4f0986471f4bfaa7f2 and it works as does the stable version Looking for the cause...

if it helps,
sth is wrong in start*.elf and fixup*.dat,
because of this commit,
https://github.com/raspberrypi/firmware/commit/900dcf95fe85d9d5bcb34ea43e77cfbc23cf4733

popcornmix, 2022-08-26:
rpi-update has a potential firmware fix for this. Please update and report if it's fixed.

jvdsn, 2022-08-26:
> rpi-update has a potential firmware fix for this. Please update and report if it's fixed.

This commit seems to fix the issue for me. I manually created a PKGBUILD:
```
pkgname=raspberrypi-bootloader
pkgver=20220826
pkgrel=1
 _commit=ecc243c52789f4d8e837c7300b6effb059dc18c0
arch=('any')
 url="https://github.com/raspberrypi/firmware"
 license=('custom')
 options=(!strip)
 source=("bootloader-$pkgver-${_commit:0:10}.tar.gz::https://github.com/raspberrypi/firmware/archive/${_commit}.tar.gz")
 md5sums=('937c117b0b99173bdd396a953367ec22')

 package() {
   pkgdesc="Bootloader files for Raspberry Pi"
  replaces=('raspberrypi-firmware-bootloader' 'raspberrypi-bootloader-x' 'raspberrypi-firmware-bootloader-x')
  conflicts=('raspberrypi-bootloader-x')
  mkdir -p "${pkgdir}"/boot
  cp "${srcdir}"/firmware-${_commit}/boot/{*.dat,*.bin,*.elf} "${pkgdir}"/boot
}
```

So unless I did something wrong there, it should be good. md5sums of my firmware files for confirmation:
```
e7cc09ea93ce5cb806bfb852e294cb31  bootcode.bin
8adf33e6ed14d469727cce7bd1e95966  fixup4cd.dat
255085ad868de9484fc4380807d76222  fixup4.dat
435200b56c611758b14d7124dc75f4c6  fixup4db.dat
e9b688bd0d7cee8631c83d28e28e0660  fixup4x.dat
8adf33e6ed14d469727cce7bd1e95966  fixup_cd.dat
11265335cd197a93640344d86647549c  fixup.dat
97b07418448440f4f191d455baa878bc  fixup_db.dat
6602fc22fdb578f6e7baab38a474385f  fixup_x.dat
1f8a7551b0f7a7407b110c5a8224f38c  start4cd.elf
a03affb35761a5b9c3a673d198b85c1b  start4db.elf
252707916dd7130682bbbd3b87a59b50  start4.elf
adc312d61afb5b56bc5dca85000f80fc  start4x.elf
1a8350e8b53439bc10eae0edc2327ab5  start_cd.elf
556ce32818de4f0478e3c592e34a0ff7  start_db.elf
2859b183f59f71512d07e559742d0b67  start.elf
89eec50e2870dd3303069de85540052f  start_x.elf
```

riconero, 2022-08-26:
the fix works for me. thanks.

paolo1983, 2022-08-26:
I confirm that the fix works

closed 2022-08-26 by peterharperuk