raspberrypi/firmware #1745: NETCONSOLE - MAC address setting ignored?

martinkarsten, 2022-09-01:
**Is this the right place for my bug report?**
bootloader problem

**Describe the bug**
The bootloader (configuration) ignores the destination MAC address setting in NETCONSOLE. I can see the UDP packets using Wireshark, but the destination MAC address is still 00:00:00:00:00:00. I hope with a proper MAC address, it would be possible to capture those debug message using a regular userland tool, such as 'nc'.

**To reproduce**
cat > ~/bootconf.txt <<EOF
NETCONSOLE=6665@192.168.253.65/,6666@192.168.253.1/F0:1E:34:1F:39:6B
EOF

sudo rpi-eeprom-config -o netboot-pieeprom.bin -c ~/bootconf.txt\
 /lib/firmware/raspberrypi/bootloader/beta/pieeprom-2022-03-10.bin
sudo rpi-eeprom-update -d -f netboot-pieeprom.bin
sudo reboot

**Expected behaviour**
UDP packets generated with destination MAC F0:1E:34:1F:39:6B.

**Actual behaviour**
UDP packets generated with destination MAC 00:00:00:00:00:00.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi4B

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 27a8050c3c06e567c794620394a8c2d74262a516, stage2

* Which firmware version (`vcgencmd version`)?
Mar 24 2022 13:19:26 
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux

**Logs**
N/A

**Additional context**
I also tried using lower-case letters with the same result. The bootloader does seem to parse the MAC address. If I supply an incorrect form (e.g., no colons), the NETCONSOLE setting is completely ignored and no output generated.

timg236, 2022-09-01:
Yes, unfortunately, there is a parsing bug and the MAC address in the netconsole is ignored

timg236, 2022-09-01:
This should really be reported against rpi-eeprom, I'll see if someone with suitable Github access can move it

martinkarsten, 2022-09-01:
Thanks! Please let me know if I can help in any way with this, but I believe the firmware is not open source? I am planning to use Raspberry Pis in a course (see below) and it would be seriously cool to have this issue fixed rather sooner than later. Thanks!

https://student.cs.uwaterloo.ca/~cs452/