raspberrypi/firmware #1746: CM4 NVMe failed to boot becuase wrong initrd.img compress format

ww898, 2022-09-15:
Hardware: CM4 d03140
EPROM: 2022-09-02 with `BOOT_ORDER=0xf61`
OS: Ubuntu 22.04 / 22.04.1
Kernel: 5.15.0-1014-raspi

The original image of Ubuntu 22.04.1 for Raspberry Pi taken from https://cdimage.ubuntu.com/releases/22.04.1/release/ubuntu-22.04.1-preinstalled-server-arm64+raspi.img.xz failed to boot from NVMe, but works correctly from SD-Card. The error is: `Initramfs unpacking failed: Decoding failed`. The problem is with `initrd.img` - wrong packing format. The device normally booting with Ubuntu 21.10 where `COMPRESS=zstd`, but failed to boot with Ubuntu 22.04.01 where `COMPRESS=lz4`. I created the new version of `initrd.img` with `COMPRESS=gzip` and CM4 was started correctly.

P.S. The device stop working again after update OS because `initrd.img` was overriten.

timg236, 2022-09-15:
There was a recent firmware (start4.elf) fix for loading large files from NVMe. If you update to the latest stable firmware release from this repo (or APT) then this should work.

ww898, 2022-09-15:
@timg236 Confirm, the new version of `start4.elf` / `fixup4.dat` helps. When these files will be available in Ubuntu release?

timg236, 2022-09-15:
Sorry, that's a question for Ubuntu

closed 2022-09-15 by pelwell