raspberrypi/firmware #1749: RPi4: Any change regarding the boot time issue?

advancel, 2022-09-21:
Hello, i've been researching so long time about raspberry pi 4 boot time. 
Now i get;
1.234s (kernel) | + 2.571s (userspace) = 3.805s
But this "Delay before kernel issue" is looks unsolvable. Because i couldn't find anything about it, except -> https://github.com/raspberrypi/firmware/issues/1375
so after this research i thought i should ask the community and the contributors.

Any progress on this situation?

pelwell, 2022-09-21:
Progress since when? In the years since Pi 4 was launched we've reduced the boot time considerably, but unless we happen to stumble across some wasted time somewhere (which seems unlikely) it is going to remain as it is now.

advancel, 2022-09-21:
Progress since this -> https://github.com/raspberrypi/firmware/issues/1375. Its still same, taking 9 seconds to get display(kernel).
This is so much time for any computer. I thought boot time is only about operating system and its style to handling hardware. Maybe you're right there is nothing to do about it for now.

popcornmix, 2023-01-24:
Closing as we know of no ways to boot faster.

closed 2023-01-24 by popcornmix