raspberrypi/firmware #1752: RPI4B EEPROM bootloader from Tue Aug 2 15:55:05 UTC 2022 (1659455705) causes Pi to hang at rainbow

JerryK13, 2022-10-04:
Raspberry Pi 4B 4GB. Sandisk extreme 32GB sdcard

On a fresh Raspberry Pi OS Lite 64bit (September 22nd 2022) and also rpi-updated to ed6ef1c57077cc3969ae62a4bd9dc1abe5e2a7b7 (kernel 5.15.70-v8+).

With the rpi-eeprom-update set to stable branch and updated, the bootloader Tue Aug 2 15:55:05 UTC 2022 (1659455705) installed, the Pi boots normally. Upon executing reboot command, the Pi always gets stuck on the rainbow with 7 flashes from the power led. Removing the power cable and reattaching it, the Pi boots normally. Executing shutdown and then reattaching power boots normally.
Returning to critical branch and bootloader Tue Apr 26 10:24:28 UTC 2022 (1650968668) alleviates the problem and the Pi reboots normally.

JerryK13, 2022-10-04:
Sorry, it's not the power led, but the green CPU led that is flashing 7 times while being stuck at rainbow.

peterharperuk, 2022-10-04:
I can't seem to reproduce this. 7 flashes implies a problem loading the kernel, which is odd.

JerryK13, 2022-10-04:
```
$ dmesg | grep "Machine model:"
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
```
I have two more RPI4B 4GB that are a little bit younger than this one and will try this sdcard with this image and both bootloader versions to see if it's the card or the Pi.

JerryK13, 2022-10-04:
I have tried this specific sdcard without changing anything in those two additional RPI4B 4GB.  One is `Machine model: Raspberry Pi 4 Model B Rev 1.1` and the other `Machine model: Raspberry Pi 4 Model B Rev 1.2`. I booted them up, ran rpi-eeprom-update, updated their eeproms to 1659455705, rebooted numerous times and it works without a problem. Then I took their sdcards, inserted tem in to problematic Pi, booted up, rebooted numerous times and it also worked. Then I rpi-updated their kernels and retested, reboot works without a problem. Reinserted the original sdcard in to the problematic Pi, booted up, rebooted and it got stuck again at rainbow with 7 flashes. Tried numerous reboots and got stuck every time. Then I deleted /boot/.firmware_revision and reran rpi-update and now reboot works.

What, why, where and how is beyond me. Looks like the sdcard controller was returning something amiss to this particular Pi at those conditions and now that the metadata and data on the partition got rewritten **it is fixed**.

I am sorry for wasting your time. This issue now can be closed.

timg236, 2022-10-05:
Thanks for updating the report with your findings, it's useful for us to track these issues. I think sometimes the card FTL layer gets into a state where it takes a long time to recover, if so, and the bootloader can detect that it could take some action. However, that's just a theory so no promises :)

JerryK13, 2022-10-05:
I don't envy you at all with all the wonky sdcards from hell flooding the market. You are doing an awesome job juggling this, considering the state of affairs from 10 years ago and all the progress made. Keep up the good work!