raspberrypi/firmware #1754: Boot loader files cases eth0 to be disabled on Pi400

solskogen, 2022-10-05:
I'm using Arch Linux ARM and with the latest boot loader files (*.elf, *.bin, *.dat in /boot) I'm no longer able to activate eth0 on my Pi400. Pi4 is not affected. b179e7bf2b0c48577916043909c24173b2d129d1 works, while something happen between that and 2b3cef2f4e9987ab4ad5e07578c2f6192aa7787d which cases the problem.

The latest commit c18d9a5f440838a7a9b3775cd9c281595e58e400 still have the issue.

pelwell, 2022-10-05:
Thanks for the report, and especially for narrowing it down so much. The cause is depressingly obvious (but not to our code checking tools, apparently), and we'll get a fix out ASAP.

pelwell, 2022-10-05:
Thanks, @popcornmix - that fixes my test 400.

solskogen, 2022-10-06:
Fix is confirmed. Thanks all!

closed 2022-10-06 by solskogen