raspberrypi/firmware #1755: CM3-bootloader: Configure Pin 0 and 1 for HAT EEPROM

zhan-kunbus, 2022-10-11:
**Describe the bug**
The HAT EEPROM works not with CM3 without pin defines in dt-blob.bin

**Expected behaviour**
The HAT EEPROM works with CM3 without extra configurations like that on CM4(s)

**System**
* Which model of Raspberry Pi? CM3
* Which OS and version (`cat /etc/rpi-issue`)? Raspberry Pi reference 2022-01-28
* Which firmware version (`vcgencmd version`)? Dec  1 2021 15:07:06
* Which kernel version (`uname -a`)? 5.10
However, as the pin configuration are implemented in the bootloader, the OS and kernel version should not be relevant. 

Here I would like to ask whether it's planned to support HAT EEPROM in bootloader for CM3. If yes, would like to further know when that would be possible.

pelwell, 2022-10-11:
The current behaviour is by design. CM3 is a very flexible platform, and there is no expectation that it would be used in conjunction with HATs, or even that GPIOs 0 & 1 would be used for I2C of any kind.

zhan-kunbus, 2022-10-11:
thanks a lot for the answer.

closed 2022-10-13 by zhan-kunbus