raspberrypi/firmware #1756: bootloader: Configure the GPIO pins in Bank 1 from HAT EEPROM

zhan-kunbus, 2022-10-12:
**Describe the bug**
The GPIO pins in Bank 1 can not be set from HAT EEPROM:
https://github.com/raspberrypi/hats/blob/9616b5cd2bdf3e1d2d0330611387d639c1916100/eeprom-format.md#gpio-map-atom-data-type0x0002

**To reproduce**
Do the configuration for GPIO 28 or above follow the HAT ID EEPROM FORMAT SPECIFICATION, it doesn't work as it does for GPIO 0-27

**Expected behaviour**
The GPIO pins in Bank 1 and Bank 2 (if possible) can also be set from HAT EEPROM 

**System**
* Which model of Raspberry Pi? CM4, CM3
* Which OS and version (`cat /etc/rpi-issue`)?  
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

For the applications we are handling, it helps a lot with the ability of configuring pins in bank 1 (and bank 2) from HAT EEPROM for CM3 and CM4.

pelwell, 2022-10-12:
HATs were not intended for use on Compute Modules. It is not possible to support GPIOs other than in bank 0 without requiring all users to upgrade in order to use HATs with the new format.

Have you considered putting the GPIO settings in a Device Tree overlay? Or on config.txt using the `gpio=` directive (https://www.raspberrypi.com/documentation/computers/config_txt.html#gpio-control)?

iluminat23, 2022-10-12:
Hi @pelwell,

as I discussed with @JamesH65 here https://github.com/raspberrypi/firmware/issues/1666 we use the "HAT" EEPROM to distinguish between variants. 
With CM4 it is not an issue with the bank limitation, but with the CM3 and CM4S based devices it would make our life much easier if we could change the bootup pin configuration of the 2nd bank as well.

We do all pinconfig in the devicetree where appropriate. But for some devices we need to make sure the configuration is correct from boot. Also to my knowledge it is currently not possible to set the drive strength from the kernel.
As we want to provide one image for all our devices, we don't want to change the `config.txt` or the `dt-blob.bin` as this makes the one image for all devices much harder.

Best regards,
Philipp

pelwell, 2022-10-12:
The tools for HAT EEPROM creation are open source. As you are the only user in 8 years to ask for this facility, why not create a PR for the tool so we can decide if we want to take it further?

iluminat23, 2022-10-13:
I opened a draft PR (https://github.com/raspberrypi/hats/pull/92)  with a proposal how this might be implemented. If this looks ok to you we will provide a PR to change the tooling.

iluminat23, 2022-11-08:
New PR also with the implementation: https://github.com/raspberrypi/hats/pull/93

pelwell, 2022-11-11:
https://github.com/raspberrypi/hats/pull/93 is now merged. Also, the latest `rpi-update` firmware should support the bank1 GPIOs.

closed 2022-11-14 by zhan-kunbus