raspberrypi/firmware #1758: Changes to Filesystem gone after reboot - Raspberry Pi

Bra1nsen, 2022-11-02:
I know this topic has been raised on a few forums, but I can't find any clear instructions on how best to approach this problem. 

The Raspberry is on a roof as part of a weather station and is very difficult to access, so I'd like to rule out everything else before touching the hardware.

Can you please tell me how the offical raspberry's recommendations look like regarding to troubleshooting this issue?

kind regards
Paul

pelwell, 2022-11-02:
To avoid any misunderstandings and wasted effort, please explain precisely what the problem is, including descriptions of the relevant hardware, software and configuration.

Bra1nsen, 2022-11-02:
Thanks for reply.

Operating System: Raspbian GNU/Linux 11 (bullseye)
Kernel: Linux 5.15.61-v7l+
Architecture: arm

Any changes to the filesystem e.g. editing a file, creating a folder are gone after rebooting the pi. How to proceed in such a case?

pelwell, 2022-11-02:
Is this a standard Raspberry Pi OS image? It sounds like your SD card is being treated as if it were read-only.

1. Does `dmesg -l warn,err` report anything that looks relevant?
2. Have you tried with a different card?

Bra1nsen, 2022-11-02:
looks relevant:

```
pi@allskyLA:~ $ dmesg -l warn,err
[    0.181747] usb_phy_generic phy: supply vcc not found, using dummy regulator
[    0.181939] usb_phy_generic phy: dummy supplies not allowed for exclusive requests
[    3.662194] systemd[1]: /lib/systemd/system/plymouth-start.service:16: Unit configured to use KillMode=none. This is unsafe, as it disables systemd's process lifecycle management for the service. Please update your service to use a safer KillMode=, such as 'mixed' or 'control-group'. Support for KillMode=none is deprecated and will eventually be removed.
[    4.728827] systemd-journald[142]: File /var/log/journal/ef4df6d3a9a544f2a1705ace80cce517/system.journal corrupted or uncleanly shut down, renaming and replacing.
[    9.521497] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
[    9.579984] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[    9.903194] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[    9.923730] bcm2835_isp: module is from the staging directory, the quality is unknown, you have been warned.
[   10.013183] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
[   10.079700] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[   10.104212] rpivid_hevc: module is from the staging directory, the quality is unknown, you have been warned.
[   10.471445] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[   11.081430] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.bin failed with error -2
[   19.559035] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   19.559059]  MAI: soc_pcm_open() failed (-19)
[   19.559677] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   19.559696]  MAI: soc_pcm_open() failed (-19)
[   19.560246] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   19.560264]  MAI: soc_pcm_open() failed (-19)
[   19.567928] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   19.567953]  MAI: soc_pcm_open() failed (-19)
[   19.568221] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   19.568236]  MAI: soc_pcm_open() failed (-19)
[   19.568485] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   19.568499]  MAI: soc_pcm_open() failed (-19)
[   19.597447] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   19.597472]  MAI: soc_pcm_open() failed (-19)
[   19.598035] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   19.598053]  MAI: soc_pcm_open() failed (-19)
[   19.598688] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   19.598710]  MAI: soc_pcm_open() failed (-19)
[   19.606399] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   19.606423]  MAI: soc_pcm_open() failed (-19)
[   19.606729] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   19.606748]  MAI: soc_pcm_open() failed (-19)
[   19.607013] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   19.607030]  MAI: soc_pcm_open() failed (-19)
[   21.598517] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   21.598551]  MAI: soc_pcm_open() failed (-19)
[   21.599139] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   21.599162]  MAI: soc_pcm_open() failed (-19)
[   21.599723] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   21.599744]  MAI: soc_pcm_open() failed (-19)
[   21.607370] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   21.607401]  MAI: soc_pcm_open() failed (-19)
[   21.607680] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   21.607696]  MAI: soc_pcm_open() failed (-19)
[   21.607955] hdmi-audio-codec hdmi-audio-codec.4.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   21.607971]  MAI: soc_pcm_open() failed (-19)
[   21.639067] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   21.639092]  MAI: soc_pcm_open() failed (-19)
[   21.639744] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   21.639762]  MAI: soc_pcm_open() failed (-19)
[   21.640285] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   21.640299]  MAI: soc_pcm_open() failed (-19)
[   21.647636] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   21.647669]  MAI: soc_pcm_open() failed (-19)
[   21.647947] hdmi-audio-codec hdmi-audio-codec.3.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
[   21.647965]  MAI: soc_pcm_open() failed (-19)
[... cut ...]
[   42.083177] mmc0: Got data interrupt 0x00000002 even though no data operation was in progress.
[  132.289786] v3d fec00000.v3d: MMU error from client CLE (4) at 0x2901000, pte invalid
[  243.675673] INFO: task kworker/2:4:711 blocked for more than 122 seconds.
[  243.675702]       Tainted: G         C        5.15.61-v7l+ #1579
[  243.675720] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  258.046311] unicam fe801000.csi: Failed to start media pipeline: -22

```

not yet, i have to climb on my roof and disassemble my weather station for that..

Bra1nsen, 2022-11-04:
pelwell?

pelwell, 2022-11-04:
You could have a try, but I don't like the look of these error messages:
```
[   33.855175] mmc0: Got data interrupt 0x00000002 even though no data operation was in progress.
[   42.083177] mmc0: Got data interrupt 0x00000002 even though no data operation was in progress.
```
That suggests that card itself (or some other aspect of the hardware) is misbehaving.

Bra1nsen, 2022-11-04:
Anything else you would try?

Gives a way to better protect the SD card in the future. I can only explain that this situation came about because the Raspberry is often switched off hard (current I/O).

Might bring something to command `sudo rpi-update` ?

Bra1nsen, 2022-11-04:
Currently, this always comes on your recommended command:
```
pi@allskyLA:~ $ dmesg -l warn,err
[92816.852757] systemd-journald[142]: Failed to write entry (25 items, 1342 bytes), ignoring: Cannot assign requested address
[92816.853039] systemd-journald[142]: Failed to write entry (25 items, 1333 bytes), ignoring: Cannot assign requested address
[92816.899275] systemd-journald[142]: Failed to write entry (25 items, 1342 bytes), ignoring: Cannot assign requested address
```

pelwell, 2022-11-04:
Have you tried `df`? Is your filesystem full?

In the longer term you need to think about whether your weather station needs to log results persistently. If not, you may be able to make the SD card filing system read-only and overlay a writeable RAM disk, or perhaps to create a separate data partition that is writeable and overlay that on top of the read-only root fs.

pelwell, 2022-11-04:
By the way, this isn't the right venue for general help requests - you're more likely to find someone with personal experience of similar problems and solutions in the Advanced Users forum: https://forums.raspberrypi.com/viewforum.php?f=29

Bra1nsen, 2022-11-04:
```
pi@allskyLA:~ $ sudo systemctl --force reboot
Failed to execute operation: Transport endpoint is not connected

```

pelwell, 2022-11-04:
It's dead, Jim.

kripton, 2022-11-04:
Well. before climbing to the roof, I would try an unexpected reboot by cutting the power and re-applying it. That should be possible from inside I hope :) If it boots: Fine, try running fsck. If not: was worth a try, now you might need to climb

Bra1nsen, 2022-11-04:
hey kripton, iam still logind. do you have any instructions/helpful links for me?

Bra1nsen, 2022-11-04:
If I turn off on power supply, the pi boots up again. but still cant save changes to the filesystem...

closed 2022-11-04 by Bra1nsen