raspberrypi/firmware #1760: Missing or Incomplete kernel headers  for version 5.15.78+?

Jibun-no-Kage, 2022-11-14:
Error?
# make
make ARCH=armv6l CROSS_COMPILE= -C /lib/modules/5.15.78+/build M=/root/8188eus-20210615 modules
make[1]: *** /lib/modules/5.15.78+/build: No such file or directory.  Stop.
make: *** [Makefile:2489: modules] Error 2

# uname -r
5.15.78+

Unable to compile any drivers.

popcornmix, 2022-11-14:
The stable kernel in apt is 5.15.76, and the apt kernel headers should match that.

If you are using rpi-update kernel then you might want to try https://github.com/RPi-Distro/rpi-source

Jibun-no-Kage, 2022-11-14:
Is rpi-source still active?  I thought it was dead?  Will give it a shot.  As for using rpi-update, I am testing a new overlay update, so in a bit of a catch 22.  I need rpi-update latest for the new overlay, but since the headers have yet to be published, can't test the entire image, since I need additional wireless dongle drivers as well.

Jibun-no-Kage, 2022-11-14:
Ah, since rpi-source is based on python2?  Another catch 22.  So as I recalled, rpi-source broken.

popcornmix, 2022-11-14:
python2 can still be installed on bullseye.
```
i@pi4:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@pi4:~ $ sudo apt install python2
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python2 is already the newest version (2.7.18-3).
python2 set to manually installed.
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 13 not upgraded.
pi@pi4:~ $ python2 --version
Python 2.7.18
```

Jibun-no-Kage, 2022-11-14:
Yes, just not a great solution.  I started looking at the rpi-source to get an idea of how ugly it might be to update it.  I know python, but not sure my python skills/experience is up to the task, but figured I would try.

popcornmix, 2022-11-14:
I'm sure we'd welcome a PR to update it to python 3.
if there's anything that is problematic, then that can be discussed in the PR.

Jibun-no-Kage, 2022-11-14:
Yup