raspberrypi/firmware #1762: MMAL: mmal_vc_port_info_set: failed to set port info (2:0): EINVAL on H.264 input

dr-m, 2022-11-22:
**Is this the right place for my bug report?**
I believe yes.

**Describe the bug**
[zillevdr/vdr-plugin-softhddevice-drm](https://github.com/zillevdr/vdr-plugin-softhddevice-drm/) compiled with `MMAL=1` crashes on interlaced H.264 input (DVB-T2 channel). MPEG-2 DVB-T streams work fine.

This seems to be a regression somewhere between 518ee7c871aaa9aaa88116953d57e73787ee6e43 and 4877a7d918be6fab22ce5d345feefd9c43200a90.

**To reproduce**
Compile or `sudo apt install vdr` and compile [zillevdr/vdr-plugin-softhddevice-drm](https://github.com/zillevdr/vdr-plugin-softhddevice-drm/) for it.
Try to play back a H.264 recording.
**Expected behaviour**
The interlaced H.264 video recording should play back with GPU acceleration.

**Actual behaviour**
You may see the on-screen display (OSD) to indicate that the channel was changed or the playback started. Before any image is displayed, the process will crash with SIGSEGV as noted in zillevdr/vdr-plugin-softhddevice-drm#11.

**System**
This is with 518ee7c871aaa9aaa88116953d57e73787ee6e43 and not an affected version like 4877a7d918be6fab22ce5d345feefd9c43200a90 or 74c4307951701f93bd94b4fd37e6adc44910d6e0:
```
System Information
------------------

Raspberry Pi 2 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2022-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, a17501d7c91a584085cd794ab9c007c9d1b9b435, stage2

Linux raspberrypi 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux
Revision	: a01041
Serial		: 00000000d2444097
Model		: Raspberry Pi 2 Model B Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Apr 30 2021 13:47:07 
Copyright (c) 2012 Broadcom
version d7f29d96450abfc77cd6cf011af1faf1e03e5e56 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/root       29441852 10632328  17585364  38% /
devtmpfs          348568        0    348568   0% /dev
tmpfs             381848        0    381848   0% /dev/shm
tmpfs             381848     9928    371920   3% /run
tmpfs               5120        4      5116   1% /run/lock
tmpfs             381848        0    381848   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095    49126    208969  20% /boot
tmpfs              76368        0     76368   0% /run/user/1000

Filename				Type		Size		Used		Priority
/var/swap                               file		102396		0		-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: (none)
raspberrypi-sys-mods:
[... 506 lines cut ...]
Failed send buffer to deinterlacer output port (3, EINVAL)
Failed send buffer to deinterlacer output port (3, EINVAL)
Failed send buffer to deinterlacer output port (3, EINVAL)

Thread 15 "softhddev video" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x612ff400 (LWP 2379)]
VideoRenderFrame (render=0x2ea2e0, video_ctx=<optimized out>, 
    frame=<optimized out>) at video_mmal.c:889
889		memcpy(qbuffer->data, buffer->data, buffer->length);
(gdb) bt
#0  VideoRenderFrame (render=0x2ea2e0, video_ctx=<optimized out>, 
    frame=<optimized out>) at video_mmal.c:889
#1  0x76454fe4 in CodecVideoReceiveFrame (decoder=0x309328, 
    no_deint=no_deint@entry=0) at codec.c:376
#2  0x7644e9d0 in VideoDecodeInput (stream=0x7646a858 <MyVideoStream>)
    at softhddev.c:1117
#3  0x76450498 in DisplayHandlerThread (arg=0x2ea2e0) at video_mmal.c:723
#4  0x76f4f494 in start_thread (arg=0x612ff400) at pthread_create.c:486
#5  0x76bb0dd8 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73
   from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

**Additional context**
MPEG-2 and H.264 video playback works fine via the OMX interface (https://github.com/reufer/rpihddevice/).

dr-m, 2022-11-22:
The H.264 playback fails also with the currently latest available firmware according to `sudo rpi-update`: raspberrypi/rpi-firmware@bcd35b7abb058b210b33922ba60efc1cc93f533d (5.15.79). MPEG-2 and H.264 via OMX using rpihddevice works fine also with this kernel.

dr-m, 2022-11-24:
I added a few `mmal_log_dump_port()` calls and set the environment variable. Here is the output for raspberrypi/rpi-firmware@bcd35b7abb058b210b33922ba60efc1cc93f533d starting a little before the first `EINVAL`:
```
mmal: mmal_vc_component_create:  handle 48
mmal: mmal_vc_port_info_get: get port info (1:0)
mmal: mmal_vc_port_info_get: get port info (2:0)
mmal: mmal_vc_port_info_get: get port info (3:0)
mmal: mmal_pool_create_with_allocator: allocating 48 + 24 + 472 * 4 bytes for pool
mmal: mmal_pool_initialise_buffer_headers: allocating 740 bytes for payload 0/4
mmal: mmal_pool_initialise_buffer_headers: allocating 740 bytes for payload 1/4
mmal: mmal_pool_initialise_buffer_headers: allocating 740 bytes for payload 2/4
mmal: mmal_pool_initialise_buffer_headers: allocating 740 bytes for payload 3/4
mmal: mmal_component_create_core: created 'vc.ril.image_fx' 8 0x63421280
mmal: mmal_port_parameter_set: vc.ril.image_fx(3:0) port 0x63421d40, param 0x612fbd30 (10032,40)
mmal: mmal_port_enable: vc.ril.image_fx:ctr:0 port 0x63421670, cb 0x7644f7d0, buffers (4/0/4,740/0/740)
mmal: mmal_port_format_commit: vc.ril.image_fx(2:0) port 0x63421980 format 3:OPQV
mmal: mmal_vc_port_info_set: set port info (2:0)
mmal: mmal_vc_port_info_set: failed to set port info (2:0): EINVAL
mmal: mmal_vc_port_set_format: mmal_vc_port_info_set failed 0x63421980 (EINVAL)
Failed to commit deinterlace intput format (status=3 EINVAL)
mmal: mmal_log_dump_port: vc.ril.image_fx:ctr:0(0x63421670)
mmal: mmal_log_dump_format: type: unknown, fourcc:     
mmal: mmal_log_dump_format:  bitrate: 0, framed: 0
mmal: mmal_log_dump_format:  extra data: 0, (nil)
mmal: mmal_log_dump_port:  buffers num: 4(opt 0, min 4), size: 740(opt 0, min: 740), align: 0
mmal: mmal_log_dump_port: vc.ril.image_fx:in:0(OPQV)(0x63421980)
mmal: mmal_log_dump_format: type: video, fourcc: OPQV
mmal: mmal_log_dump_format:  bitrate: 0, framed: 1
mmal: mmal_log_dump_format:  extra data: 0, (nil)
mmal: mmal_log_dump_format:  width: 1920, height: 1080, (0,0,1920,1080)
mmal: mmal_log_dump_format:  pixel aspect ratio: 1/1, frame rate: 1571292/65536
mmal: mmal_log_dump_port:  buffers num: 1(opt 1, min 1), size: 3840(opt 3840, min: 3840), align: 0
mmal: mmal_port_parameter_set: vc.ril.image_fx(2:0) port 0x63421980, param 0x612fbd0c (20013,12)
mmal: mmal_port_enable: vc.ril.image_fx:in:0(OPQV) port 0x63421980, cb 0x764500d4, buffers (1/1/1,3840/3840/3840)
mmal: mmal_vc_port_enable: failed to enable port vc.ril.image_fx:in:0(OPQV): EINVAL
mmal: mmal_port_enable: failed to enable port vc.ril.image_fx:in:0(OPQV)(0x63421980) (EINVAL)
Failed to enable deinterlace input port vc.ril.image_fx:in:0(OPQV) (3, EINVAL)
mmal: mmal_log_dump_port: vc.ril.image_fx:ctr:0(0x63421670)
mmal: mmal_log_dump_format: type: unknown, fourcc:     
mmal: mmal_log_dump_format:  bitrate: 0, framed: 0
mmal: mmal_log_dump_format:  extra data: 0, (nil)
mmal: mmal_log_dump_port:  buffers num: 4(opt 0, min 4), size: 740(opt 0, min: 740), align: 0
mmal: mmal_log_dump_port: vc.ril.image_fx:in:0(OPQV)(0x63421980)
mmal: mmal_log_dump_format: type: video, fourcc: OPQV
mmal: mmal_log_dump_format:  bitrate: 0, framed: 1
mmal: mmal_log_dump_format:  extra data: 0, (nil)
mmal: mmal_log_dump_format:  width: 1920, height: 1080, (0,0,1920,1080)
mmal: mmal_log_dump_format:  pixel aspect ratio: 1/1, frame rate: 1571292/65536
mmal: mmal_log_dump_port:  buffers num: 1(opt 1, min 1), size: 3840(opt 3840, min: 3840), align: 0
mmal: mmal_port_format_commit: vc.ril.image_fx(3:0) port 0x63421d40 format 3:OPQV
mmal: mmal_vc_port_info_set: set port info (3:0)
mmal: mmal_vc_port_info_set: failed to set port info (3:0): EINVAL
mmal: mmal_vc_port_set_format: mmal_vc_port_info_set failed 0x63421d40 (EINVAL)
Failed to commit deinterlace output format (status=3 EINVAL)
mmal: mmal_log_dump_port: vc.ril.image_fx:ctr:0(0x63421670)
mmal: mmal_log_dump_format: type: unknown, fourcc:     
mmal: mmal_log_dump_format:  bitrate: 0, framed: 0
mmal: mmal_log_dump_format:  extra data: 0, (nil)
mmal: mmal_log_dump_port:  buffers num: 4(opt 0, min 4), size: 740(opt 0, min: 740), align: 0
mmal: mmal_log_dump_port: vc.ril.image_fx:in:0(OPQV)(0x63421980)
mmal: mmal_log_dump_format: type: video, fourcc: OPQV
mmal: mmal_log_dump_format:  bitrate: 0, framed: 1
mmal: mmal_log_dump_format:  extra data: 0, (nil)
mmal: mmal_log_dump_format:  width: 1920, height: 1080, (0,0,1920,1080)
mmal: mmal_log_dump_format:  pixel aspect ratio: 1/1, frame rate: 1571292/65536
mmal: mmal_log_dump_port:  buffers num: 1(opt 1, min 1), size: 3840(opt 3840, min: 3840), align: 0
mmal: mmal_component_enable: vc.ril.image_fx 8
mmal: mmal_port_enable: vc.ril.image_fx:out:0(OPQV) port 0x63421d40, cb 0x7644fc78, buffers (1/1/1,3840/3840/3840)
mmal: mmal_vc_port_enable: failed to enable port vc.ril.image_fx:out:0(OPQV): EINVAL
mmal: mmal_port_enable: failed to enable port vc.ril.image_fx:out:0(OPQV)(0x63421d40) (EINVAL)
Failed to enable deinterlacer output port vc.ril.image_fx:out:0(OPQV) (3, EINVAL)
[... 20 lines cut ...]
mmal: mmal_log_dump_format: type: video, fourcc: OPQV
mmal: mmal_log_dump_format:  bitrate: 0, framed: 1
mmal: mmal_log_dump_format:  extra data: 0, (nil)
mmal: mmal_log_dump_format:  width: 1920, height: 1080, (0,0,1920,1080)
mmal: mmal_log_dump_format:  pixel aspect ratio: 1/1, frame rate: 1571292/65536
mmal: mmal_log_dump_port:  buffers num: 1(opt 1, min 1), size: 3840(opt 3840, min: 3840), align: 0
Failed send buffer to deinterlacer output port (3, EINVAL)
[... cut ...]
mmal: mmal_log_dump_port: vc.ril.image_fx:out:0(OPQV)(0x63421d40)
mmal: mmal_log_dump_format: type: video, fourcc: OPQV
mmal: mmal_log_dump_format:  bitrate: 0, framed: 1
mmal: mmal_log_dump_format:  extra data: 0, (nil)
mmal: mmal_log_dump_format:  width: 1920, height: 1080, (0,0,1920,1080)
mmal: mmal_log_dump_format:  pixel aspect ratio: 1/1, frame rate: 1571292/65536
mmal: mmal_log_dump_port:  buffers num: 1(opt 1, min 1), size: 3840(opt 3840, min: 3840), align: 0

Thread 15 "softhddev video" received signal SIGSEGV, Segmentation fault.
```

6by9, 2022-11-24:
There is a commit in that time frame relating to the padded height (```port->format->es->video.height```) on image_fx.
The deinterlacing algorithm needs the frame to be padded to at least a multiple of 16 lines above the height.

When using opaque buffers the format is slightly redundant as the buffers include the full geometry, but I suspect is still being validated.
Can you try adding
```render->deint->input[0]->format->es->video.height = VCOS_ALIGN_UP(render->deint->input[0]->format->es->video.height, 16);```
after the ```mmal_port_format_copy at ``` in https://github.com/zillevdr/vdr-plugin-softhddevice-drm/blob/drm/video_mmal.c#L427

dr-m, 2022-11-24:
Thank you, I will try your suggestion later.

I just downgraded to 518ee7c871aaa9aaa88116953d57e73787ee6e43. Here is the corresponding output (with no crash). Both were from a live DVB-T2 stream, so some parameters might be different:
```
mmal: mmal_vc_component_create:  handle 48
mmal: mmal_vc_port_info_get: get port info (1:0)
mmal: mmal_vc_port_info_get: get port info (2:0)
mmal: mmal_vc_port_info_get: get port info (3:0)
mmal: mmal_pool_create_with_allocator: allocating 48 + 24 + 472 * 4 bytes for pool
mmal: mmal_pool_initialise_buffer_headers: allocating 740 bytes for payload 0/4
mmal: mmal_pool_initialise_buffer_headers: allocating 740 bytes for payload 1/4
mmal: mmal_pool_initialise_buffer_headers: allocating 740 bytes for payload 2/4
mmal: mmal_pool_initialise_buffer_headers: allocating 740 bytes for payload 3/4
mmal: mmal_component_create_core: created 'vc.ril.image_fx' 8 0x6341f230
mmal: mmal_port_parameter_set: vc.ril.image_fx(3:0) port 0x6341fcf0, param 0x612fbd30 (10032,40)
mmal: mmal_port_enable: vc.ril.image_fx:ctr:0 port 0x6341f620, cb 0x7644f7d0, buffers (4/0/4,740/0/740)
mmal: mmal_port_format_commit: vc.ril.image_fx(2:0) port 0x6341f930 format 3:OPQV
mmal: mmal_vc_port_info_set: set port info (2:0)
mmal: mmal_vc_port_info_get: get port info (2:0)
mmal: mmal_vc_port_info_get: get port info (3:0)
mmal: mmal_port_parameter_set: vc.ril.image_fx(2:0) port 0x6341f930, param 0x612fbd0c (20013,12)
mmal: mmal_port_enable: vc.ril.image_fx:in:0(OPQV) port 0x6341f930, cb 0x764500d4, buffers (10/10/3,128/128/128)
mmal: mmal_port_format_commit: vc.ril.image_fx(3:0) port 0x6341fcf0 format 3:OPQV
mmal: mmal_vc_port_info_set: set port info (3:0)
mmal: mmal_vc_port_info_get: get port info (3:0)
mmal: mmal_component_enable: vc.ril.image_fx 8
mmal: mmal_port_enable: vc.ril.image_fx:out:0(OPQV) port 0x6341fcf0, cb 0x7644fc78, buffers (10/10/3,128/128/128)
mmal: mmal_pool_create_with_allocator: allocating 48 + 48 + 472 * 10 bytes for pool
mmal: mmal_pool_initialise_buffer_headers: allocating 128 bytes for payload 0/10
mmal: mmal_port_payload_alloc: vc.ril.image_fx(2:0) port 0x6341f930, size 128
mmal: mmal_vc_port_payload_alloc: vc.ril.image_fx:in:0(OPQV): allocating 128 bytes, format OPQV, is_zero_copy 0
mmal: mmal_vc_port_payload_alloc: vc.ril.image_fx:in:0(OPQV): allocated at 0x634216d0
mmal: mmal_port_acquire: port vc.ril.image_fx:in:0(OPQV)(0x6341f930), refcount 0
mmal: mmal_pool_initialise_buffer_headers: allocating 128 bytes for payload 1/10
mmal: mmal_port_payload_alloc: vc.ril.image_fx(2:0) port 0x6341f930, size 128
mmal: mmal_vc_port_payload_alloc: vc.ril.image_fx:in:0(OPQV): allocating 128 bytes, format OPQV, is_zero_copy 0
mmal: mmal_vc_port_payload_alloc: vc.ril.image_fx:in:0(OPQV): allocated at 0x63421770
mmal: mmal_port_acquire: port vc.ril.image_fx:in:0(OPQV)(0x6341f930), refcount 1
mmal: mmal_pool_initialise_buffer_headers: allocating 128 bytes for payload 2/10
mmal: mmal_port_payload_alloc: vc.ril.image_fx(2:0) port 0x6341f930, size 128
mmal: mmal_vc_port_payload_alloc: vc.ril.image_fx:in:0(OPQV): allocating 128 bytes, format OPQV, is_zero_copy 0
mmal: mmal_vc_port_payload_alloc: vc.ril.image_fx:in:0(OPQV): allocated at 0x63421820
mmal: mmal_port_acquire: port vc.ril.image_fx:in:0(OPQV)(0x6341f930), refcount 2
mmal: mmal_pool_initialise_buffer_headers: allocating 128 bytes for payload 3/10
mmal: mmal_port_payload_alloc: vc.ril.image_fx(2:0) port 0x6341f930, size 128
mmal: mmal_vc_port_payload_alloc: vc.ril.image_fx:in:0(OPQV): allocating 128 bytes, format OPQV, is_zero_copy 0
mmal: mmal_vc_port_payload_alloc: vc.ril.image_fx:in:0(OPQV): allocated at 0x634218c0
mmal: mmal_port_acquire: port vc.ril.image_fx:in:0(OPQV)(0x6341f930), refcount 3
mmal: mmal_pool_initialise_buffer_headers: allocating 128 bytes for payload 4/10
mmal: mmal_port_payload_alloc: vc.ril.image_fx(2:0) port 0x6341f930, size 128
mmal: mmal_vc_port_payload_alloc: vc.ril.image_fx:in:0(OPQV): allocating 128 bytes, format OPQV, is_zero_copy 0
mmal: mmal_vc_port_payload_alloc: vc.ril.image_fx:in:0(OPQV): allocated at 0x63421970
mmal: mmal_port_acquire: port vc.ril.image_fx:in:0(OPQV)(0x6341f930), refcount 4
mmal: mmal_pool_initialise_buffer_headers: allocating 128 bytes for payload 5/10
mmal: mmal_port_payload_alloc: vc.ril.image_fx(2:0) port 0x6341f930, size 128
mmal: mmal_vc_port_payload_alloc: vc.ril.image_fx:in:0(OPQV): allocating 128 bytes, format OPQV, is_zero_copy 0
mmal: mmal_vc_port_payload_alloc: vc.ril.image_fx:in:0(OPQV): allocated at 0x63421a10
mmal: mmal_port_acquire: port vc.ril.image_fx:in:0(OPQV)(0x6341f930), refcount 5
mmal: mmal_pool_initialise_buffer_headers: allocating 128 bytes for payload 6/10
mmal: mmal_port_payload_alloc: vc.ril.image_fx(2:0) port 0x6341f930, size 128
mmal: mmal_vc_port_payload_alloc: vc.ril.image_fx:in:0(OPQV): allocating 128 bytes, format OPQV, is_zero_copy 0
mmal: mmal_vc_port_payload_alloc: vc.ril.image_fx:in:0(OPQV): allocated at 0x63421ac0
mmal: mmal_port_acquire: port vc.ril.image_fx:in:0(OPQV)(0x6341f930), refcount 6
mmal: mmal_pool_initialise_buffer_headers: allocating 128 bytes for payload 7/10
mmal: mmal_port_payload_alloc: vc.ril.image_fx(2:0) port 0x6341f930, size 128
mmal: mmal_vc_port_payload_alloc: vc.ril.image_fx:in:0(OPQV): allocating 128 bytes, format OPQV, is_zero_copy 0
mmal: mmal_vc_port_payload_alloc: vc.ril.image_fx:in:0(OPQV): allocated at 0x63426780
mmal: mmal_port_acquire: port vc.ril.image_fx:in:0(OPQV)(0x6341f930), refcount 7
[... cut ...]
mmal: mmal_port_acquire: port vc.ril.image_fx:in:0(OPQV)(0x6341f930), refcount 8
mmal: mmal_pool_initialise_buffer_headers: allocating 128 bytes for payload 9/10
mmal: mmal_port_payload_alloc: vc.ril.image_fx(2:0) port 0x6341f930, size 128
mmal: mmal_vc_port_payload_alloc: vc.ril.image_fx:in:0(OPQV): allocating 128 bytes, format OPQV, is_zero_copy 0
mmal: mmal_vc_port_payload_alloc: vc.ril.image_fx:in:0(OPQV): allocated at 0x634268d0
mmal: mmal_port_acquire: port vc.ril.image_fx:in:0(OPQV)(0x6341f930), refcount 9
```

dr-m, 2022-11-25:
There are two `mmal_port_format_copy()` calls in the file. I first added the call to the wrong place, and it did not stop the crashes.

After I added the suggested tweak to both call sites, it stopped crashing on raspberrypi/rpi-firmware@bcd35b7abb058b210b33922ba60efc1cc93f533d, but I did not get any H.264 video either (only audio); the initial DVB-T MPEG-2 live stream was playing fine. I then reverted my changes and applied only your suggested fix, to see if it would be better. Same thing: the video disappeared but audio continued playing, with some interruptions. Even after I switched back to a DVB-T channel, the video remained absent and the audio was choppy. (Initially, the DVB-T output was playing fine).

Here is my reduced patch against zillevdr/vdr-plugin-softhddevice@cbccef7cc37c3122c75f58000a50227fe468ca45:
```diff
diff --git a/Makefile b/Makefile
index 93391b5..f51e21b 100644
--- a/Makefile
+++ b/Makefile
@@ -12,7 +12,7 @@ PLUGIN = softhddevice-drm
 ### Configuration (edit this for your needs)
 
 	# enable this for MMAL (RaspberryPi 2)
-MMAL ?= 0
+MMAL ?= 1
 
 CONFIG := #-DDEBUG 				# enable debug output+functions
 #CONFIG += -DAV_SYNC_DEBUG		# enable debug messages AV_SYNC
diff --git a/video_mmal.c b/video_mmal.c
index 5b9dd65..7c5181c 100644
--- a/video_mmal.c
+++ b/video_mmal.c
@@ -425,6 +425,7 @@ static void MmalChangeResolution(VideoRender * render,
 				render->vout->control->name, status, mmal_status_to_string(status));
 
 		mmal_format_copy(render->deint->input[0]->format, render->vout->input[0]->format);
+render->deint->input[0]->format->es->video.height = VCOS_ALIGN_UP(render->deint->input[0]->format->es->video.height, 16);
 		status = mmal_port_format_commit(render->deint->input[0]);
 		if (status != MMAL_SUCCESS)
 			printf("Failed to commit deinterlace intput format (status=%"PRIx32" %s)\n",
```

dr-m, 2022-11-25:
On b324aea801f669b6ab18441f970e74a5a7346684 the playback continues fine even with the above patch.