raspberrypi/firmware #1763: RPi4 reboot hangs with "kernel not found" blinks when running at 4kp60

HiassofT, 2022-12-06:
**Describe the bug**
When rebooting my RPi4 4GB rev 1.1 connected to a 4K TV rebooting often hangs at the rainbow splash and UART reports that it could not read cmdline.txt and kernel.img from SD card.

The issue is a bit intermittent, sometimes they survive about 5 reboots but quite often the first reboot already fails.

I reproduced that with 2 rev 1.1 RPi4s and 3 different SD cards (two Sandisk Extreme 32G and one Sandisk Extreme Pro 32GB).

**To reproduce**
The key point seems to be that 4kp60 is enabled and the RPi is connected to a 4kp60 capable TV - or the edid of the 4k TV is used - I've used the latter to reproduce it without any peripherals except serial console connected.

I used this config.txt with the attached edid file [lg-55c8.bin.zip](https://github.com/raspberrypi/firmware/files/10170214/lg-55c8.bin.zip)
```
uart_2ndstage=1

dtoverlay=disable-bt

hdmi_enable_4kp60=1
hdmi_edid_filename:0=lg-55c8.bin
hdmi_edid_file=1
```

I dd a lot of reboots with and without 4kp60 enabled when connected to a 1920x1200 monitor and that worked fine (boot-looping for an hour).

**Expected behaviour**
RPi4 reboots fine without locking up in firmware.

**Actual behaviour**
On an unsuccessful reboot firmware reports that it couldn't load files from the SD card.

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
RPi4 4GB rev 1.1. cpuinfo:
```
Hardware	: BCM2711
Revision	: c03111
Serial		: 100000008eb25f41
Model		: Raspberry Pi 4 Model B Rev 1.1
```

* Which OS and version (`cat /etc/rpi-issue`)?
RPi OS Bullseye Lite, fully updated yesterday
```
Raspberry Pi reference 2022-01-28
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f01430c9d8f67a4b9719cc00e74a2079d3834d5d, stage2
```

* Which firmware version (`vcgencmd version`)?
```
Nov 11 2022 11:13:01 
Copyright (c) 2012 Broadcom
version 1a79bba5cc84ea52d15809fff557479fdf084529 (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
```
Linux raspberrypi 5.15.80-v7l+ #1602 SMP Tue Nov 29 14:45:31 GMT 2022 armv7l GNU/Linux
```

bootloader:
```
root@raspberrypi:~# rpi-eeprom-update 
BOOTLOADER: up to date
   CURRENT: Fri 25 Nov 2022 10:54:13 AM UTC (1669373653)
    LATEST: Fri 25 Nov 2022 10:54:13 AM UTC (1669373653)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138a1
[... 492 lines cut ...]
[    5.732363] systemd[1]: Starting Create list of static device nodes for the current kernel...
[    5.766527] systemd[1]: Starting Load Kernel Module configfs...
[    5.798202] systemd[1]: Starting Load Kernel Module drm...
[    5.828565] systemd[1]: Starting Load Kernel Module fuse...
[    5.856233] systemd[1]: Condition check resulted in Set Up Additional Binary Formats being skipped.
[    5.878545] systemd[1]: Starting File System Check on Root Device...
[    5.892472] fuse: init (API version 7.34)
[    5.926239] systemd[1]: Starting Journal Service...
[    5.967784] systemd[1]: Starting Load Kernel Modules...
[    6.029569] systemd[1]: Starting Coldplug All udev Devices...
[    6.081649] systemd[1]: Mounted POSIX Message Queue File System.
[    6.107640] systemd[1]: Mounted RPC Pipe File System.
[    6.132092] systemd[1]: Mounted Kernel Debug File System.
[    6.156205] systemd[1]: Mounted Kernel Trace File System.
[    6.181729] systemd[1]: Finished Restore / save the current clock.
[    6.216647] systemd[1]: Finished Create list of static device nodes for the current kernel.
[    6.245505] systemd[1]: Started Journal Service.

Raspbian GNU/Linux 11 raspberrypi ttyAMA0

raspberrypi login:
```

**Additional context**
I initially noticed the issue on LibreELEC master, running kernel 6.0 or 6.1, which showed the same behaviour

popcornmix, 2022-12-07:
Do you think this is a regression or has this issue always been present for you?
If a regression, any feel for whether it came with a bootloader or firmware update?

HiassofT, 2022-12-07:
I think it's a regression in the firmware but I can't tell when it occurred.

I've been running LE 9.2.8 (with ancient 2b76cfc6f57d4943144b9ceb5b57d3d455d6a8fd firmware) and rather recent bootloaders on that RPi and didn't notice any reboot issues with that.

I had been using netboot to test LE10/11 on that RPi and just recently switched to running LE11 from SD card and noticed the issue.

I had been testing with SD cards and recent firmwares/bootloaders on other RPi4s in my office room but they were hooked up to non-4k monitors - and they were fine

timg236, 2022-12-07:
The bootloader change seems good so I think we'll likely merge this since it's easy to verify it via 3V3.

There's possibly a secondary problem in start4.elf if there is no HDMI display connected and core_freq is boosted above 500MHz and force_turbo=1.

HiassofT, 2022-12-07:
@timg236 thanks a lot, 2022-12-07 bootloader fixed the issue for me. I've found another bootloader bug though:

With 2022-04-26 and 2022-12-07 bootloader my scope confirmed that 3V3 is going low for about 50 ms on reboots and I saw the two bootloader init messages on console - 2022-05-20 and later didn't drop 3V3 and resulted in read errors.

However when doing a rpi-eeprom-update from 2022-04-26 to 2022-12-07 3V3 wasn't dropped and I was left with a similar non-boot situation:

```
BMD "vl805.bin" not found
SIG pieeprom.sig a61e9c976fb0d8aa7f900f33badd56af74bf66d047ccbb0ea50e1ff9c0f13d04 1670448824
Reading EEPROM: 524288
Writing EEPROM
+++++++++++.++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++***....+
Verify BOOT EEPROM
Reading EEPROM: 524288
BOOT-EEPROM: UPDATED
rename recovery.bin to RECOVERY.000
RESET

RPi: BOOTLOADER release VERSION:0fc8fc8e DATE: 2022/12/07 TIME: 14:24:15
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1670423055 serial 8eb25f41 boardrev c03111 stc 405164
PM_RSTS: 0x00001020
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Micron' 16Gb x2 total-size: 32 Gbit 3200
DDR 3200 1 0 32 152

XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 1
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
Reset USB port-power 1000 ms
xhci_set_port_power 1 0
xhci_set_port_power 2 0
xhci_set_port_power 3 0
xhci_set_port_power 4 0
xhci_set_port_power 5 0
xhci_set_port_power 1 1
xhci_set_port_power 2 1
xhci_set_port_power 3 1
xhci_set_port_power 4 1
xhci_set_port_power 5 1
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 18
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 19
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
Boot mode: SD (01) order f2
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
OCR c0ff8000 [14]
CID: 00035344534533324780bae4399a0122
CSD: 400e00325b590000edc87f800a404000
SD: bus-width: 4 spec: 2 SCR: 0x02358443 0x00000000
SD HOST: 200000000 CTL0: 0x00800f04 BUS: 50000000 Hz actual: 50000000 HZ div: 4 (2) status: 0x1fff0000 delay: 2
MBR: 0x00002000,  524288 type: 0x0c
MBR: 0x00082000,61801472 type: 0x83
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 516190
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 516190
Read config.txt bytes      109 hnd 0x4c9
SIG pieeprom.sig a61e9c976fb0d8aa7f900f33badd56af74bf66d047ccbb0ea50e1ff9c0f13d04 1670448824
SELF-UPDATE timestamp current 1670448824 new 1670448824 skip
Read start4.elf bytes  2250688 hnd 0xcecb
Read fixup4.dat bytes     5398 hnd 0x15507
0x00c03111 0x00000000 0x00001fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Firmware: 1a79bba5cc84ea52d15809fff557479fdf084529 Nov 11 2022 11:13:01
Starting start4.elf @ 0xfec00200 partition 0
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 8
+

MESS:00:00:04.151035:0: arasan: arasan_emmc_open
MESS:00:00:04.596765:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:04.599514:0: brfs: File read: 109 bytes
MESS:00:00:04.627759:0: brfs: File read: /mfs/sd/lg-55c8.bin
MESS:00:00:04.644432:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:04.648939:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:04.654081:0: brfs: File read: 256 bytes
MESS:00:00:04.665074:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.116198:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:05.123517:0: *** Restart logging
MESS:00:00:05.124922:0: brfs: File read: 109 bytes
MESS:00:00:05.147400:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.152422:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:05.163041:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.168068:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:05.173665:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:05.182430:0: HDMI0: hdmi_pixel_encoding: 600000000
MESS:00:00:05.187900:0: HDMI1: hdmi_pixel_encoding: 162000000
MESS:00:00:05.198179:0: dtb_file 'bcm2711-rpi-4-b.dtb'
MESS:00:00:17.209486:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:17.214696:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:17.322635:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:17.370238:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:17.390421:0: Loaded 'bcm2711-rpi-4-b.dtb' to 0x100 size 0x0
MESS:00:00:17.481460:0: Failed to read command line file 'cmdline.txt'
MESS:00:00:17.586922:0: kernel=
MESS:00:00:17.589254:0: Loaded 'kernel7l.img' to 0x8000 size 0x0
MESS:00:00:17.592687:0: Device tree loaded to 0x2eff2d00 (size 0xd29c)
MESS:00:00:17.600505:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:17.608003:0: uart: Baud rate change done...
MESS:00:00:17.610021:0:
```

With 2022-12-07 bootloader I haven't noticed any issues with 4kp60 yet but it's still puzzling why rebooting eg with 2022-05-20 bootloader (or 2022-11-25 as in my initial report) seem to be working fine...

HiassofT, 2022-12-07:
Just noticed updating from 2022-04-26 to 2022-12-07 wasn't a too good data point for testing 2022-12-07 bootloader but updating 2022-12-07 to 2022-12-07 resulted in the same issue - 3V3 not dropped, read errors.

```
root@raspberrypi:~# rpi-eeprom-update -f pieeprom-2022-12-07.bin 
*** INSTALLING pieeprom-2022-12-07.bin  ***

   CURRENT: Wed 07 Dec 2022 02:24:15 PM UTC (1670423055)
    UPDATE: Wed 07 Dec 2022 02:24:15 PM UTC (1670423055)
    BOOTFS: /boot
Using recovery.bin for EEPROM update

EEPROM updates pending. Please reboot to apply the update.
To cancel a pending update run "sudo rpi-eeprom-update -r".
root@raspberrypi:~# reboot
```

```
[  100.795449] reboot: Restarting system
BMD "vl805.bin" not found
SIG pieeprom.sig a61e9c976fb0d8aa7f900f33badd56af74bf66d047ccbb0ea50e1ff9c0f13d04 1670450713
Reading EEPROM: 524288
Writing EEPROM
...............................................................................................................................+
Verify BOOT EEPROM
Reading EEPROM: 524288
BOOT-EEPROM: UPDATED
rename recovery.bin to RECOVERY.000
RESET

RPi: BOOTLOADER release VERSION:0fc8fc8e DATE: 2022/12/07 TIME: 14:24:15
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1670423055 serial 8eb25f41 boardrev c03111 stc 405202
PM_RSTS: 0x00000020
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Micron' 16Gb x2 total-size: 32 Gbit 3200
DDR 3200 1 0 32 152

XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 1
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
Reset USB port-power 1000 ms
xhci_set_port_power 1 0
xhci_set_port_power 2 0
xhci_set_port_power 3 0
xhci_set_port_power 4 0
xhci_set_port_power 5 0
xhci_set_port_power 1 1
xhci_set_port_power 2 1
xhci_set_port_power 3 1
xhci_set_port_power 4 1
xhci_set_port_power 5 1
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 18
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 19
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
Boot mode: SD (01) order f2
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
OCR c0ff8000 [14]
CID: 00035344534533324780bae4399a0122
CSD: 400e00325b590000edc87f800a404000
SD: bus-width: 4 spec: 2 SCR: 0x02358443 0x00000000
SD HOST: 200000000 CTL0: 0x00800f04 BUS: 50000000 Hz actual: 50000000 HZ div: 4 (2) status: 0x1fff0000 delay: 2
MBR: 0x00002000,  524288 type: 0x0c
MBR: 0x00082000,61801472 type: 0x83
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 516190
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 516190
Read config.txt bytes      109 hnd 0xe
SIG pieeprom.sig a61e9c976fb0d8aa7f900f33badd56af74bf66d047ccbb0ea50e1ff9c0f13d04 1670450713
SELF-UPDATE timestamp current 1670450713 new 1670450713 skip
Read start4.elf bytes  2250688 hnd 0xcecb
Read fixup4.dat bytes     5398 hnd 0x15507
0x00c03111 0x00000000 0x00001fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Firmware: 1a79bba5cc84ea52d15809fff557479fdf084529 Nov 11 2022 11:13:01
Starting start4.elf @ 0xfec00200 partition 0
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 8
+

MESS:00:00:04.146219:0: arasan: arasan_emmc_open
MESS:00:00:04.592569:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:04.595304:0: brfs: File read: 109 bytes
MESS:00:00:04.623689:0: brfs: File read: /mfs/sd/lg-55c8.bin
MESS:00:00:04.640362:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:04.644869:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:04.650011:0: brfs: File read: 256 bytes
MESS:00:00:04.661105:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.112523:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:05.119842:0: *** Restart logging
MESS:00:00:05.121246:0: brfs: File read: 109 bytes
MESS:00:00:05.143828:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.148849:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:05.159469:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.164496:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:05.170094:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:05.178858:0: HDMI0: hdmi_pixel_encoding: 600000000
MESS:00:00:05.184328:0: HDMI1: hdmi_pixel_encoding: 162000000
MESS:00:00:05.194591:0: dtb_file 'bcm2711-rpi-4-b.dtb'
MESS:00:00:17.205680:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:17.210890:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:17.318829:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
MESS:00:00:17.367045:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_WRITE_PROTECTED not defined
MESS:00:00:17.387231:0: Loaded 'bcm2711-rpi-4-b.dtb' to 0x100 size 0x0
MESS:00:00:17.478316:0: Failed to read command line file 'cmdline.txt'
MESS:00:00:17.583818:0: kernel=
MESS:00:00:17.586153:0: Loaded 'kernel7l.img' to 0x8000 size 0x0
MESS:00:00:17.589583:0: Device tree loaded to 0x2eff2d00 (size 0xd29c)
MESS:00:00:17.597402:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:17.604900:0: uart: Baud rate change done...
MESS:00:00:17.606917:0:
```

timg236, 2022-12-07:
Please could you check the version of recovery.bin installed to /boot before applying the update (strings /boot/recovery.bin | grep VERSION)
It should be 0fc8fc8e i.e. same as the EEPROM release since it will be recovery.bin and not the bootloader switching 3V3 during the update.

HiassofT, 2022-12-07:
Ah, sorry, my fault. Of course I was using old recovery.bin (from apt update a few hours ago, 14.0-1).

apt just got me rpi-eeprom 15.0-1 with new recovery.bin and with that I see 3V3 is being dropped on eeprom update and updating 2022-12-07 to 2022-12-07 succeeded (I was lazy and just used rpi-eeprom-config -e).

I'll do a few more tests the next days but so far it's looing good here now.

The 4kp60 (probably core_freq?) thing is really puzzling.... You folks with access to firmware source might have a better chance than me finding out why it seems to have triggered the failure.

Anyways: though my initial issue seems to be solved now and I couldn't notice other issue I guess it might be good to keep this issue open for a week or so in case other people get hit by it (no one looks at closed issues...)

Feel free to close it any time in case I forget though - I'll reopen or create another issue if/when I have more info or an actual failure to report.

timg236, 2022-12-08:
Thanks for confirming the recovery.bin thing. Let’s keep this issue open until we understand the clock issue.
Might take a few days to find the root cause!

timg236, 2022-12-08:
I'm still investigating the root-cause but I've been testing a pending change to switch the firmware to the newer and more reliable SDHCI controller as used by Linux (and now the bootloader).

With this firmware I can no longer reproduces the 4kp60 issues (connected or not).

Originally, this change was added in order to see if it helped with boot time, since that wasn't the case it was parked because of the risk of regressions. However, the experience with the bootloader is convincing me that this is is actually more robust (soak testing on various board revisions will be required though!

Would you mind giving it a try? I can build the other start*elf variants if needed.

[rpi-fw-new-sdhci.zip](https://github.com/raspberrypi/firmware/files/10185527/rpi-fw-new-sdhci.zip)

HiassofT, 2022-12-08:
@timg236 sure, I'll do some tests with it later.

Could you also build the 4x variant? That's the one we use in LE by default

timg236, 2022-12-08:
No problem, here's the slightly more official releases from the CI system

[rpi-fw-new-sdhci-all-variants.zip](https://github.com/raspberrypi/firmware/files/10186684/rpi-fw-new-sdhci-all-variants.zip)

HiassofT, 2022-12-11:
Testing with a 32GB Samsung Evo+ card and an 8GB rev 1.4 RPi4 looks fine here as well

timg236, 2022-12-12:
I left some boards in a reboot loop over the weekend and they were still running so this has been pushed to rpi-update master by @popcornmix. Not much else has changed but please shout if you encounter any issues in this area.

HiassofT, 2022-12-12:
Thanks a lot to everyone involved in fixing this issue!

Latest rpi-firmware works fine here so I'm closing this

closed 2022-12-12 by HiassofT

capiman, 2022-12-17:
@timg236 Can you tell if this (SD card problem) also affect a compute module 4?

timg236, 2022-12-18:
@capiman The 3V3 switch issue resolved by the bootloader update would not have affected CM4 because all CM4s have an SD power switch. 

The firmware change for headless boot with 4kp60 enabled (or overclocked core freq) would be relevant for CM4 Lite + SD. I didn't see it fail with CM4 + EMMC though.