raspberrypi/firmware #1764: actually using this

LinuxGamernotexe, 2022-12-14:
Right so to use this do i just copy all the files to an sd card and insert it into a pi
don't wanna break something

popcornmix, 2022-12-14:
In general no.
You should have a suitable kernel and gpu firmware on whatever distribution you are using (RPiOS is the most common one).

You will periodically gets updates using your distribution's update mechanism (RPiOS uses apt).

Updates in here may be newer, but should be treated as testing rather than stable.

If you have a specific need to update, then rpi-update is the common tool for getting these updates.

But typically you should only update if you want to contribute to the testing effort, or have been advised to update by an engineer to fix a specific issue.

Having a backup system in place is always advisable.

LinuxGamernotexe, 2023-01-06:
my computers imager is not working and if i try to download any others they don't work so you got any ideas of how else i can install raspi os?

JamesH65, 2023-01-06:
For technical support on using imager, please use our forums. Lots of help is available there. Imager should work on Windows, Mac and Linux machines.

LinuxGamernotexe, 2023-01-06:
it should work but it doesn't idk why i have tried everything it just refuses to restore a disk image

JamesH65, 2023-01-06:
forums.raspberrypi.com is the place for tech help.

LinuxGamernotexe, 2023-01-06:
Ok Thanks for the help

closed 2023-01-06 by LinuxGamernotexe