raspberrypi/firmware #1765: Regression post 1.20220331: blk_update_request: I/O error, dev sr0

macmpi, 2022-12-16:
Reading Audio CD seems to fail with recent kernel and post 1.20220331 firmware (on PiZeroW)

All this started with failing cdda2wav on a new system install (Alpine 3.17-based):
https://pastebin.com/EF9rsQ24

**Note last line:** `Driver and/or firmware bug detected! Drive cannot play the very last sector (248604)!`

On the very same hardware (same PiZeroW, CD-ROM drive, AudioCD, SD card, Power Supply,....), I installed a several base Alpine system (no bells & whistles) and looked at `dmesg`

Working combo: Alpine 3.16 (kernel 5.15.78, firmware 1.20220331):
https://pastebin.com/JDUHDJTy

Failing combo: Alpine 3.17 (kernel 5.15.79, firmware 6cbf00359959cf7381f4e3773037c7d5573d94b2):
https://pastebin.com/B8HqN0Pv

I  then copied Alpine 3.17 firmware onto initially working 3.16 and got it failing too:
https://pastebin.com/U606fAqx

Finally the opposite test (Alpine 3.17 with  firmware 1.20220331), which seems OK:
https://pastebin.com/srsYgPEX

With that, it very much seem recent firmware changes (bootcode.bin, *.dat, *.elf) post 1.20220331 are colliding with CD driver.
Thoughts?

macmpi, 2022-12-21:
Lucky `cdda2wav` gives more clues than `cdparanoia` or `mpd` which just get blocked by struggling IO.
@popcornmix any hint by chance?

popcornmix, 2022-12-21:
It seem unlikely that start.elf firmware could cause an issue like this in the kernel driver (but it's conceivable that some prior setup has had an effect).

Best way to confirm this for sure is to manually bisect the firmware.
I think you have said this firmware:
```[    0.130639] raspberrypi-firmware soc:firmware: Attached to firmware from 2022-03-24T13:20:54, variant start```
is good, and this is bad:
```[    0.130616] raspberrypi-firmware soc:firmware: Attached to firmware from 2022-11-11T11:14:37, variant start```

If you look [here](https://github.com/raspberrypi/firmware/commits/master/boot/start.elf) you can see  ~25 firmwares between these two points. Bisecting them should take about 5 steps.

For each commit you'll want to download start.elf and fixup.dat and replace the ones on your sdcard.
The download addresses will look like:
```https://github.com/raspberrypi/firmware/raw/ea4c803c57697aad951a27fcd64c01ef16f9fe7f/boot/start.elf```
```https://github.com/raspberrypi/firmware/raw/ea4c803c57697aad951a27fcd64c01ef16f9fe7f/boot/fixup.dat```

This url is for the March 24 firmware (which should work), but replace the hash (ea4c803c57697aad951a27fcd64c01ef16f9fe7f/) with each of the bisect points. 
You are expecting March 24 firmware to work, and Nov 11 firmware to fail. Report the first version that has the failure.

macmpi, 2022-12-21:
Thanks for feedback. I have indeed started bisecting and it failed in the first few changes following 1.20220331.
I'll report exactly which, as soon as I'm in reach with that setup again.

macmpi, 2022-12-26:
I can confirm things start to fail after 22/04/13 commit https://github.com/raspberrypi/firmware/commit/afc7631bc9a58f94a157503955881ab27a6287d0 (`firmware: arm_loader: Support longer file paths`)
It's actually first firmware files change following 1.20220331 if I'm correct.
Would that ring a bell? (seems quite unrelated at first sight...)
`dmesg` is not happy (below) and `cdda2wav` complains as reported earlier (`cdparanoia` also fails)
```
[   21.083740] sr 0:0:0:0: Power-on or device reset occurred
[   21.086307] sr 0:0:0:0: [sr0] scsi3-mmc drive: 24x/24x writer cd/rw xa/form2 cdda tray
[   21.086346] cdrom: Uniform CD-ROM driver Revision: 3.20
[   21.621214] sr 0:0:0:0: Attached scsi CD-ROM sr0
[   22.278091] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   22.281678] sr 0:0:0:0: [sr0] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=DRIVER_OK cmd_age=0s
[   22.281722] sr 0:0:0:0: [sr0] tag#0 Sense Key : 0x5 [current] 
[   22.281741] sr 0:0:0:0: [sr0] tag#0 ASC=0x64 ASCQ=0x0 
[   22.281765] sr 0:0:0:0: [sr0] tag#0 CDB: opcode=0x28 28 00 00 03 ca e0 00 00 02 00
[   22.281781] blk_update_request: I/O error, dev sr0, sector 994176 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
```
*Note*: kernel (and related dtbs & al) remain 5.15.79 in this test, just rolling back .elf/.dat).
Happy to provide more detailed info if needed.

popcornmix, 2022-12-26:
That commit seems surprising. Are you sure that is the first firmware update that has the issue?
i.e. the previous firmware "firmware: arm_loader_dvfs: Make arm only see its own boosts" is good?

macmpi, 2022-12-26:
Looking at commit messages it's the first with commit message referring to firmware change, following 1.20220331 which is "known-good".
The one you are referring to ea4c803c57697aad951a27fcd64c01ef16f9fe7f on March 24th seems to predates 1.20220331, right?

macmpi, 2022-12-28:
> That commit seems surprising. Are you sure that is the first firmware update that has the issue?

Yes I double-checked: issue does come-up with afc7631bc9a58f94a157503955881ab27a6287d0
`dmesg` warning are here earlier (maybe normal as it's Audio CD which won't mount per say), but `cdda2wav`/`cdparanoia` issues do start with that firmware.
Thanks for any tip.

macmpi, 2022-12-30:
It looks as if DTBs are not selected the same way after that https://github.com/raspberrypi/firmware/commit/afc7631bc9a58f94a157503955881ab27a6287d0 change.

Indeed by defaults `/mmcblk0p1` contains both downstream `bcm27*.dtb` and upstream `bcm28*.dtb`
When using firmware from https://github.com/raspberrypi/firmware/commit/afc7631bc9a58f94a157503955881ab27a6287d0 and on, if I remove/move away `bcm28*.dtb` then the issue is fixed!...
So after that change it seems upstream DTBs are selected if available (and BTW those may have a problem): it was seemingly not the case before.

How can I check which DTB file is selected on a running system?

For records, defauls Alpine `config.txt` is as follows, with all DTBs in `/mmcblk0p1`:
```
[pi0]
kernel=boot/vmlinuz-rpi
initramfs boot/initramfs-rpi
[pi0w]
kernel=boot/vmlinuz-rpi
initramfs boot/initramfs-rpi
[pi1]
kernel=boot/vmlinuz-rpi
initramfs boot/initramfs-rpi
[pi02]
kernel=boot/vmlinuz-rpi2
initramfs boot/initramfs-rpi2
[pi2]
kernel=boot/vmlinuz-rpi2
initramfs boot/initramfs-rpi2
[pi3]
kernel=boot/vmlinuz-rpi2
initramfs boot/initramfs-rpi2
[pi3+]
kernel=boot/vmlinuz-rpi2
initramfs boot/initramfs-rpi2
[all]
include usercfg.txt
```

macmpi, 2022-12-30:
@pelwell any thought as it seemingly relates to https://github.com/raspberrypi/firmware/issues/1720 ?

popcornmix, 2022-12-30:
> How can I check which DTB file is selected on a running system?

```sudo vcdbg log msg```

should include the dtb files loaded.

macmpi, 2022-12-30:
hum, unfortunately it does not seem to be built on Alpine, while `vcgencmd` is.
I'll need to find-out how to build it then...

pelwell, 2022-12-30:
You can use the vclog utility instead: https://github.com/raspberrypi/utils/tree/master/vclog

pelwell, 2022-12-30:
Do you have an `os_prefix` set in config.txt? The older, smaller filename limit was sufficient for the known DTB files unless they had been moved into a subdirectory. And what about `upstream_kernel`?

macmpi, 2022-12-30:
> You can use the vclog utility instead: https://github.com/raspberrypi/utils/tree/master/vclog

Not available either. [Package](https://git.alpinelinux.org/aports/tree/main/raspberrypi/APKBUILD) is built with `-DCMAKE_BUILD_TYPE=MinSizeRel`. Any hint on proper option to get all the missing binaries?

>Do you have an os_prefix set in config.txt? The older, smaller filename limit was sufficient for the known DTB files unless they had been moved into a subdirectory. And what about upstream_kernel?

No, none are set, as shown in copied [config.txt](https://github.com/raspberrypi/firmware/issues/1765#issuecomment-1368044827) (DTBs are in root of SD aside firmware files and overlay directory, not in boot/ that has initramfs and vmlinuz)
```
ls /media/mmcblk0p1/
bcm2710-rpi-zero-2.dtb    bcm2837-rpi-3-a-plus.dtb
bcm2708-rpi-b-plus.dtb    bcm2711-rpi-4-b.dtb       bcm2837-rpi-3-b-plus.dtb
bcm2708-rpi-b-rev1.dtb    bcm2711-rpi-400.dtb       bcm2837-rpi-3-b.dtb
bcm2708-rpi-b.dtb         bcm2711-rpi-cm4.dtb       bcm2837-rpi-cm3-io3.dtb
bcm2708-rpi-cm.dtb        bcm2711-rpi-cm4s.dtb      boot
bcm2708-rpi-zero-w.dtb    bcm2835-rpi-a-plus.dtb    bootcode.bin
bcm2708-rpi-zero.dtb      bcm2835-rpi-a.dtb 
bcm2709-rpi-2-b.dtb       bcm2835-rpi-b-plus.dtb    cmdline.txt
bcm2709-rpi-cm2.dtb       bcm2835-rpi-b-rev2.dtb    config.txt
bcm2710-rpi-2-b.dtb       bcm2835-rpi-b.dtb         fixup.dat
bcm2710-rpi-3-b-plus.dtb  bcm2835-rpi-cm1-io1.dtb
bcm2710-rpi-3-b.dtb       bcm2835-rpi-zero-w.dtb    overlays
bcm2710-rpi-cm3.dtb       bcm2835-rpi-zero.dtb      start.elf
bcm2710-rpi-zero-2-w.dtb  bcm2836-rpi-2-b.dtb
```

popcornmix, 2022-12-31:
Just clone the git repo and follow build instructions in README. It only takes a minute to build and has minimal dependencies.

macmpi, 2022-12-31:
Thanks @popcornmix  will try to do so.
Meanwhile, the tests & results so far show that:
- when `bcm2708-rpi-zero-w.dtb`  is the only relevant DTB available, AudioCD programs work as expected with firmware after https://github.com/raspberrypi/firmware/commit/afc7631bc9a58f94a157503955881ab27a6287d0
- if upstream `bcm2835-rpi-zero-w.dtb` file **is also available** then AudioCD programs fail with those later firmware, while they worked with earlier firmware.
- [`config.txt`](https://github.com/raspberrypi/firmware/issues/1765#issuecomment-1368044827) has no `upstream_kernel` / `os_prefix` / `upstream_kernel` statement, so only downstream DTBs in root FAT filesystem should be selected(even if there are upstream files next to them).

Can't we conclude https://github.com/raspberrypi/firmware/commit/afc7631bc9a58f94a157503955881ab27a6287d0 change somehow broke default downstream DTBs selection in root FAT filesystem, with no further tests?

Let me know if/which further info is needed to help for a fix.

popcornmix, 2022-12-31:
> Can't we conclude https://github.com/raspberrypi/firmware/commit/afc7631bc9a58f94a157503955881ab27a6287d0 change somehow broke default downstream DTBs selection in root FAT filesystem, with no further tests?

Based on the evidence you've provided that sounds likely. The commit mentioned ("Support longer file paths") doesn't look blatantly wrong, but it perhaps has caused some unintended change in behaviour.

I'm sure once we've reproduced this, the fix will be straightforward. But most of us supporting here are away with family for the holiday period, so we can ask questions to narrow down the issue, but actually stepping through the code with a debugger is unlikely to happen before the office opens next week.

It's possible the output of vclog will make it clear how it's failing.

macmpi, 2022-12-31:
Of course, no hurry: enjoy a great & refreshing family break !
Let's recap next week.

macmpi, 2023-01-04:
For records, few pending issues compiling `vclog` as described there: https://github.com/raspberrypi/utils/issues/14 , https://github.com/raspberrypi/utils/pull/15

macmpi, 2023-01-04:
Tried again (twice) after a cold reboot (poweroff + unplugging for 15mn+) with "working" firmware (1.20220331):
```
# vclog --msg
Could not read from Device Tree log starts and log size
```
Doas `vclog` require later firmware ?

timg236, 2023-01-04:
vclog requires a recent APT (or rpi update stable) release of the firmware in order to discover the location of the log buffer via device tree

timg236, 2023-01-04:
uart_2ndstage=1 will also emit all of the dt loading message up till the kernel is started via the uart

popcornmix, 2023-01-05:
You may be able to use a static build of vcdbg. See https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/67#issuecomment-653209729

popcornmix, 2023-01-05:
The two vcdbg logs are identical (ignoring the timestamps).

> I even have some successes with https://github.com/raspberrypi/firmware/commit/afc7631bc9a58f94a157503955881ab27a6287d0 (though it fails frequently).

Are you sure the preceding commit is 100% working?
When failures are intermittent, it's easy to think a version is good, when you get a few lucky successes.

I could produce some additional firmware builds in between the two versions released just to be sure it is the (internal tree) commit we are assuming, but that only works if you can reliably say whether a build has the issue or not.

macmpi, 2023-01-05:
It seems system do not always properly recover from eventual:
`sr 0:0:0:0: Power-on or device reset occurred`

This often occur after CD totally spins down, and `cdda2wav` or `cdparanoia` are then called.
CD spins-up (may draw too much current initially?) and then reads may be affected (apps become very slow to respond but may go through).
Subsequent CD app launch are very slow and it seems it would stay so until a cold reboot...
It may eventually go doggy with `cdparanoia` blocking IO ops to death for example.
```
[ 1825.802613] Indeed it is in host mode hprt0 = 00001101
[ 1826.012513] usb 1-1: reset high-speed USB device number 2 using dwc_otg
```
If those apps are called with CD already spinning (and no previous lockup), then they operate fine...

Could root cause be some [power limitation](https://raw.githubusercontent.com/wiki/macmpi/tiny-radioCD/images/openview.jpg) (though `vcgencmd get_throttled` reports `throttled=0x0`) somehow altering CD USB interface/driver/firmware food-chain, it not being able to recover/flush the situation in some cases?...

Any known command that would just spin-up CD (without reading) I could call prior testing access with `cdda2wav` or `cdparanoia`  ?

macmpi, 2023-01-06:
Further refined testing pattern trying to be more immune to transient effects.
- swapped PSU to similarly rated 2.4A and [known to be very stable & low noise](https://lygte-info.dk/review/USBpower%20Anker%2040W%205-port%2071AN7105%20UK.html)
- unplug  & let setup rest for about 20mn between each test suite where firmware files are changed (let caps discharge & try avoid memory effects).
- run `cdda2wav -D /dev/cdrom --info-only` command several time, either with CD spinning or letting it spin down for a couple of minutes

With that, apparently no more  `sr 0:0:0:0: Power-on or device reset occurred` in `dmesg` (apart initial one at about 20sec during system init), and no more `cdda2wav` errors.
Tested with 1.20220331 stable, https://github.com/raspberrypi/firmware/commit/afc7631bc9a58f94a157503955881ab27a6287d0 and https://github.com/raspberrypi/firmware/commit/6cbf00359959cf7381f4e3773037c7d5573d94b2

So, I'd pretend some transient power instabilities may affect CD interface board, and/or USB communication with Pi device _under some circumstances_, while Pi device itself does not seem affected (`throttled=0x0`) by such power instabilities (USB port wiring is indeed in parallel in my [PiZeroWed-Mac Mini](https://raw.githubusercontent.com/wiki/macmpi/tiny-radioCD/images/openview.jpg)).
Now, whether or not firmware changes may marginally influence such power sensitivity on USB port/built-in hub, I don't know...

I'll keep monitoring this and eventually close it if all shame is to be put on PSU  (though was happy with the original 2.4A so far...).
>I could produce some additional firmware builds

Thanks for the kind offer, but I'm reluctant to consume more of your time on this, unless you think some of the firmware changes may adversely affect USB communication.
Thanks again for your great support.

macmpi, 2023-01-06:
interface details for records (bulk)
```
# lsusb -v

Bus 001 Device 002: ID 14cd:6600 Super Top  USB 2.0  IDE DEVICE    
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x14cd 
  idProduct          0x6600 
  bcdDevice            2.01
  iManufacturer           1 Super Top 
  iProduct                3 USB 2.0  IDE DEVICE    
  iSerial                 2 ††††††††††
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 
      bInterfaceSubClass      6 
      bInterfaceProtocol     80 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0001
  Self Powered

Bus 001 Device 001: ID 1d6b:0002 Linux 5.15.79-0-rpi dwc_otg_hcd DWC OTG Controller
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 
  bDeviceSubClass         0 
  bDeviceProtocol         1 
  bMaxPacketSize0        64
  idVendor           0x1d6b 
  idProduct          0x0002 
  bcdDevice            5.15
  iManufacturer           3 Linux 5.15.79-0-rpi dwc_otg_hcd
  iProduct                2 DWC OTG Controller
  iSerial                 1 20980000.usb
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0019
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0008
    Ganged power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood        1 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
Device Status:     0x0001
  Self Powered
```

macmpi, 2023-01-06:
is PiZeroW QC3.0 compatible on its OTG port (when host)?
i.e would it be able to reclaim 5v/3A from a QC3.0 charger for instance?