raspberrypi/firmware #1766: Boot process hangs during start.elf when MT7921K is connected

jolla, 2022-12-21:
**Describe the bug**
Boot process intermittantly hangs during (or just after) start.elf stage when Mediatek MT7921K (Wi-Fi 6E module) is connected

**To reproduce**
Install MT7921K Wi-Fi module and power on or reboot system

**Expected behaviour**
start.elf process should complete and linux kernel should start booting

**Actual behaviour**
System hangs during (or just after) start.elf process (see log output below)

**System**
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
CM4
* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2022-12-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9ecb50a577fa3772440cd37b7dbff5f327a6191f, wlanpi1

* Which firmware version (`vcgencmd version`)?
Oct 26 2022 11:09:36 
Copyright (c) 2012 Broadcom
version c72ad6b26ff40c91ef776b847436094ee63fabee (clean) (release) (start_db)

* Which kernel version (`uname -a`)?
Linux wlanpi-627 5.19.17-v8-wlanpi+ #1 SMP PREEMPT Sun Dec 18 21:51:50 UTC 2022 aarch64 GNU/Linux

**Logs**
If applicable, add the relevant output from `dmesg` or similar.
This is where it hangs during the boot process

...
Starting start4db.elf @ 0xfe200200 partition 0
+

ASRT:00:00:03.091404:0: ../../../../../vcfw/drivers/chip/vciv/2708/clock_2711.c::clock_wait_busy!2151: timeout != 0
MESS:00:00:03.189266:0: arasan: arasan_emmc_open
MESS:00:00:03.350188:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:03.353609:0: brfs: File read: 1714 bytes
MESS:00:00:03.378635:0: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:03.383139:0: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:03.400113:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:03.404620:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:03.411391:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:03.415907:0: gpioman: gpioman_get_pin_num: pin DISPLAY_SDA not defined
MESS:00:00:03.422201:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:03.919527:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:03.925335:0: *** Restart logging
MESS:00:00:03.927817:0: brfs: File read: 1714 bytes
MESS:00:00:03.937892:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:03.942914:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:03.953539:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:03.958568:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:03.964167:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:03.977959:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:03.982989:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:03.993612:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:03.998634:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:04.004232:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:04.012996:0: HDMI0: hdmi_pixel_encoding: 300000000
MESS:00:00:04.018464:0: HDMI1: hdmi_pixel_encoding: 300000000
MESS:00:00:04.024303:0: gpioman: gpioman_get_pin_num: pin CAMERA_0_I2C_PORT not defined
MESS:00:00:04.035930:0: dtb_file 'bcm2711-rpi-cm4.dtb'
MESS:00:00:04.044017:0: brfs: File read: /mfs/sd/bcm2711-rpi-cm4.dtb
MESS:00:00:04.047265:0: Loaded 'bcm2711-rpi-cm4.dtb' to 0x100 size 0xcf46
MESS:00:00:04.067115:0: brfs: File read: 53062 bytes
MESS:00:00:04.086101:0: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
MESS:00:00:04.166255:0: brfs: File read: 2259 bytes
MESS:00:00:04.170965:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:04.173646:0: brfs: File read: 1714 bytes
MESS:00:00:04.189733:0: brfs: File read: /mfs/sd/overlays/pcie-32bit-dma.dtbo
MESS:00:00:04.200366:0: Loaded overlay 'pcie-32bit-dma'
MESS:00:00:04.212859:0: brfs: File read: 438 bytes
MESS:00:00:04.220786:0: brfs: File read: /mfs/sd/overlays/dwc2.dtbo
MESS:00:00:04.229998:0: Loaded overlay 'dwc2'
MESS:00:00:04.231241:0: dtparam: dr_mode=host
MESS:00:00:04.246294:0: brfs: File read: 801 bytes
MESS:00:00:04.253125:0: brfs: File read: /mfs/sd/overlays/disable-wifi.dtbo
MESS:00:00:04.265530:0: Loaded overlay 'disable-wifi'
MESS:00:00:04.281089:0: brfs: File read: 387 bytes
MESS:00:00:04.288422:0: brfs: File read: /mfs/sd/overlays/disable-bt.dtbo
MESS:00:00:04.307102:0: Loaded overlay 'disable-bt'
MESS:00:00:04.342558:0: brfs: File read: 1073 bytes
MESS:00:00:04.364216:0: brfs: File read: /mfs/sd/overlays/vc4-fkms-v3d-pi4.dtbo
MESS:00:00:04.386059:0: Loaded overlay 'vc4-fkms-v3d'
MESS:00:00:04.423744:0: brfs: File read: 1446 bytes
MESS:00:00:04.430661:0: brfs: File read: /mfs/sd/overlays/disable-wifi.dtbo
MESS:00:00:04.443067:0: Loaded overlay 'disable-wifi'
MESS:00:00:04.458637:0: brfs: File read: 387 bytes
MESS:00:00:04.466197:0: brfs: File read: /mfs/sd/overlays/dwc2.dtbo
MESS:00:00:04.475479:0: Loaded overlay 'dwc2'
MESS:00:00:04.476719:0: dtparam: dr_mode=otg
MESS:00:00:04.480925:0: dtparam: i2c_arm=on
MESS:00:00:04.493247:0: dtparam: spi=on
MESS:00:00:04.502124:0: dtparam: act_led_trigger=heartbeat
MESS:00:00:04.542296:0: brfs: File read: 801 bytes
MESS:00:00:04.546354:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:04.549381:0: Read command line from file 'cmdline.txt':
MESS:00:00:04.555259:0: 'console=serial0,115200 console=tty1 root=PARTUUID=81c2a5e7-02 rootfstype=ext4 fsck.repair=yes rootwait'
MESS:00:00:04.663646:0: kernel=
MESS:00:00:04.663692:0: brfs: File read: 103 bytes
MESS:00:00:05.467888:0: brfs: File read: /mfs/sd/kernel8.img
MESS:00:00:05.470441:0: Loaded 'kernel8.img' to 0x300000 size 0x7d15db
MESS:00:00:06.643298:0: Kernel relocated to 0x200000
MESS:00:00:06.645150:0: Device tree loaded to 0x2eff2a00 (size 0xd533)
MESS:00:00:06.654233:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:06.660566:0: uart: Baud rate change done...
MESS:00:00:06.662625:0: uart: Baud rate change done...
MESS:00:00:06.671792:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined

jolla, 2022-12-21:
After a successful reboot I can trigger a kernel panic by disabling and enabling PCIe via an overlay, it sometimes requires disabling and enabling PCIe several times, but eventually I can trigger a kernel panic. Here is the output.

[   69.005037] SError Interrupt on CPU0, code 0x00000000bf000002 -- SError
[   69.005051] CPU: 0 PID: 3491 Comm: dtoverlay Tainted: G         C        5.19.17-v8-wlanpi+ #1
[   69.005057] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[   69.005060] pstate: 200000c5 (nzCv daIF -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   69.005065] pc : pci_generic_config_read+0x44/0xe0
[   69.005080] lr : pci_generic_config_read+0x2c/0xe0
[   69.005085] sp : ffffffc0092c33c0
[   69.005087] x29: ffffffc0092c33c0 x28: ffffff8003ccb800 x27: 0000000000000001
[   69.005096] x26: 0000000000000001 x25: ffffffd9536299b0 x24: 0000000000000000
[   69.005102] x23: ffffffc0092c34f4 x22: ffffff8003ccb400 x21: 0000000000000087
[   69.005108] x20: 0000000000000004 x19: ffffffc0092c3434 x18: 0000000000000000
[   69.005113] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
[   69.005118] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
[   69.005124] x11: 0000000000000000 x10: 0000000000001a80 x9 : ffffffd9526c09ec
[   69.005129] x8 : 0000000080b5111d x7 : 000000000000ea60 x6 : ffffffc0092c34f4
[   69.005135] x5 : ffffffc009280000 x4 : ffffff8003cc8f40 x3 : ffffffc009289000
[   69.005140] x2 : 0000000000008000 x1 : 00000000deaddead x0 : ffffffc009288000
[   69.005147] Kernel panic - not syncing: Asynchronous SError Interrupt
[   69.005151] CPU: 0 PID: 3491 Comm: dtoverlay Tainted: G         C        5.19.17-v8-wlanpi+ #1
[   69.005155] Hardware name: Raspberry Pi Compute Module 4 Rev 1.0 (DT)
[   69.005158] Call trace:
[   69.005159]  dump_backtrace+0xe8/0xf8
[   69.005165]  show_stack+0x20/0x58
[   69.005168]  dump_stack_lvl+0x8c/0xb8
[   69.005175]  dump_stack+0x18/0x34
[   69.005179]  panic+0x184/0x37c
[   69.005183]  add_taint+0x0/0xb0
[   69.005188]  arm64_serror_panic+0x6c/0x80
[   69.005192]  do_serror+0x3c/0x88
[   69.005194]  el1h_64_error_handler+0x38/0x50
[   69.005199]  el1h_64_error+0x64/0x68
[   69.005203]  pci_generic_config_read+0x44/0xe0
[   69.005208]  pci_bus_read_config_dword+0x88/0xf0
[   69.005212]  pci_bus_generic_read_dev_vendor_id+0x3c/0x1b8
[   69.005217]  pci_bus_read_dev_vendor_id+0x54/0x78
[   69.005220]  pci_scan_single_device+0x88/0xf8
[   69.005223]  pci_scan_slot+0x48/0x128
[   69.005227]  pci_scan_child_bus_extend+0x5c/0x2c0
[   69.005230]  pci_scan_child_bus+0x1c/0x28
[   69.005234]  pci_scan_bridge_extend+0x188/0x598
[   69.005237]  pci_scan_child_bus_extend+0x15c/0x2c0
[   69.005240]  pci_scan_root_bus_bridge+0x6c/0xe0
[   69.005244]  pci_host_probe+0x20/0xd0
[   69.005247]  brcm_pcie_probe+0x26c/0x5b8
[   69.005251]  platform_probe+0x70/0xe0
[   69.005257]  really_probe+0x124/0x2c8
[   69.005260]  __driver_probe_device+0x80/0xe8
[   69.005263]  driver_probe_device+0x44/0xf8
[   69.005266]  __device_attach_driver+0xc0/0x100
[   69.005269]  bus_for_each_drv+0x80/0xd8
[   69.005275]  __device_attach+0x104/0x198
[   69.005278]  device_initial_probe+0x1c/0x28
[   69.005281]  bus_probe_device+0xa4/0xb0
[   69.005286]  device_add+0x390/0x830
[   69.005290]  of_device_add+0x5c/0x70
[   69.005295]  of_platform_device_create_pdata+0x98/0x100
[   69.005299]  of_platform_notify+0xec/0x180
[   69.005302]  blocking_notifier_call_chain+0x74/0xa8
[   69.005308]  of_property_notify+0x68/0xb0
[   69.005313]  __of_changeset_entry_notify+0xb8/0x118
[   69.005318]  __of_changeset_revert_notify+0x4c/0x80
[   69.005323]  of_overlay_remove+0x1c0/0x2b8
[   69.005327]  cfs_overlay_release+0x34/0x60
[   69.005332]  config_item_release+0x64/0x138
[   69.005338]  config_item_put+0x70/0x120
[   69.005342]  configfs_rmdir+0x1f0/0x2d0
[   69.005346]  vfs_rmdir+0x100/0x210
[   69.005350]  do_rmdir+0x194/0x1c0
[   69.005354]  __arm64_sys_unlinkat+0x70/0x90
[   69.005358]  invoke_syscall+0x4c/0x110
[   69.005364]  el0_svc_common.constprop.0+0x68/0x128
[   69.005369]  do_el0_svc+0x34/0xc0
[   69.005374]  el0_svc+0x30/0x98
[   69.005377]  el0t_64_sync_handler+0xb8/0xc0
[   69.005380]  el0t_64_sync+0x18c/0x190
[   69.005385] SMP: stopping secondary CPUs
[   69.005393] Kernel Offset: 0x194a000000 from 0xffffffc008000000
[   69.005396] PHYS_OFFSET: 0x0
[   69.005398] CPU features: 0x2000,04027810,00001086
[   69.005401] Memory Limit: none
[   69.379250] ---[ end Kernel panic - not syncing: Asynchronous SError Interrupt ]---

pelwell, 2022-12-21:
I don't think enabling PCie at runtime is supported. I wouldn't be surprised (can't check right now) if the firmware has to turn on some clocks etc.

jolla, 2022-12-22:
If PCIe is disabled at boot by loading the following overlay, then the system boots fine, however with no PCIe...

```
/dts-v1/;

/ {
	compatible = "brcm,bcm2711-pcie";

	fragment@0 {
		target = <0xffffffff>;

		__overlay__ {
			status = "disabled";
		};
	};

	__fixups__ {
		pcie0 = "/fragment@0:target:0";
	};
};
```

jolla, 2022-12-22:
Adding dtparm=pcie=off to boot/config.txt I was able to disable PCIe and get the system to reliably boot without hanging, however after the system is booted if I enable PCIe with the MT7921K Wi-Fi module inserted sometimes I get a kernel panic, I'm assuming this is the same reason the system hangs trying to load the kernel. Any ideas how to debug what is causing this?

 recover4.elf not found (6)
recovery.elf not found (6)
Read start4db.elf bytes  3746856 hnd 0x000163a2 hash '06629a6b6aeaa30e'
Read fixup4db.dat bytes     8382 hnd 0x0000040e hash 'd93f7d3e8ffe6eb8'
0x00a03140 0x00000000 0x00001fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4db.elf @ 0xfec00200 partition 0

ASRT:00:00:05.120497:0: ../../../../../vcfw/drivers/chip/vciv/2708/clock_2711.c::clock_wait_busy!2151: timeout != 0
MESS:00:00:05.217023:0: arasan: arasan_emmc_open
MESS:00:00:05.379322:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.382697:0: brfs: File read: 1577 bytes
MESS:00:00:05.402809:0: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.407314:0: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:05.424363:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.428870:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:05.435576:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.440105:0: gpioman: gpioman_get_pin_num: pin DISPLAY_SDA not defined
MESS:00:00:05.446450:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:05.949045:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:05.954851:0: *** Restart logging
MESS:00:00:05.957333:0: brfs: File read: 1577 bytes
MESS:00:00:05.967394:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.972416:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:05.983042:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.988071:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:05.993669:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:06.007461:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:06.012491:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:06.023114:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:06.028135:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:06.033733:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:06.042498:0: HDMI0: hdmi_pixel_encoding: 300000000
MESS:00:00:06.047966:0: HDMI1: hdmi_pixel_encoding: 300000000
MESS:00:00:06.053791:0: gpioman: gpioman_get_pin_num: pin CAMERA_0_I2C_PORT not defined
MESS:00:00:06.083743:0: dtb_file 'bcm2711-rpi-cm4.dtb'
MESS:00:00:06.091669:0: brfs: File read: /mfs/sd/bcm2711-rpi-cm4.dtb
MESS:00:00:06.094916:0: Loaded 'bcm2711-rpi-cm4.dtb' to 0x100 size 0xcddf
MESS:00:00:06.114633:0: brfs: File read: 52703 bytes
MESS:00:00:06.134484:0: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
MESS:00:00:06.210310:0: brfs: File read: 2231 bytes
MESS:00:00:06.227425:0: brfs: File read: /mfs/sd/overlays/rpi-poe.dtbo
MESS:00:00:06.248925:0: Loaded overlay 'rpi-poe'
MESS:00:00:06.330913:0: brfs: File read: 4168 bytes
MESS:00:00:06.335103:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:06.338193:0: brfs: File read: 1577 bytes
MESS:00:00:06.354523:0: brfs: File read: /mfs/sd/overlays/pcie-32bit-dma.dtbo
MESS:00:00:06.365378:0: Loaded overlay 'pcie-32bit-dma'
MESS:00:00:06.367505:0: dtparam: pcie=off
MESS:00:00:06.392913:0: brfs: File read: 438 bytes
MESS:00:00:06.399950:0: brfs: File read: /mfs/sd/overlays/dwc2.dtbo
MESS:00:00:06.409292:0: Loaded overlay 'dwc2'
MESS:00:00:06.410535:0: dtparam: dr_mode=host
MESS:00:00:06.425769:0: brfs: File read: 801 bytes
MESS:00:00:06.432012:0: brfs: File read: /mfs/sd/overlays/disable-wifi.dtbo
MESS:00:00:06.444817:0: Loaded overlay 'disable-wifi'
MESS:00:00:06.461118:0: brfs: File read: 387 bytes
MESS:00:00:06.467445:0: brfs: File read: /mfs/sd/overlays/disable-bt.dtbo
MESS:00:00:06.486359:0: Loaded overlay 'disable-bt'
MESS:00:00:06.522485:0: brfs: File read: 1073 bytes
MESS:00:00:06.543724:0: brfs: File read: /mfs/sd/overlays/vc4-fkms-v3d-pi4.dtbo
MESS:00:00:06.566179:0: Loaded overlay 'vc4-fkms-v3d'
MESS:00:00:06.604574:0: brfs: File read: 1446 bytes
MESS:00:00:06.610871:0: brfs: File read: /mfs/sd/overlays/disable-wifi.dtbo
MESS:00:00:06.623733:0: Loaded overlay 'disable-wifi'
MESS:00:00:06.639902:0: brfs: File read: 387 bytes
MESS:00:00:06.646533:0: brfs: File read: /mfs/sd/overlays/dwc2.dtbo
MESS:00:00:06.655881:0: Loaded overlay 'dwc2'
[... 365 lines cut ...]
[   18.516303]  bus_probe_device+0xa4/0xb0
[   18.516307]  device_add+0x390/0x830
[   18.516311]  of_device_add+0x5c/0x70
[   18.516316]  of_platform_device_create_pdata+0x98/0x100
[   18.516321]  of_platform_notify+0xec/0x180
[   18.516324]  blocking_notifier_call_chain+0x74/0xa8
[   18.516331]  of_property_notify+0x68/0xb0
[   18.516337]  __of_changeset_entry_notify+0xb8/0x118
[   18.516342]  __of_changeset_apply_notify+0x4c/0x80
[   18.516347]  of_overlay_fdt_apply+0x6e8/0x8b8
[   18.516351]  cfs_overlay_item_dtbo_write+0x64/0xc0
[   18.516356]  configfs_release_bin_file+0x98/0xa8
[   18.516363]  __fput+0x78/0x238
[   18.516368]  ____fput+0x18/0x28
[   18.516372]  task_work_run+0x88/0x180
[   18.516375]  do_notify_resume+0x1f0/0x4a0
[   18.516378]  el0_svc+0x80/0x98
[   18.516382]  el0t_64_sync_handler+0xb8/0xc0
[   18.516386]  el0t_64_sync+0x18c/0x190
[   18.516391] SMP: stopping secondary CPUs
[   18.516397] Kernel Offset: 0x1fe6000000 from 0xffffffc008000000
[   18.516400] PHYS_OFFSET: 0x0
[   18.516401] CPU features: 0x2000,04027810,00001086
[   18.516404] Memory Limit: none
[   18.877135] ---[ end Kernel panic - not syncing: Asynchronous SError Interrupt ]---

CGDevHusky92, 2022-12-28:
We are running into the exact same issue on our custom board. Except on the other end of our PCI lane is a PCI Packet Switch (PI7C9X2G606PRDNJAE) off of that is a Texas Instruments USB host controller (TUSB7340IRKMT). We plug in various E key wifi modules, but even without any plugged in we run into this unfortunate reboot issue. This is a piece of industrial equipment that will be placed in hard to reach places. Unreliable reboots are going to be an issue that we need to resolve before we can deploy. Any insight on how to handle this would be appreciated.

Boot Uart final output before getting stuck:

[  201.589330] reboot: Restarting system

RPi: BOOTLOADER release VERSION:0fc8fc8e DATE: 2022/12/07 TIME: 14:24:15
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1670423055 serial 81d5d5c7 boardrev c03140 stc 477177
PM_RSTS: 0x00000020
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Samsung' 16Gb x2 total-size: 32 Gbit 3200
DDR 3200 1 0 32 152

Boot mode: USB-MSD (04) order f2561
USB xHC init failed
Boot mode: SD (01) order f256
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
SD HOST: 200000000 CTL0: 0x00800f00 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
EMMC
SD retry 1 oc 0
SD HOST: 200000000 CTL0: 0x00800000 BUS: 400000 Hz actual: 390625 HZ div: 512 (256) status: 0x1fff0000 delay: 276
OCR c0ff8080 [0]
CID: 0015010038475446345206095552c387
SD HOST: 200000000 CTL0: 0x00800f04 BUS: 25000000 Hz actual: 25000000 HZ div: 8 (4) status: 0x1fff0000 delay: 4
SD HOST: 200000000 CTL0: 0x00800f04 BUS: 50000000 Hz actual: 50000000 HZ div: 4 (2) status: 0x1fff0000 delay: 2
MBR: 0x00000800,  524288 type: 0x0c
MBR: 0x00080800,14743519 type: 0x83
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
type: 32 lba: 2048 oem: 'mkfs.fat' volume: ' system-boot'
rsc 32 fat-sectors 4033 c-count 516190 c-size 1
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 516190
Trying partition: 0
type: 32 lba: 2048 oem: 'mkfs.fat' volume: ' system-boot'
rsc 32 fat-sectors 4033 c-count 516190 c-size 1
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 516190
Read config.txt bytes     2810 hnd 0x24938
gpio_cmd: '2=op,dh' pins: 2-2 drive: 1 fsel: 1 term: -1
gpio_cmd: '12=op,dh' pins: 12-12 drive: 1 fsel: 1 term: -1
Invalid GPIO command 14=in,pu
Invalid GPIO command 15=in,pu
Invalid GPIO command 18=in,pu
gpio_cmd: '20=op,dh' pins: 20-20 drive: 1 fsel: 1 term: -1
gpio_cmd: '21=op,dh' pins: 21-21 drive: 1 fsel: 1 term: -1
Invalid GPIO command 23=in,pu
Invalid GPIO command 24=in,pn/np
gpio_cmd: '25=ip,pu' pins: 25-25 drive: -1 fsel: 0 term: 2
Read start4.elf bytes  2241504 hnd 0x4fcd5
Read fixup4.dat bytes     5411 hnd 0x58386
0x00c03140 0x00000000 0x00000fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Firmware: d9b293558b4cef6aabedcc53c178e7604de90788 Nov 18 2021 16:16:49
Starting start4.elf @ 0xfec00200 partition 0
+

A successful boot looks like this:

RPi: BOOTLOADER release VERSION:0fc8fc8e DATE: 2022/12/07 TIME: 14:24:15
BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1670423055 serial 81d5d5c7 boardrev c03140 stc 478850
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Samsung' 16Gb x2 total-size: 32 Gbit 3200
DDR 3200 1 0 32 152

Boot mode: USB-MSD (04) order f2561
[... 114 lines cut ...]
[    0.553944] pcieport 0000:00:00.0:    [20] UnsupReq               (First)
[    0.553968] pcieport 0000:00:00.0: AER:   TLP Header: 30000000 02280032 00000000 00000000
[    0.554897] pcieport 0000:00:00.0: PCIe Bus Error: severity=Uncorrected (Non-Fatal), type=Transaction Layer, (Requester ID)
[    0.554928] pcieport 0000:00:00.0:   device [14e4:2711] error status/mask=00100000/00000000
[    0.554957] pcieport 0000:00:00.0:    [20] UnsupReq               (First)
[    0.554981] pcieport 0000:00:00.0: AER:   TLP Header: 30000000 02280032 00000000 00000000
[    0.556747] pcieport 0000:00:00.0: PCIe Bus Error: severity=Uncorrected (Non-Fatal), type=Transaction Layer, (Requester ID)
[    0.556790] pcieport 0000:00:00.0:   device [14e4:2711] error status/mask=00100000/00000000
[    0.556818] pcieport 0000:00:00.0:    [20] UnsupReq               (First)
[    0.556843] pcieport 0000:00:00.0: AER:   TLP Header: 30000000 02280032 00000000 00000000
[    0.557574] pcieport 0000:00:00.0: PCIe Bus Error: severity=Uncorrected (Non-Fatal), type=Transaction Layer, (Requester ID)
[    0.557605] pcieport 0000:00:00.0:   device [14e4:2711] error status/mask=00100000/00000000
[    0.557633] pcieport 0000:00:00.0:    [20] UnsupReq               (First)
[    0.557658] pcieport 0000:00:00.0: AER:   TLP Header: 30000000 02280032 00000000 00000000
[    0.559966] pcieport 0000:00:00.0: PCIe Bus Error: severity=Uncorrected (Non-Fatal), type=Transaction Layer, (Requester ID)
[    0.560008] pcieport 0000:00:00.0:   device [14e4:2711] error status/mask=00100000/00000000
[    0.560037] pcieport 0000:00:00.0:    [20] UnsupReq               (First)
[    0.560062] pcieport 0000:00:00.0: AER:   TLP Header: 30000000 02280032 00000000 00000000
[    7.412007] systemd[1]: Failed to start Load Kernel Modules.

Ubuntu 22.04 LTS node4969 ttyS0

node4969 login:

Lots of pcieport errors, but once it is up pci works great and without any issues. Any pointers on where to start looking for a solution would be great!

jolla, 2023-01-26:
@CGDevHusky92 the workaround we've come up with is to disable PCIe on boot and then enable PCIe after boot. Not ideal, but it does work around the issue and prevents the unit from hanging on boot.

Subrosian, 2024-10-09:
@jolla I'm dealing with this EXACT same issue. You talk about disabling pcie using config.txt and enabling it after boot. Could you explain in a bit more detail on how you are doing this?

CGDevHusky92, 2024-10-09:
I forgot about this post, but I actually found a patch that helped a few months after commenting here. I can't for the life of me find where this patch exists on the internet, but it fixed all of my pci reset issues without the need to shut off and turn on pci. I have posted a copy of the patch to the pastebin below and I think I had modified it slightly for openwrt os, but hopefully it can help you or anyone else that finds it.

https://pastebin.com/uPEn3axg

CGDevHusky92, 2024-10-09:
> @CGDevHusky92
> I'm using OpenWRT as well. Looking at your patch, I found a VERY similar patch on the snapshot tree. Did you have the freezing right before starting the linux kernel, or the "PCIe Bus Error" messages, or both?
> Patch:
> https://github.com/openwrt/openwrt/blob/c3251f5d52aef636f0cec432a27bf90f7537e1e7/target/linux/bcm27xx/patches-6.6/950-0697-drivers-pci-brcmstb-optionally-extend-Tperst_clk-tim.patch

Oh nice for some reason I couldn’t find it. It was mostly the freeze prior to kernel start. On a different custom board using a different pci packet switch we would get the bus error and then it would come up with no devices. This patch fixed it for both boards in our case.

Subrosian, 2024-10-10:
K, just to make a note for others, running OpenWRT on the snapshot branch, I was able to overcome my issues by adding "pcie_tperst_clk_ms=300" to /boot/config.txt. The snapshot branch had the patch discussed above already installed.