raspberrypi/firmware #1768: Wifi: Firmware crashing regularly on RPi 0W2

porst17, 2023-01-01:
**Describe the bug**
The firmware of the onboard WiFi on my RPi 0W2 crashes almost every day:
```
[78167.189463] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
[78167.237807] brcmfmac: CONSOLE:
[78167.237835] brcmfmac: CONSOLE: FWID 01-3b307371
[78167.237846] brcmfmac: CONSOLE: flags 1
[78167.237856] brcmfmac: CONSOLE:
[78167.237868] brcmfmac: CONSOLE: TRAP c(6ff7c): pc 837364, lr 8372bb, sp 3f828, cpsr 4000000c, spsr 41000000
[78167.237881] brcmfmac: CONSOLE:   r0 0, r1 4, r2 0, r3 0, r4 0, r5 0, r6 60cbc
[78167.237893] brcmfmac: CONSOLE:   r7 4f390, r8 61190, r9 0, r10 168, r11 0, r12 7c
[78167.237904] brcmfmac: CONSOLE:
[78167.237916] brcmfmac: CONSOLE:    sp+0 00000000 00802fa9 00060de8 00061218
[78167.237928] brcmfmac: CONSOLE:   sp+10 00065e00 00061190 00065e00 00060cbc
[78167.237939] brcmfmac: CONSOLE:
[78167.237949] brcmfmac: CONSOLE: sp+4 00802fa9
[78167.237959] brcmfmac: CONSOLE: sp+34 00019ab5
[78167.237970] brcmfmac: CONSOLE: sp+5c 00844f3f
[78167.237980] brcmfmac: CONSOLE: sp+7c 008472b5
[78167.237991] brcmfmac: CONSOLE: sp+94 0085e3f5
[78167.238001] brcmfmac: CONSOLE: sp+fc 0081e9e9
[78167.238012] brcmfmac: CONSOLE: sp+130 00844c95
[78167.238023] brcmfmac: CONSOLE: sp+134 00844fe9
[78167.238033] brcmfmac: CONSOLE: sp+164 0002f6df
[78167.238044] brcmfmac: CONSOLE: sp+1bc 00807f53
[78167.238055] brcmfmac: CONSOLE: sp+1f4 00875b39
[78167.238065] brcmfmac: CONSOLE: sp+234 0001b6b1
[78167.238076] brcmfmac: CONSOLE: sp+274 0001e085
[78167.238087] brcmfmac: CONSOLE: sp+29c 0082eed3
[78167.238097] brcmfmac: CONSOLE: sp+2c4 0082f215
[78167.238108] brcmfmac: CONSOLE: sp+2d4 0081f351
[78167.285400] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
[78167.285433] ieee80211 phy0: brcmf_cfg80211_get_tx_power: error (-5)
[78167.937870] brcmfmac: brcmf_sdiod_probe: Failed to set F1 blocksize
[78167.937934] brcmfmac: brcmf_sdio_bus_reset: Failed to probe after sdio device reset: ret -123
```
It seems I can only get it back online by restarting the RPi. I am tracking this issue for the past months and already tried running with `over_voltage=2` as suggested elsewhere, but no luck.

**Expected behaviour**
WiFi firmware does not crash.

**Actual behaviour**
WiFi firmware crashes.

**System & Logs**
See the attached [raspinfo.txt](https://github.com/raspberrypi/firmware/files/10329128/raspinfo.txt).

**Additional context**
I can't confirm it 100% yet, but the crash seems to happen whenever the connection is handed over from my WiFi router to my WiFi repeater or vice versa.

pelwell, 2023-01-04:
There's a ticket open with Synaptics for this issue. I'll report back with any questions or findings.

Smyl3, 2023-01-06:
I had the same issue with a Raspberry Zero 2 W and Dietpi OS. With like every day unrecoverable brcmf_fw_crash.
It seems its related to overlapping wifi networks.
We used 2 routers here with the same SSID.
I changed one of them SSID to something else, and since then I didnt see wifi issue. I installed a bash script to check my router ip access every 5 min and restart kernel module if needed but it wasnt.
What is weird I didnt upgrade the Raspberry packages (Dietpi OS) and we used this router setup long time ago, so I dont know the reason why it started facing the issue some weeks ago.

Example for the crash:
Dec 29 19:00:10 zero2 kernel: [122694.272662] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
Dec 29 19:00:11 zero2 kernel: [122695.912670] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
Dec 29 19:00:12 zero2 kernel: [122695.985029] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
Dec 29 19:00:12 zero2 kernel: [122695.985074] ieee80211 phy0: brcmf_cfg80211_get_tx_power: error (-5)
Dec 29 19:00:12 zero2 kernel: [122696.617005] brcmfmac: brcmf_sdiod_probe: Failed to set F1 blocksize
Dec 29 19:00:12 zero2 kernel: [122696.617137] brcmfmac: brcmf_sdio_bus_reset: Failed to probe after sdio device reset: r
et -123
Dec 29 19:00:12 zero2 kernel: [122696.618115] mmc1: card 0001 removed
Dec 29 19:00:12 zero2 kernel: [122696.660762] mmc1: queuing unknown CIS tuple 0xc0 [dd 38 c1 95 4f f1 8a 61 89 b7 68 bf
e2 45 00] (15 bytes)
Dec 29 19:00:12 zero2 kernel: [122696.751641] mmc1: queuing unknown CIS tuple 0x2a [80 ff e6 ec 00 00 01 0a 1e 2c 00 f9
92 b6 b4 0c ff 01 13 48 ff 00 e4 41 48 09 38 6b 4c 1f ff 00 58 0a 12 83 7c 7f f4 00 ff ad ff ec 22 08 c6 59 df 10 f8 33
da 1f 48 ff a5 e2 28 2f b4 ff f0 20] (253 bytes)
Dec 29 19:00:12 zero2 kernel: [122696.753777] mmc1: queuing unknown CIS tuple 0xf1 [03 35] (2 bytes)
Dec 29 19:00:12 zero2 kernel: [122696.805589] mmc1: queuing unknown CIS tuple 0x2e [e2 12 22 c1 04 72 76 ff 8d ff fe a4
49 44 00 00 4b 16 86 12 c3 3d 75 07 93 00 01 b1 62 ed 2e 49 16 ff 27 13 6a 0b 02 30 06 b0 c2 02 16 7f 00 41 7a 48 6d 02
24 61 00 20 00 00 9f 42 99 06 c1 04] (144 bytes)
Dec 29 19:00:57 zero2 kernel: [122740.941905] mmc1: error -22 whilst initialising SDIO card
Dec 29 19:00:57 zero2 kernel: [122740.997200] mmc1: queuing unknown CIS tuple 0xc0 [dd 38 c1 95 4f f1 8a 61 89 b7 68 bf
e2 45 00] (15 bytes)
Dec 29 19:00:57 zero2 kernel: [122741.110450] mmc1: queuing unknown CIS tuple 0x2a [80 ff e6 ec 00 00 01 0a 1e 2c 00 f9
92 b6 b4 0c ff 01 13 48 ff 00 e4 41 48 09 38 6b 4c 1f ff 00 58 0a 12 83 7c 7f f4 00 ff ad ff ec 22 08 c6 59 df 10 f8 33
da 1f 48 ff a5 e2 28 2f b4 ff f0 20] (253 bytes)
Dec 29 19:00:57 zero2 kernel: [122741.112705] mmc1: queuing unknown CIS tuple 0xf1 [03 35] (2 bytes)
Dec 29 19:00:57 zero2 kernel: [122741.177792] mmc1: queuing unknown CIS tuple 0x2e [e2 12 22 c1 04 72 76 ff 8d ff fe a4
49 44 00 00 4b 16 86 12 c3 3d 75 07 93 00 01 b1 62 ed 2e 49 16 ff 27 13 6a 0b 02 30 06 b0 c2 02 16 7f 00 41 7a 48 6d 02
24 61 00 20 00 00 9f 42 99 06 c1 04] (144 bytes)
Dec 29 19:01:52 zero2 kernel: [122796.785417] mmc1: error -22 whilst initialising SDIO card
Dec 29 19:01:52 zero2 kernel: [122796.850698] mmc1: queuing unknown CIS tuple 0xc0 [dd 38 c1 95 4f f1 8a 61 89 b7 68 bf
e2 45 00] (15 bytes)

pelwell, 2023-01-06:
Can you do what @porst17 did and add `brcmfmac.debug=0x100000` to `/boot/cmdline.txt` (keeping it all in one line) and reboot? Any firmware crash should then generate a log like the one in the first post:
```
[78167.189463] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
[78167.237807] brcmfmac: CONSOLE:
[78167.237835] brcmfmac: CONSOLE: FWID 01-3b307371
[78167.237846] brcmfmac: CONSOLE: flags 1
[78167.237856] brcmfmac: CONSOLE:
[78167.237868] brcmfmac: CONSOLE: TRAP c(6ff7c): pc 837364, lr 8372bb, sp 3f828, cpsr 4000000c, spsr 41000000
[78167.237881] brcmfmac: CONSOLE:   r0 0, r1 4, r2 0, r3 0, r4 0, r5 0, r6 60cbc
[78167.237893] brcmfmac: CONSOLE:   r7 4f390, r8 61190, r9 0, r10 168, r11 0, r12 7c
...
```

porst17, 2023-01-08:
I also want to add that my setup uses the same SSID for the main router and the repeater. This is the default setup of the wifi mesh system (AVM Fritz!Box) I have been using for years without problems on the many WiFi devices I own. It's just the RPi 0W2 that doesn't seem to like it. 🤷‍♂️

WLBQE, 2023-01-09:
I have also been experiencing this on my 0W2 for two weeks, ever since I moved the pi from a wall socket to a power strip, so it might have something to do with voltage.
Until a couple of days ago I handled it by plugging the pi out and then back in. The firmware could hold up for at least a day before crashing again. Then I switched to a cron script that reboots whenever the firmware crashes, and in the past 24 hours it has crashed 3 times.
One thing I notice is that the crash always happens after attempting to renew the DHCP lease. Meanwhile, as other folks, I'm also on a mesh Wi-Fi network (Deco).
I'll set the firmware to debug mode and post logs here the next time it crashes.

WLBQE, 2023-01-19:
Logs from a recent crash:
```
Jan 18 17:50:25 homebridge kernel: [87244.740253] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
Jan 18 17:50:25 homebridge kernel: [87244.798582] brcmfmac: CONSOLE: 
Jan 18 17:50:25 homebridge kernel: [87244.798611] brcmfmac: CONSOLE: FWID 01-3b307371
Jan 18 17:50:25 homebridge kernel: [87244.798624] brcmfmac: CONSOLE: flags 1
Jan 18 17:50:25 homebridge kernel: [87244.798636] brcmfmac: CONSOLE: 
Jan 18 17:50:25 homebridge kernel: [87244.798648] brcmfmac: CONSOLE: TRAP 3(6ff7c): pc 87b7c6, lr 867231, sp 3f71c, cpsr 20000003, spsr 21000000
Jan 18 17:50:25 homebridge kernel: [87244.798661] brcmfmac: CONSOLE:   r0 f7d19ee4, r1 61190, r2 3f808, r3 0, r4 3f784, r5 1002, r6 3f780
Jan 18 17:50:25 homebridge kernel: [87244.798675] brcmfmac: CONSOLE:   r7 0, r8 3f808, r9 0, r10 61190, r11 6058c, r12 3f7d4
Jan 18 17:50:25 homebridge kernel: [87244.798687] brcmfmac: CONSOLE: 
Jan 18 17:50:25 homebridge kernel: [87244.798699] brcmfmac: CONSOLE:    sp+0 0003f784 00001002 00867231 00000001
Jan 18 17:50:25 homebridge kernel: [87244.798711] brcmfmac: CONSOLE:   sp+10 0085ec05 ffffffff 0005a080 00000000
Jan 18 17:50:25 homebridge kernel: [87244.798723] brcmfmac: CONSOLE: 
Jan 18 17:50:25 homebridge kernel: [87244.798734] brcmfmac: CONSOLE: sp+8 00867231
Jan 18 17:50:25 homebridge kernel: [87244.798746] brcmfmac: CONSOLE: sp+10 0085ec05
Jan 18 17:50:25 homebridge kernel: [87244.798758] brcmfmac: CONSOLE: sp+48 00867607
Jan 18 17:50:25 homebridge kernel: [87244.798769] brcmfmac: CONSOLE: sp+88 008364af
Jan 18 17:50:25 homebridge kernel: [87244.798780] brcmfmac: CONSOLE: sp+a0 00839441
Jan 18 17:50:25 homebridge kernel: [87244.798792] brcmfmac: CONSOLE: sp+b8 008566f3
Jan 18 17:50:25 homebridge kernel: [87244.798803] brcmfmac: CONSOLE: sp+d8 0084807f
Jan 18 17:50:25 homebridge kernel: [87244.798815] brcmfmac: CONSOLE: sp+110 00830fc9
Jan 18 17:50:25 homebridge kernel: [87244.798826] brcmfmac: CONSOLE: sp+15c 00001531
Jan 18 17:50:25 homebridge kernel: [87244.798838] brcmfmac: CONSOLE: sp+164 00011531
Jan 18 17:50:25 homebridge kernel: [87244.798850] brcmfmac: CONSOLE: sp+1c4 0087702d
Jan 18 17:50:25 homebridge kernel: [87244.798861] brcmfmac: CONSOLE: sp+1c8 0087710d
Jan 18 17:50:25 homebridge kernel: [87244.798872] brcmfmac: CONSOLE: sp+1d0 00807b3f
Jan 18 17:50:25 homebridge kernel: [87244.798884] brcmfmac: CONSOLE: sp+230 0000586f
Jan 18 17:50:25 homebridge kernel: [87244.798895] brcmfmac: CONSOLE: sp+28c 008302d5
Jan 18 17:50:25 homebridge kernel: [87244.798907] brcmfmac: CONSOLE: sp+2a0 00833417
Jan 18 17:50:25 homebridge kernel: [87245.147269] ieee80211 phy0: brcmf_netdev_start_xmit: xmit rejected state=0
Jan 18 17:50:27 homebridge kernel: [87247.291281] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Jan 18 17:50:27 homebridge kernel: [87247.292223] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
Jan 18 17:50:27 homebridge kernel: [87247.292588] brcmfmac: dongle trap info: type 0x3 @ epc 0x0087b7c6
Jan 18 17:50:27 homebridge kernel: [87247.292588]   cpsr 0x20000003 spsr 0x21000000 sp 0x0003f71c
Jan 18 17:50:27 homebridge kernel: [87247.292588]   lr   0x00867231 pc   0x0087b7c6 offset 0x6ff7c
Jan 18 17:50:27 homebridge kernel: [87247.292588]   r0   0xf7d19ee4 r1   0x00061190 r2 0x0003f808 r3 0x00000000
Jan 18 17:50:27 homebridge kernel: [87247.292588]   r4   0x0003f784 r5   0x00001002 r6 0x0003f780 r7 0x00000000
Jan 18 17:50:27 homebridge kernel: [87247.292660] ieee80211 phy0: brcmf_cfg80211_disconnect: error (-110)
Jan 18 17:50:27 homebridge kernel: [87247.292753] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
Jan 18 17:50:27 homebridge kernel: [87247.292777] ieee80211 phy0: _brcmf_set_multicast_list: Setting BRCMF_C_SET_PROMISC failed, err=-5
Jan 18 17:50:27 homebridge kernel: [87247.292805] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
Jan 18 17:50:27 homebridge kernel: [87247.292827] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
Jan 18 17:50:28 homebridge kernel: [87247.807106] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
Jan 18 17:50:28 homebridge kernel: [87247.807172] ieee80211 phy0: brcmf_cfg80211_get_tx_power: error (-5)
Jan 18 17:50:28 homebridge kernel: [87247.807612] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
Jan 18 17:50:28 homebridge kernel: [87247.807643] ieee80211 phy0: brcmf_cfg80211_reg_notifier: Country code iovar returned err = -5
Jan 18 17:50:28 homebridge kernel: [87247.886394] brcmfmac: brcmf_sdiod_probe: Failed to set F1 blocksize
Jan 18 17:50:28 homebridge kernel: [87247.886499] brcmfmac: brcmf_sdio_bus_reset: Failed to probe after sdio device reset: ret -123
Jan 18 17:50:28 homebridge kernel: [87247.887801] mmc1: card 0001 removed
Jan 18 17:50:28 homebridge kernel: [87248.012263] mmc1: queuing unknown CIS tuple 0x02 [0b 9f 10 37 b5 cb 93 26 ff 98 4a e6 44 80 00 d6 03 c9 ff 76 4d 53 ff bd 60 c7 42 ca 03 e3 50 3f 0c cf 69 d6 5f ff 12 c8 44 cd 48 cc c0 ca b3 39 00 51 4d ff 41 8c d1 a8 c8 ff 84 ff c1 cc 52 9a] (196 bytes)
Jan 18 17:50:28 homebridge kernel: [87248.014664] mmc1: queuing unknown CIS tuple 0x6e [12 81 29 22 77] (5 bytes)
Jan 18 17:50:28 homebridge kernel: [87248.075837] mmc1: queuing unknown CIS tuple 0x0c [b9 ff f5 42 31 4b 51 1a 95 69 ff 03 eb 44 7f c0 e3 24 fc 4a 30 ab 4c 71 77 40 ff 28 b3 c2 fd c0 6e 81 73 12 ff d6 29 8e 39 2a be 86 49 02 04 03 7b 22 30 13 7e 3f e9 13 e0 59 e4 42 e4 fa a4 84] (178 bytes)
Jan 18 17:50:28 homebridge kernel: [87248.096219] mmc1: queuing unknown CIS tuple 0xf0 [51 ff 32 c8 45 04 cf 43 56 47 59 a8 52 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00] (58 bytes)
```

mati75, 2023-01-29:
I have the same issue with the firmware from next branch.

pelwell, 2023-01-30:
I've received two test firmwares from Synaptics, numbered 1 and 2. There was no accompanying information, so I don't know if they might be a fixes or whether they are just to gather more information.

Firmware 1: https://drive.google.com/file/d/1UghnB9zZ7ydpVlYaTbp8F1M0rqEX_DZK/view?usp=share_link
Firmware 2: https://drive.google.com/file/d/1P3QZDMFPR3I5wPGYWNu2SCEJli9oTCC6/view?usp=share_link

To back up the original firmware:
```
$ sudo cp /lib/firmware/brcm/brcmfmac43436-sdio.bin{,.orig} 
```
To install (change `test1` to `test2` for the second test firmware):
```
$ sudo cp fw_bcmdhd_test1.bin /lib/firmware/brcm/brcmfmac43436-sdio.bin
```
To revert to the original firmware:
```
$ sudo cp /lib/firmware/brcm/brcmfmac43436-sdio.bin{.orig,} 
```

Smyl3, 2023-01-30:
Unfortunately it would be pointless to test those test firmwares for me I think, because I still couldnt reproduce the issuse for some reason.

porst17, 2023-01-31:
I am running the `test2` firmware since yesterday and so far there haven't been any firmware crashes. I'll report back in a couple of days.

pelwell, 2023-01-31:
@Smyl3 I understand - enjoy the lack of crashes.

@porst17 Thank you - that's sounding promising.

porst17, 2023-02-03:
Unfortunately, the Wifi firmware (`test2`) crashed again two times over the past couple of days:
```
Feb  1 22:14:41 openhab kernel: [212302.000222] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
Feb  1 22:14:41 openhab kernel: [212302.046161] brcmfmac: CONSOLE:
Feb  1 22:14:41 openhab kernel: [212302.046190] brcmfmac: CONSOLE: FWID 01-c43ac73d
Feb  1 22:14:41 openhab kernel: [212302.046201] brcmfmac: CONSOLE: flags 1
Feb  1 22:14:41 openhab kernel: [212302.046211] brcmfmac: CONSOLE:
Feb  1 22:14:41 openhab kernel: [212302.046224] brcmfmac: CONSOLE: TRAP c(6ff7c): pc 83cf64, lr 83cf5d, sp 40ff0, cpsr 2000000c, spsr 21000000
Feb  1 22:14:41 openhab kernel: [212302.046237] brcmfmac: CONSOLE:   r0 1, r1 41138, r2 2151, r3 137, r4 41138, r5 0, r6 67940
Feb  1 22:14:41 openhab kernel: [212302.046250] brcmfmac: CONSOLE:   r7 5057c, r8 30, r9 1, r10 85e671, r11 41170, r12 4
Feb  1 22:14:41 openhab kernel: [212302.046261] brcmfmac: CONSOLE:
Feb  1 22:14:41 openhab kernel: [212302.046272] brcmfmac: CONSOLE:    sp+0 0005d1e0 00041138 00000000 0083cef5
Feb  1 22:14:41 openhab kernel: [212302.046284] brcmfmac: CONSOLE:   sp+10 00041138 0005b810 00000016 0085e6c5
Feb  1 22:14:41 openhab kernel: [212302.046295] brcmfmac: CONSOLE:
Feb  1 22:14:41 openhab kernel: [212302.046305] brcmfmac: CONSOLE: sp+c 0083cef5
Feb  1 22:14:41 openhab kernel: [212302.046316] brcmfmac: CONSOLE: sp+1c 0085e6c5
Feb  1 22:14:41 openhab kernel: [212302.046328] brcmfmac: CONSOLE: sp+4c 0085e7c9
Feb  1 22:14:41 openhab kernel: [212302.046211] brcmfmac: CONSOLE:
Feb  1 22:14:41 openhab kernel: [212302.046224] brcmfmac: CONSOLE: TRAP c(6ff7c): pc 83cf64, lr 83cf5d, sp 40ff0, cpsr 2000000c, spsr 21000000
Feb  1 22:14:41 openhab kernel: [212302.046237] brcmfmac: CONSOLE:   r0 1, r1 41138, r2 2151, r3 137, r4 41138, r5 0, r6 67940
Feb  1 22:14:41 openhab kernel: [212302.046250] brcmfmac: CONSOLE:   r7 5057c, r8 30, r9 1, r10 85e671, r11 41170, r12 4
Feb  1 22:14:41 openhab kernel: [212302.046261] brcmfmac: CONSOLE:
Feb  1 22:14:41 openhab kernel: [212302.046272] brcmfmac: CONSOLE:    sp+0 0005d1e0 00041138 00000000 0083cef5
Feb  1 22:14:41 openhab kernel: [212302.046284] brcmfmac: CONSOLE:   sp+10 00041138 0005b810 00000016 0085e6c5
Feb  1 22:14:41 openhab kernel: [212302.046295] brcmfmac: CONSOLE:
Feb  1 22:14:41 openhab kernel: [212302.046305] brcmfmac: CONSOLE: sp+c 0083cef5
Feb  1 22:14:41 openhab kernel: [212302.046316] brcmfmac: CONSOLE: sp+1c 0085e6c5
Feb  1 22:14:41 openhab kernel: [212302.046328] brcmfmac: CONSOLE: sp+4c 0085e7c9
Feb  1 22:14:41 openhab kernel: [212302.046338] brcmfmac: CONSOLE: sp+8c 0085e929
Feb  1 22:14:41 openhab kernel: [212302.046349] brcmfmac: CONSOLE: sp+98 0085e671
Feb  1 22:14:41 openhab kernel: [212302.046359] brcmfmac: CONSOLE: sp+11c 0085e515
Feb  1 22:14:41 openhab kernel: [212302.046370] brcmfmac: CONSOLE: sp+130 0085e635
Feb  1 22:14:41 openhab kernel: [212302.046380] brcmfmac: CONSOLE: sp+134 0085e671
Feb  1 22:14:41 openhab kernel: [212302.046391] brcmfmac: CONSOLE: sp+18c 0000013b
Feb  1 22:14:41 openhab kernel: [212302.046401] brcmfmac: CONSOLE: sp+1b4 0085ea59
Feb  1 22:14:41 openhab kernel: [212302.046412] brcmfmac: CONSOLE: sp+1d4 00836547
Feb  1 22:14:41 openhab kernel: [212302.046422] brcmfmac: CONSOLE: sp+22c 0085e2af
Feb  1 22:14:41 openhab kernel: [212302.046433] brcmfmac: CONSOLE: sp+25c 008390b3
Feb  1 22:14:41 openhab kernel: [212302.046443] brcmfmac: CONSOLE: sp+300 00832175
Feb  1 22:14:41 openhab kernel: [212302.046454] brcmfmac: CONSOLE: sp+304 00832165
Feb  1 22:14:41 openhab kernel: [212302.046464] brcmfmac: CONSOLE: sp+308 00832191
```

```
Feb 02 04:15:08 openhab kernel: ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE:
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE: FWID 01-c43ac73d
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE: flags 1
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE:
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE: TRAP c(6ff7c): pc 83cf64, lr 83cf5d, sp 40ff0, cpsr 2000000c, spsr 21000000
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE:   r0 1, r1 41138, r2 2151, r3 137, r4 41138, r5 0, r6 67940
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE:   r7 50d60, r8 30, r9 1, r10 85e671, r11 41170, r12 4
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE:
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE:    sp+0 0005d1e0 00041138 00000000 0083cef5
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE:   sp+10 00041138 0005b810 00000016 0085e6c5
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE:
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE:
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE: TRAP c(6ff7c): pc 83cf64, lr 83cf5d, sp 40ff0, cpsr 2000000c, spsr 21000000
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE:   r0 1, r1 41138, r2 2151, r3 137, r4 41138, r5 0, r6 67940
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE:   r7 50d60, r8 30, r9 1, r10 85e671, r11 41170, r12 4
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE:
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE:    sp+0 0005d1e0 00041138 00000000 0083cef5
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE:   sp+10 00041138 0005b810 00000016 0085e6c5
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE:
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE: sp+c 0083cef5
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE: sp+1c 0085e6c5
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE: sp+4c 0085e7c9
[... cut ...]
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE: sp+18c 0000013b
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE: sp+1b4 0085ea59
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE: sp+1d4 00836547
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE: sp+22c 0085e2af
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE: sp+25c 008390b3
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE: sp+300 00832175
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE: sp+304 00832165
Feb 02 04:15:08 openhab kernel: brcmfmac: CONSOLE: sp+308 00832191
```

porst17, 2023-02-03:
I was running `test2` so far and switched over to the `test1` firmware just now. I'll keep you updated.

pelwell, 2023-02-03:
Thanks again. Until we are able to reproduce the issue we are relying on people like you.

mati75, 2023-02-04:
In my case test1 crashes just as often, additionally bluetooth loses connections. test2 works fine so far.

pelwell, 2023-02-04:
Interesting. Do you have an example log from a test1 crash?

porst17, 2023-02-10:
> In my case test1 crashes just as often, additionally bluetooth loses connections. test2 works fine so far.

I've been running `test1` now for about a week without a single crash. 🤞
Strange that others seem to observe the opposite behavior.

pelwell, 2023-02-10:
Indeed - it is strange.

@mati75 It would be helpful if you could find the time to capture and post a crash log from the test1 firmware.

Smyl3, 2023-02-10:
Maybe there are more different issues in the firmware that needs different solutions.

mati75, 2023-02-10:
@pelwell I will try catch crash at the weekend. I only caught that the crash occurs when dhcpv6 loses connection, but I wasn't able to save dmesg.

mati75, 2023-02-10:
```
[Fri Feb 10 18:56:22 2023] brcmfmac: CONSOLE: wlc_join_adopt_bss, 5751
[Fri Feb 10 18:56:22 2023] brcmfmac: CONSOLE: wlc_link, 36727
[Fri Feb 10 18:56:22 2023] brcmfmac: CONSOLE: wlc_link, 36849
[Fri Feb 10 18:56:22 2023] brcmfmac: CONSOLE: wl0: link up (wl0)
[Fri Feb 10 18:56:22 2023] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x1 tid 0
[Fri Feb 10 18:56:22 2023] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[Fri Feb 10 18:56:22 2023] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[Fri Feb 10 18:56:22 2023] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[Fri Feb 10 18:56:22 2023] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[Fri Feb 10 18:56:22 2023] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[Fri Feb 10 18:56:22 2023] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[Fri Feb 10 18:56:22 2023] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[Fri Feb 10 18:56:28 2023] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[Fri Feb 10 18:56:33 2023] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x1 tid 4
[Fri Feb 10 18:56:34 2023] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[Fri Feb 10 18:56:34 2023] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[Fri Feb 10 18:56:35 2023] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
```

```
ping6 google.com -c 3
PING google.com(bud02s23-in-x0e.1e100.net (2a00:1450:400d:802::200e)) 56 data bytes

--- google.com ping statistics ---
3 packets transmitted, 0 received, 100% packet loss, time 2108ms
```

connections down, in the same network pi0w, rpi4 working fine

aa777888-2, 2023-02-15:
All: I'm also seeing this problem, I think. My Pi Z2W is living in a Ubiquiti wireless environment. Four UAP-AC-LR access points, all advertising three different SSIDs, on SSID reserved for IoT devices and not broadcasting it's ID. The Z2W is on the IoT WiFi SSID. It can really only see two of the access points. I used Unifi controller to lock the Z2W to a single AP. This reduced spontaneous reboots from a couple of times a day to about once or twice a week, but still has not fixed the problem. Running box stock Raspian 64 bit.

Here is a syslog snippet from the latest reboot. You can see the usual FIPS logging nonsense, some trouble with the WiFi connection, then BOOM!

`Feb 15 09:34:23 genmon rngd[397]: stats: bits received from HRNG source: 6200064
Feb 15 09:34:23 genmon rngd[397]: stats: bits sent to kernel pool: 6146144
Feb 15 09:34:23 genmon rngd[397]: stats: entropy added to kernel pool: 6146144
Feb 15 09:34:23 genmon rngd[397]: stats: FIPS 140-2 successes: 310
Feb 15 09:34:23 genmon rngd[397]: stats: FIPS 140-2 failures: 0
Feb 15 09:34:23 genmon rngd[397]: stats: FIPS 140-2(2001-10-10) Monobit: 0
Feb 15 09:34:23 genmon rngd[397]: stats: FIPS 140-2(2001-10-10) Poker: 0
Feb 15 09:34:23 genmon rngd[397]: stats: FIPS 140-2(2001-10-10) Runs: 0
Feb 15 09:34:23 genmon rngd[397]: stats: FIPS 140-2(2001-10-10) Long run: 0
Feb 15 09:34:23 genmon rngd[397]: stats: FIPS 140-2(2001-10-10) Continuous run: 0
Feb 15 09:34:23 genmon rngd[397]: stats: HRNG source speed: (min=906.070; avg=1070.052; max=1088.698)Kibits/s
Feb 15 09:34:23 genmon rngd[397]: stats: FIPS tests speed: (min=23.176; avg=36.915; max=43.747)Mibits/s
Feb 15 09:34:23 genmon rngd[397]: stats: Lowest ready-buffers level: 2
Feb 15 09:34:23 genmon rngd[397]: stats: Entropy starvations: 0
Feb 15 09:34:23 genmon rngd[397]: stats: Time spent starving for entropy: (min=0; avg=0.000; max=0)us
Feb 15 09:57:29 genmon dhcpcd[491]: wlan0: carrier lost
Feb 15 09:57:29 genmon dhcpcd[491]: wlan0: deleting route to 192.168.1.0/24
Feb 15 09:57:29 genmon dhcpcd[491]: wlan0: deleting default route via 192.168.1.1
Feb 15 09:57:36 genmon dhcpcd[491]: wlan0: carrier acquired
Feb 15 09:57:36 genmon dhcpcd[491]: wlan0: IAID 01:44:3d:a1
Feb 15 09:57:36 genmon dhcpcd[491]: wlan0: probing address 192.168.1.62/24
Feb 15 09:57:37 genmon dhcpcd[491]: wlan0: soliciting an IPv6 router
Feb 15 09:57:41 genmon dhcpcd[491]: wlan0: using static address 192.168.1.62/24
Feb 15 09:57:41 genmon dhcpcd[491]: wlan0: adding route to 192.168.1.0/24
Feb 15 09:57:41 genmon dhcpcd[491]: wlan0: adding default route via 192.168.1.1
Feb 15 09:57:49 genmon dhcpcd[491]: wlan0: carrier lost
Feb 15 09:57:49 genmon dhcpcd[491]: wlan0: deleting route to 192.168.1.0/24
Feb 15 09:57:49 genmon dhcpcd[491]: wlan0: deleting default route via 192.168.1.1
Feb 15 09:57:51 genmon dhcpcd[491]: wlan0: carrier acquired
Feb 15 09:57:51 genmon dhcpcd[491]: wlan0: IAID 01:44:3d:a1
Feb 15 09:57:51 genmon dhcpcd[491]: wlan0: probing address 192.168.1.62/24
Feb 15 09:57:51 genmon dhcpcd[491]: wlan0: soliciting an IPv6 router
Feb 15 09:57:56 genmon dhcpcd[491]: wlan0: using static address 192.168.1.62/24
Feb 15 09:57:56 genmon dhcpcd[491]: wlan0: adding route to 192.168.1.0/24
Feb 15 09:57:56 genmon dhcpcd[491]: wlan0: adding default route via 192.168.1.1
Feb 15 10:02:44 genmon dhcpcd[491]: wlan0: carrier lost
Feb 15 10:02:44 genmon dhcpcd[491]: wlan0: deleting route to 192.168.1.0/24
Feb 15 10:02:44 genmon dhcpcd[491]: wlan0: deleting default route via 192.168.1.1
Feb 15 10:02:46 genmon dhcpcd[491]: wlan0: carrier acquired
Feb 15 10:02:46 genmon dhcpcd[491]: wlan0: IAID 01:44:3d:a1
Feb 15 10:02:46 genmon dhcpcd[491]: wlan0: probing address 192.168.1.62/24
Feb 15 10:02:47 genmon dhcpcd[491]: wlan0: soliciting an IPv6 router
Feb 15 10:02:51 genmon dhcpcd[491]: wlan0: using static address 192.168.1.62/24
Feb 15 10:02:51 genmon dhcpcd[491]: wlan0: adding route to 192.168.1.0/24
Feb 15 10:02:51 genmon dhcpcd[491]: wlan0: adding default route via 192.168.1.1
Feb 15 10:17:01 genmon CRON[182485]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Feb 15 10:34:23 genmon rngd[397]: stats: bits received from HRNG source: 6220064
Feb 15 10:34:23 genmon rngd[397]: stats: bits sent to kernel pool: 6176384
Feb 15 10:34:23 genmon rngd[397]: stats: entropy added to kernel pool: 6176384
Feb 15 10:34:23 genmon rngd[397]: stats: FIPS 140-2 successes: 311
Feb 15 10:34:23 genmon rngd[397]: stats: FIPS 140-2 failures: 0
Feb 15 10:34:23 genmon rngd[397]: stats: FIPS 140-2(2001-10-10) Monobit: 0
Feb 15 10:34:23 genmon rngd[397]: stats: FIPS 140-2(2001-10-10) Poker: 0
Feb 15 10:34:23 genmon rngd[397]: stats: FIPS 140-2(2001-10-10) Runs: 0
Feb 15 10:34:23 genmon rngd[397]: stats: FIPS 140-2(2001-10-10) Long run: 0
Feb 15 10:34:23 genmon rngd[397]: stats: FIPS 140-2(2001-10-10) Continuous run: 0
Feb 15 10:34:23 genmon rngd[397]: stats: HRNG source speed: (min=906.070; avg=1070.062; max=1088.698)Kibits/s
Feb 15 10:34:23 genmon rngd[397]: stats: FIPS tests speed: (min=23.176; avg=36.866; max=43.747)Mibits/s
Feb 15 10:34:23 genmon rngd[397]: stats: Lowest ready-buffers level: 2
Feb 15 10:34:23 genmon rngd[397]: stats: Entropy starvations: 0
Feb 15 10:34:23 genmon rngd[397]: stats: Time spent starving for entropy: (min=0; avg=0.000; max=0)us
Feb 15 10:17:07 genmon fake-hwclock[132]: Wed 15 Feb 15:17:01 UTC 2023
Feb 15 10:17:07 genmon systemd-fsck[146]: e2fsck 1.46.2 (28-Feb-2021)
[... cut ...] 73920 blocks
Feb 15 10:17:07 genmon systemd[1]: Finished Remount Root and Kernel File Systems.
Feb 15 10:17:07 genmon systemd[1]: Condition check resulted in Rebuild Hardware Database being skipped.
Feb 15 10:17:07 genmon systemd[1]: Starting Flush Journal to Persistent Storage...
Feb 15 10:17:07 genmon systemd[1]: Condition check resulted in Platform Persistent Storage Archival being skipped.
Feb 15 10:17:07 genmon kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
`

pelwell, 2023-02-15:
[ Stop editing the log - I've fixed it up once already ]

aa777888-2, 2023-02-15:
Sorry, Phil, but I had no way of knowing you and I were editing in parallel. Your last edit truncated the log. It appears fine now. I could not get it to show properly in code tags for some reason.

pelwell, 2023-02-15:
The WiFi firmware crashing (if that's what's happening to you - you don't appear to have added brcmfmac.debug=0x100000 to /boot/cmdline.txt as requested above) shouldn't cause your Pi to reboot.

How are you powering the Pi and its peripherals?

aa777888-2, 2023-02-15:
I made the change and rebooted as requested. Now we need to wait for a spontaneous reboot. Where should I look for the relevant log entries?

The Z2W is powered by a large, commercial, 5VDC power supply that can provide 5A. The only peripheral the Z2W is powering is a single RS232-TTL level translator IC on the 3.3VDC bus which consumes well under 10ma. The Z2W never reports undervoltage or throttling. The SD card has been replaced once as a possible source of the problem.

pelwell, 2023-02-15:
The additional log entries will go into the kernel log, which should appear in the main system log as above.

Smyl3, 2023-02-18:
So, Im still using the original fw. that still didnt crash but I had a weird wifi connection issue today until my script was restarted brcmfmac kernel driver at 14:20. Dont know is it related to this issue or not.
The issue started with a `DHCPREQUEST` and I see a lot of
`wpa_supplicant[589307]: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=MAC status_code=16` messages in syslog....

[kern.log.23.02.18.1415.1420.log](https://github.com/raspberrypi/firmware/files/10774656/kern.log.23.02.18.1415.1420.log)
[syslog.23.02.18.1415.1420.log](https://github.com/raspberrypi/firmware/files/10774657/syslog.23.02.18.1415.1420.log)

w-u-2-o, 2023-02-19:
OK, it took about 4 days since adding brcmfmac.debug=0x100000 before suffering another spontaneous reboot. Syslog is attached. Reboot at 15:17:36. As before, it happens after wlan0 connection problems. Unifi controller reports no problem with any other client on the AP in question, only the Pi.
[syslog.txt](https://github.com/raspberrypi/firmware/files/10777915/syslog.txt)

porst17, 2023-02-20:
> > In my case test1 crashes just as often, additionally bluetooth loses connections. test2 works fine so far.
> 
> I've been running `test1` now for about a week without a single crash. 🤞 Strange that others seem to observe the opposite behavior.

Another week without a crash since running the `test1` firmware (about ~17 days in total). Looks very good to me! I'll still continue monitoring it.

porst17, 2023-02-20:
> OK, it took about 4 days since adding brcmfmac.debug=0x100000 before suffering another spontaneous reboot. Syslog is attached. Reboot at 15:17:36. As before, it happens after wlan0 connection [...]

I noticed a typo in your kernel command line:
```
$ curl -sL https://github.com/raspberrypi/firmware/files/10777915/syslog.txt | grep brcmfmac | head -n 2
Feb 19 15:17:36 genmon kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Composite-1:720x480@60i smsc95xx.macaddr=B8:27:EB:23:89:A6 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=tty1 root=PARTUUID=94dff47d-02 rootfstype=ext4 fsck.repair=yes rootwait modules-load=dwc2,g_ether brcmfmac.debu=0x100000
Feb 19 15:17:36 genmon kernel: [   11.249067] brcmfmac: unknown parameter 'debu' ignored
```
You added `brcmfmac.debu=0x100000` instead of `brcmfmac.debug=0x100000` and `brcmfmac` doesn't know what to do with it. I suggest fixing the typo and reporting the next crash.

aa777888-2, 2023-02-20:
Well that's embarrassing! I will correct that immediately! Thank you!

aa777888-2, 2023-02-20:
Well that's embarrassing. I will did that immediately. Thanks!

pelwell, 2023-02-20:
Well spotted, @porst17.

aa777888-2, 2023-03-02:
Syslog file attached with a flurry of 5 reboots at the bottom. brcmfmac debug working in this one. Just search the file for "booting" and you'll find them.

Hell if I know what is going on. It seems like I can make a reboot happen by causing a lot of WiFi traffic on the device. There's plenty of amps available on the 5V power supply and I never see an undervoltage error being reported.

I had to trim the top of the syslog file because it was HUGE, too big to attach, with all the brcmfmac entries in it.

[syslog_trimmed.txt](https://github.com/raspberrypi/firmware/files/10875361/syslog_trimmed.txt)

phylax2020, 2023-03-24:
I have the same problem as Smyl3:
The crash occurs after several days. After that, only a reboot helped to reactivate the WLAN.
Before I migrated from Pi-OS Buster to Bullseye, there were no problems with WiFi.

[image]

Mar 21 18:28:36 pi-ager kernel: [    8.197340] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
Mar 21 18:28:41 pi-ager kernel: [   13.528200] Bluetooth: Core ver 2.22
Mar 21 18:28:41 pi-ager kernel: [   13.528316] NET: Registered PF_BLUETOOTH protocol family
Mar 21 18:28:41 pi-ager kernel: [   13.528324] Bluetooth: HCI device and connection manager initialized
Mar 21 18:28:41 pi-ager kernel: [   13.528345] Bluetooth: HCI socket layer initialized
Mar 21 18:28:41 pi-ager kernel: [   13.528355] Bluetooth: L2CAP socket layer initialized
Mar 21 18:28:41 pi-ager kernel: [   13.528374] Bluetooth: SCO socket layer initialized
Mar 21 18:28:42 pi-ager kernel: [   13.544732] Bluetooth: HCI UART driver ver 2.3
Mar 21 18:28:42 pi-ager kernel: [   13.544756] Bluetooth: HCI UART protocol H4 registered
Mar 21 18:28:42 pi-ager kernel: [   13.544844] Bluetooth: HCI UART protocol Three-wire (H5) registered
Mar 21 18:28:42 pi-ager kernel: [   13.545008] Bluetooth: HCI UART protocol Broadcom registered
Mar 21 18:28:42 pi-ager kernel: [   13.869326] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Mar 21 18:28:42 pi-ager kernel: [   13.869349] Bluetooth: BNEP filters: protocol multicast
Mar 21 18:28:42 pi-ager kernel: [   13.869368] Bluetooth: BNEP socket layer initialized
Mar 21 18:28:42 pi-ager kernel: [   13.877673] Bluetooth: MGMT ver 1.22
Mar 21 18:28:42 pi-ager kernel: [   13.892573] NET: Registered PF_ALG protocol family
Mar 21 18:28:42 pi-ager kernel: [   13.928797] cryptd: max_cpu_qlen set to 1000
Mar 21 18:29:06 pi-ager kernel: [   38.333622] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
Mar 21 18:29:10 pi-ager kernel: [   42.539861] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Mar 21 18:29:11 pi-ager kernel: [   43.174811] uart-pl011 3f201000.serial: no DMA platform data
**Mar 22 00:13:22 pi-ager kernel: [20684.303280] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed**
Mar 22 00:13:22 pi-ager kernel: [20684.386581] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
Mar 22 00:13:22 pi-ager kernel: [20684.386605] ieee80211 phy0: brcmf_cfg80211_get_tx_power: error (-5)
Mar 22 00:13:23 pi-ager kernel: [20685.078155] mmc1: card 0001 removed
Mar 22 00:13:23 pi-ager kernel: [20685.138549] mmc1: queuing unknown CIS tuple 0xb1 [78 41 4f 0a aa 7b 02 cc ff 68 80 c7 e2 10 00 8a de 81 4d f6 6e a8 ff 10 94 4d c7 00 b2 49 18 60 ae 00 89 37 90 ff 49 53 ac 05 72 dd 18 b3 d8 ec 54 26 85 ff 0f e2 7b d0 df ff 5e ff ee ed bb 9a] (77 bytes)
Mar 22 00:13:23 pi-ager kernel: [20685.259343] mmc1: queuing unknown CIS tuple 0x14 [78 d0 03 ff 02 28 5a 54 c1 b1 15 68 03 42 e4 82 db d5 71 bb 00 19 83 c3 41 4a b4 c4 4a 31 e1 88 00 b0 17 02 bc b9 5a b8 34 d8 c0 ca 99 21 00 a6 ff fc 20 e4 84 8f ff e5 b5 0b 6d 60 b9 90 42 bf] (243 bytes)
Mar 22 00:14:01 pi-ager kernel: [20722.901036] mmc1: error -22 whilst initialising SDIO card
Mar 22 00:14:01 pi-ager kernel: [20722.979374] mmc1: queuing unknown CIS tuple 0xb1 [78 41 4f 0a aa 7b 02 cc ff 68 80 c7 e2 10 00 8a de 81 4d f6 6e a8 ff 10 94 4d c7 00 b2 49 18 60 ae 00 89 37 90 ff 49 53 ac 05 72 dd 18 b3 d8 ec 54 26 85 ff 0f e2 7b d0 df ff 5e ff ee ed bb 9a] (77 bytes)
Mar 22 00:14:01 pi-ager kernel: [20723.151812] mmc1: queuing unknown CIS tuple 0x14 [78 d0 03 ff 02 28 5a 54 c1 b1 15 68 03 42 e4 82 db d5 71 bb 00 19 83 c3 41 4a b4 c4 4a 31 e1 88 00 b0 17 02 bc b9 5a b8 34 d8 c0 ca 99 21 00 a6 ff fc 20 e4 84 8f ff e5 b5 0b 6d 60 b9 90 42 bf] (243 bytes)
Mar 22 00:14:50 pi-ager kernel: [20772.680698] mmc1: error -22 whilst initialising SDIO card
Mar 22 00:14:51 pi-ager kernel: [20772.779446] mmc1: queuing unknown CIS tuple 0xb1 [78 41 4f 0a aa 7b 02 cc ff 68 80 c7 e2 10 00 8a de 81 4d f6 6e a8 ff 10 94 4d c7 00 b2 49 18 60 ae 00 89 37 90 ff 49 53 ac 05 72 dd 18 b3 d8 ec 54 26 85 ff 0f e2 7b d0 df ff 5e ff ee ed bb 9a] (77 bytes)
Mar 22 00:14:51 pi-ager kernel: [20773.011879] mmc1: queuing unknown CIS tuple 0x14 [78 d0 03 ff 02 28 5a 54 c1 b1 15 68 03 42 e4 82 db d5 71 bb 00 19 83 c3 41 4a b4 c4 4a 31 e1 88 00 b0 17 02 bc b9 5a b8 34 d8 c0 ca 99 21 00 a6 ff fc 20 e4 84 8f ff e5 b5 0b 6d 60 b9 90 42 bf] (243 bytes)
Mar 22 00:16:03 pi-ager kernel: [20845.536942] mmc1: error -22 whilst initialising SDIO card
Mar 22 00:16:03 pi-ager kernel: [20845.688356] mmc1: queuing unknown CIS tuple 0xb1 [78 41 4f 0a aa 7b 02 cc ff 68 80 c7 e2 10 00 8a de 81 4d f6 6e a8 ff 10 94 4d c7 00 b2 49 18 60 ae 00 89 37 90 ff 49 53 ac 05 72 dd 18 b3 d8 ec 54 26 85 ff 0f e2 7b d0 df ff 5e ff ee ed bb 9a] (77 bytes)
Mar 22 00:16:04 pi-ager kernel: [20846.149283] mmc1: queuing unknown CIS tuple 0x14 [78 d0 03 ff 02 28 5a 54 c1 b1 15 68 03 42 e4 82 db d5 71 bb 00 19 83 c3 41 4a b4 c4 4a 31 e1 88 00 b0 17 02 bc b9 5a b8 34 d8 c0 ca 99 21 00 a6 ff fc 20 e4 84 8f ff e5 b5 0b 6d 60 b9 90 42 bf] (243 bytes)
Mar 22 00:18:26 pi-ager kernel: [20987.863019] mmc1: error -22 whilst initialising SDIO card

pelwell, 2023-03-24:
Can anyone experiencing spontaneous reboots after WiFi problems please try adding `over_voltage=4` to config.txt. Even devices with amazing external power supplies can struggle with internal voltage drops and transient load spikes, things that a higher core voltage can help with.

pelwell, 2023-03-24:
The output from `vcgencmd read_ring_osc` would also be interesting.

phylax2020, 2023-03-24:
[image]

pelwell, 2023-03-24:
Thanks - that's 3.068MHz, with a voltage of 1.2813V at 44.5C.

porst17, 2023-03-26:
> @porst17 Am I right in thinking that things are still good for you with the test1 firmware?

No, unfortunately not. I did not check the logs recently and my pi is running a script that reboots the device automatically after a firmware crash. I checked the logs now and a couple of days ago, there was another firmware crash:
```
Mar 22 08:43:12 openhab kernel: [4055980.767845] brcmfmac: CONSOLE: wl0: wlc_scbinit: Cubby failed
Mar 22 08:43:12 openhab kernel: [4055980.767881] brcmfmac: CONSOLE: wl0: wlc_userscb_alloc failed with err -27
Mar 22 08:43:12 openhab kernel: [4055980.767897] brcmfmac: CONSOLE: wl0: wlc_join_BSS_select: out of scbs
Mar 22 08:43:13 openhab kernel: [4055981.649680] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
Mar 22 08:43:13 openhab kernel: [4055981.703359] brcmfmac: CONSOLE: wlc_link, 36727
Mar 22 08:43:13 openhab kernel: [4055981.703394] brcmfmac: CONSOLE: wlc_link, 36849
Mar 22 08:43:13 openhab kernel: [4055981.703413] brcmfmac: CONSOLE:
Mar 22 08:43:13 openhab kernel: [4055981.703432] brcmfmac: CONSOLE: FWID 01-ebc62bc0
Mar 22 08:43:13 openhab kernel: [4055981.703451] brcmfmac: CONSOLE: flags 1
Mar 22 08:43:13 openhab kernel: [4055981.703469] brcmfmac: CONSOLE:
Mar 22 08:43:13 openhab kernel: [4055981.703487] brcmfmac: CONSOLE: TRAP c(7ff7c): pc 80cdd0, lr 716b, sp 40f70, cpsr 2000000c, spsr 21000000
Mar 22 08:43:13 openhab kernel: [4055981.703508] brcmfmac: CONSOLE:   r0 0, r1 2, r2 2, r3 2101, r4 5d420, r5 5d848, r6 5f1f4
Mar 22 08:43:13 openhab kernel: [4055981.703527] brcmfmac: CONSOLE:   r7 0, r8 0, r9 0, r10 5e828, r11 5e828, r12 0
Mar 22 08:43:13 openhab kernel: [4055981.703541] brcmfmac: CONSOLE:
Mar 22 08:43:13 openhab kernel: [4055981.703555] brcmfmac: CONSOLE:    sp+0 00000000 00000002 00000001 0000716b
Mar 22 08:43:13 openhab kernel: [4055981.703570] brcmfmac: CONSOLE:   sp+10 ffffffea ffffffea ffffffea 0005d84a
Mar 22 08:43:13 openhab kernel: [4055981.703585] brcmfmac: CONSOLE:
Mar 22 08:43:13 openhab kernel: [4055981.703603] brcmfmac: CONSOLE: sp+c 0000716b
Mar 22 08:43:13 openhab kernel: [4055981.703618] brcmfmac: CONSOLE: sp+24 0000ffff
Mar 22 08:43:13 openhab kernel: [4055981.703631] brcmfmac: CONSOLE: sp+5c 0080fd1b
Mar 22 08:43:13 openhab kernel: [4055981.703648] brcmfmac: CONSOLE: sp+74 00000457
Mar 22 08:43:13 openhab kernel: [4055981.703661] brcmfmac: CONSOLE: sp+9c 000254a9
Mar 22 08:43:13 openhab kernel: [4055981.703674] brcmfmac: CONSOLE: sp+a0 0000ffff
Mar 22 08:43:13 openhab kernel: [4055981.703687] brcmfmac: CONSOLE: sp+ec 0080ae1d
Mar 22 08:43:13 openhab kernel: [4055981.703702] brcmfmac: CONSOLE: sp+100 0084e2e5
Mar 22 08:43:13 openhab kernel: [4055981.703717] brcmfmac: CONSOLE: sp+104 00802fa9
Mar 22 08:43:13 openhab kernel: [4055981.703730] brcmfmac: CONSOLE: sp+124 0001d901
Mar 22 08:43:13 openhab kernel: [4055981.703743] brcmfmac: CONSOLE: sp+154 0001d9d9
Mar 22 08:43:13 openhab kernel: [4055981.703756] brcmfmac: CONSOLE: sp+16c 008480ab
Mar 22 08:43:13 openhab kernel: [4055981.703769] brcmfmac: CONSOLE: sp+1a4 00830fc9
Mar 22 08:43:13 openhab kernel: [4055981.703782] brcmfmac: CONSOLE: sp+1ec 00023089
Mar 22 08:43:13 openhab kernel: [4055981.703795] brcmfmac: CONSOLE: sp+200 00008ff1
Mar 22 08:43:13 openhab kernel: [4055981.703808] brcmfmac: CONSOLE: sp+21c 008046a9
Mar 22 08:43:15 openhab kernel: [4055984.167609] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
Mar 22 08:43:15 openhab kernel: [4055984.168290] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
Mar 22 08:43:15 openhab kernel: [4055984.168567] brcmfmac: dongle trap info: type 0xc @ epc 0x0080cdd0
Mar 22 08:43:15 openhab kernel: [4055984.168567]   cpsr 0x2000000c spsr 0x21000000 sp 0x00040f70
Mar 22 08:43:15 openhab kernel: [4055984.168567]   lr   0x0000716b pc   0x0080cdd0 offset 0x7ff7c
Mar 22 08:43:15 openhab kernel: [4055984.168567]   r0   0x00000000 r1   0x00000002 r2 0x00000002 r3 0x00002101
Mar 22 08:43:15 openhab kernel: [4055984.168567]   r4   0x0005d420 r5   0x0005d848 r6 0x0005f1f4 r7 0x00000000
Mar 22 08:43:15 openhab kernel: [4055984.168627] ieee80211 phy0: brcmf_link_down: WLC_DISASSOC failed (-110)
Mar 22 08:43:15 openhab kernel: [4055984.168653] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
Mar 22 08:43:15 openhab kernel: [4055984.168686] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
Mar 22 08:43:15 openhab kernel: [4055984.168687] ieee80211 phy0: brcmf_cfg80211_get_tx_power: error (-5)
Mar 22 08:43:15 openhab kernel: [4055984.168707] ieee80211 phy0: brcmf_set_pmk: failed to change PSK in firmware (len=0)
Mar 22 08:43:16 openhab kernel: [4055984.688437] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
Mar 22 08:43:16 openhab kernel: [4055984.688476] ieee80211 phy0: brcmf_cfg80211_reg_notifier: Country code iovar returned err = -5
Mar 22 08:43:16 openhab kernel: [4055984.688601] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
Mar 22 08:43:16 openhab kernel: [4055984.688628] ieee80211 phy0: brcmf_cfg80211_reg_notifier: Country code iovar returned err = -5
Mar 22 08:43:16 openhab kernel: [4055984.849945] brcmfmac: brcmf_sdiod_probe: Failed to set F1 blocksize
Mar 22 08:43:16 openhab kernel: [4055984.850000] brcmfmac: brcmf_sdio_bus_reset: Failed to probe after sdio device reset: ret -123
Mar 22 08:43:16 openhab kernel: [4055984.850566] mmc1: card 0001 removed
[... cut ...]
Mar 22 08:43:21 openhab kernel: [4055989.627866] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
Mar 22 08:43:21 openhab kernel: [4055989.818114] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
Mar 22 08:43:26 openhab kernel: [4055994.719890] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
```
The log above has been generated while the pi was running with `over_voltage=3`:
```
$ vcgencmd read_ring_osc
read_ring_osc(2)=3.608MHz (@1.3312V) (37.6'C)
```
Now running with `over_voltage=4`:
```
$ vcgencmd read_ring_osc
read_ring_osc(2)=3.668MHz (@1.3563V) (48.3'C)
```

aa777888-2, 2023-03-26:
over_voltage=4 not having any positive effects here, either.

pelwell, 2023-03-27:
Thanks for the continued reports. So that we're all on the same page, my current theory is that there are two issues at work here: a firmware bug that causes it to crash, and a systems issue in which a WiFi problem leads to a Pi reboot. For the firmware crashes I can forward the logs to Synaptics. The reboots are a total mystery - if they aren't power-related, I can't think of a mechanism that would cause them.

aa777888-2, 2023-03-27:
Good summary, Phil, thanks. It would appear I'm in the WiFi-leads-to-reboot category. I have tried playing numerous games with my Ubiquiti WiFi environment, such as locking the Pi to a single AP, single band, etc., etc. and with no significant improvement noticed.

pelwell, 2023-03-27:
If you (or anyone) finds any differentiating factor that makes a reboot more or less likely then please pipe up.

We're also interested in exchanging a number of units suffering from spontaneous reboots, partly so we can run some tests on them and partly to see if a different unit being run in the same configuration and environment behaves the same. Email me - phil@raspberrypi.com - with your GitHub handle and we'll work something out.

pelwell, 2023-03-28:
Synaptics have delivered the third test firmware. Same procedure as before: https://drive.google.com/file/d/1ylXbInNgZFV-J6a8h3ZygMSIs6PPXfb5/view?usp=share_link

porst17, 2023-03-28:
I am running `fw_bcmdhd_test3.bin` now. I'll report back as soon as there is something to report.

Krautmaster, 2023-04-07:
for me this was a working workaround if someone wants to test and report:

please try
`wget https://github.com/RPi-Distro/firmware-nonfree/raw/d95f6614b19cb2d3ead4f2e2ea4ba0a7397ac5c0/debian/config/brcm80211/brcm/brcmfmac43436f-sdio.bin`
`cp brcmfmac43436f-sdio.bin /lib/firmware/brcm/brcmfmac43436-sdio.bin`
`reboot`

Krautmaster, 2023-04-11:
> > I am running `fw_bcmdhd_test3.bin` now. I'll report back as soon as there is something to report.
> 
> The `test3` firmware is crashing daily for me:
> ```
> Apr 10 21:06:51 openhab kernel: ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE: wl0.0: wlc_keymgmt_set_bss_tx_key_id:key id 00, key index 65535 is out of range.
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE:
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE: FWID 01-f10d5ce8
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE: flags 1
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE:
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE: TRAP c(6ff7c): pc 80cdd0, lr 7337, sp 41348, cpsr 2000000c, spsr 21000000
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE:   r0 0, r1 2, r2 2, r3 2101, r4 611e4, r5 6160c, r6 62fb8
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE:   r7 1, r8 0, r9 0, r10 625ec, r11 625ec, r12 0
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE:
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE:    sp+0 00000000 00000002 00000001 00007337
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE:   sp+10 0000ffff 00000000 00000000 0006160e
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE:
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE: sp+c 00007337
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE: sp+10 0000ffff
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE: sp+24 0000ffff
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE: sp+5c 0080a22d
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE: sp+64 00005201
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE: sp+8c 00835fa5
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE: sp+ac 0000b347
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE: sp+b4 0083803f
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE: sp+c4 00838173
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE: sp+dc 00011b3b
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE: sp+ec 0081e9e9
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE: sp+fc 00005201
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE: sp+114 008341e5
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE: sp+118 008341dd
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE: sp+11c 0082f4e3
> Apr 10 21:06:51 openhab kernel: brcmfmac: CONSOLE: sp+120 0082f4d1
> Apr 10 21:06:51 openhab kernel: ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
> Apr 10 21:06:51 openhab kernel: ieee80211 phy0: brcmf_cfg80211_get_tx_power: error (-5)
> ```

Can u try the fw from my post above?

porst17, 2023-04-19:
> for me this was a working workaround if someone wants to test and report:
>
> please try `wget [...]

I did what you proposed and it's running stable for more than a week already - no firmware crashes whatsoever. I'll stay on this firmware for a couple more days before testing the new `test4` firmware.

Krautmaster, 2023-04-19:
> It's new trial firmware time (test4): https://drive.google.com/file/d/1Ow1DRR-yV9Z_e8PZzl2zNCE0DMAHDoUk/view?usp=share_link
> You know what to do.

Will try to test the next days

Krautmaster, 2023-04-19:
how can i verify that the new FW was loaded?

phylax2020, 2023-04-19:
I also have had stable conditions with the firmware test3 for more than 10 days. However, I have to admit that I replaced the SD card with a new one before using the test3 firmware. I also removed my PowerBoost 1000C UPS uninterruptible power supply module, which provides a maximum current of 1A at 5.1V for the RPi 0W2.
As you can see in one of my earlier posts, the firmware crash happened at the same time as mmc1 problems (mmc1: card 0001 removed). Since I don't believe in coincidences, there should be a connection between the firmware crash and the SD card problem. It could probably be a problem with a voltage sag in the power supply. 
I will first work with the firmware test3 for a while and do experiments with the PowerBoost 1000C module to rule out power supply problems.

Krautmaster, 2023-04-21:
> I imagine that `dmesg | grep Firmware:` will show a date from this week. Yes - Apr 19 2023.

still crashing for me and the Pi stays offline until hard reboot

pelwell, 2023-04-21:
@Krautmaster Where is the firmware crash in that log? You will see "brcmf_fw_crashed: Firmware has halted or crashed".

Krautmaster, 2023-04-21:
tell me the commands in detail and ill check. Normally that system is online 24/7 and as you can see from the log the Wifi broke down anywhere around April 19th 10pm
Apr 19 22:04:01 raspberrypi kernel: [31970.230540] mmc1: card 0001 removed

pelwell, 2023-04-21:
> tell me the commands in detail and ill check.

https://github.com/raspberrypi/firmware/issues/1768#issuecomment-1373812749

    Apr 19 22:04:01 raspberrypi kernel: [31970.230540] mmc1: card 0001 removed

That's not a firmware crash - that looks more like a hardware issue. What happens once it gets into that state?

Krautmaster, 2023-04-21:
maybe that helps https://pastebin.com/S7JB4UiS

pelwell, 2023-04-21:
That's better. Is that with the test4 firmware? The log isn't long enough to include the Firmware: message.

Krautmaster, 2023-04-21:
yeah, should be. Full Log:  https://cloud.krautmaster.de/index.php/s/Qi2J8BiYoFB8bGd

[35 more replies]