raspberrypi/firmware #1773: FKMS: transparency artifact in drm formats AR24 and AB24

wosk, 2023-01-17:
**Describe the bug**
While run `modetest -P 75@85:800x480@AR24` in fkms mode on yocto I see artifact in top-left quarter of screen. (My screen is upside down).
The issue observed on AR24 and AB24 formats. May be another with alpha channel.
No dependency on plane resolution.
I don't observe the same issue in non-firmware KMS mode.

**To reproduce**
tested on yocto core-image-sato in fkms mode (set dtoverlay=vc4-fkms-v3d in /boot/config.txt).

kill 'GUI window manager'
modetest -p # retrieve CRTC id and plane id
modetest -P <plane_id>@<crtc_id>:800x480@**AR24**
modetest -P <plane_id>@<crtc_id>:800x480@**AB24**

also 
modetest -s 89:#0@AR24 -P 38@87:600x280@AR24

**Expected behaviour**
No artifacts should be seen

**Actual behaviour**
There is a part of image in top-left quarter of screen, which have about 50% of transparency.
modetest -P 75@85:800x480@AR24
[image]
modetest -s 89:#0@AR24 -P 38@87:600x280@AR24 -C
[image]

**System**
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi 4B 8Gb + DSI LCD 7"
* Which OS and version (`cat /etc/rpi-issue`)?
Yocto core-image-sato 
* Which firmware version (`vcgencmd version`)?
The latest from https://github.com/raspberrypi/firmware/tree/master/boot at the moment
* Which kernel version (`uname -a`)?
> Linux raspberrypi4-64 5.15.34-v8 #1 SMP PREEMPT Tue Apr 19 19:21:26 UTC 2022 aarch64 GNU/Linux

**Logs**
modetest -P 38@87:800x480@AR24
testing 800x480@AR24 overlay plane 38

[modetest_p_fkms.log](https://github.com/raspberrypi/firmware/files/10277324/modetest_p_fkms.log)
[modetest_c_fkms.log](https://github.com/raspberrypi/firmware/files/10277325/modetest_c_fkms.log)
[modetest_e_fkms.log](https://github.com/raspberrypi/firmware/files/10277326/modetest_e_fkms.log)

6by9, 2023-01-17:
5.15.34 is pretty old - rpi-5.15..y is now on 5.15.88.

AR24 = DRM_FORMAT_ARGB8888
AB24 = DRM_FORMAT_ABGR8888

DRM_FORMAT_ABGR8888 was only added to fkms in https://github.com/raspberrypi/linux/commit/dbd073e4028580a09b6ee507e0c137441cb52650 in Oct 2022, so how does your kernel from April 2022 support it?

Actually looking with kms I think this is just the premultiplied alpha flag.
modetest appears to be deliberately only setting the per-pixel alpha to opaque in the top left quarter of the image. DRM expects the default to be premultiplied unless the "pixel blend mode" property exists on the plane and is set otherwise.

Adding the premultiplied flag to the firmware is a one-liner and appears to give the same output as vc4-kms-v3d. I'm not intending to add the "pixel blend mode" property to fkms - it already exists in vc4-kms-v3d, and that is the preferred solution now.

6by9, 2023-01-17:
Firmware patch created that selects premultiplied alpha. It'll filter through in due course.

wosk, 2023-01-18:
> DRM_FORMAT_ABGR8888 was only added to fkms in [raspberrypi/linux@dbd073e](https://github.com/raspberrypi/linux/commit/dbd073e4028580a09b6ee507e0c137441cb52650) in Oct 2022, so how does your kernel from April 2022 support it?

ABGR8888 I tested on another kernel.

popcornmix, 2023-01-18:
rpi-update firmware contains potential fix for this

wosk, 2023-01-18:
Looks like the issue was fixed for first plane only and still exist on another planes.
modetest -M v3d -P 3@4096:700x380@AR24 -P 5@4096:500x330@AR24 -P 6@4096:800x230@AR24
[image]