raspberrypi/firmware #1775: Fix docs

eddiewebb, 2023-01-18:
Follow from #1774 , as the terse closer mentioned the docs explicitly say"

"Picamera2 does not use this type of memory,"

But it dont work without it.  Thats a bug in docs wording.  Full details in closed issue.

JamesH65, 2023-01-18:
Picamera2 does not use that memory. However, the video encoder does, which is presumably where the problem is, but that is not part of Picamera2.

pelwell, 2023-01-18:
You are conflating two different effects of reducing the `gpu_mem` value below the default of 32 (MB).
1. The obvious effect is to reduce the amount of memory reserved for the GPU. When using libcamera/picamera2 the firmware makes little use of its memory carve-out.
2. The other effect is to switch to the cut-down version of the firmware, which lacks the codec and ISP support.

In short, just do what the documentation says and remove all `gpu_mem=...` settings.