raspberrypi/firmware #1776: USB-MSB with Samsung T7 Shield fails with eeprom newer than pieeprom-2022-01-25.bin

amahling, 2023-01-19:
**Describe the bug**
I want to use an external USB 3 SSD 'Samsung T7 Shield' for booting my rpi 400 in USB-MSB mode.

Unfortunately this fails with the latest eeproms from critical (2022-12-07) and stable (2023-01-11), when the T7 is connected via an USB 3 (blue) interface. In both cases the bootloader falls back to SD boot. 

With the T7 connected via the USB 2 interface, USB-MSB boot is successful, which prooves that the installed OS is working.

So I decided to try the critical eeproms downwards from the newest, which is 22-12-07 at the time of this bug report. All attempts failed until I reached  2022-01-25. With the latter USB-MSB boot worked as expected.

**To reproduce**
Try USB-MSB boot with Samsung T7 Shield and a critical firmware newer than pieeprom-2022-01-25.bin

**Expected behaviour**
SSD T7 Shield should boot USB-MSB even with actual firmware from critical or stable branch 

**Actual behaviour**
as described above

**System**
 
[raspinfo.txt](https://github.com/raspberrypi/firmware/files/10456752/raspinfo.txt)

**Logs**
sorry, don't know how to get logs from the bootloader stage.

**Additional context**
Add any other relevant context for the problem.

amahling, 2023-01-20:
I managed to flash the SSD with the latest firmware and tried with pieeprom-2022-12-07.bin
The issue persists so I reverted back to pieeprom-2022-01-25.bin

peterharperuk, 2023-01-20:
It could be caused by net install I guess. Could your try adding NET_INSTALL_ENABLED=0 to your bootloader config and see if that helps? (sudo rpi-eeprom-config -e)

amahling, 2023-01-20:
Unfortunately NET_INSTALL_ENABLED=0 did not solve the issue. I will try to take a photo from screen

amahling, 2023-01-20:
I think I've found the reason:
If my monitor (SyncMaster SA650) is configured to autodetect signal input, bootloader behaves as described above.
If autodetection is disabled and a fixed input is used, boot with pieeprom-2022-12-07.bin is OK.
Pi is connected via an hdmi to dvi adapter. Any hint how to configure the raspi to boot from USB with autodetect enabled would be appreciated, on the other hand I can live with it.

timg236, 2023-01-20:
You could try setting DISABLE_HDMI=1 in the EEPROM config to stop the bootloader from reading the EDID OR set HDMI_DELAY to a large number e.g. 30 seconds so you still get the diagnostics screen on failure.

amahling, 2023-01-20:
With HDMI_DELAY=20 and monitor set to autodetect I've tried several reboots and power off/power on. Pi seems to boot stable from USB with that configuration. Thank you very much. I wonder why the old firmware was more robust in this case.

timg236, 2023-01-20:
It's possible that this SSD is sensitive to delays between MSD commands (it shouldn't be). Disabling HDMI diagnostics might remove delays.
Either that, or it's a power issue whilst the drive is initialising.

amahling, 2023-01-21:
The pi 400 is powered with the offical 3A PSU and the SSD uses a [peak current under 1A](https://www.anandtech.com/show/17478/samsung-portable-ssd-t7-shield-review-flagship-pssd-gets-ip65-avatar/3), so I hope it's not a power issue. I can do further investigation when I have access to a powered USB 3 hub. 
From my point of view the issue is solved, so I close it. Thank you for your help.

closed 2023-01-21 by amahling